thesofproject/sof #5113: [BUG]Topology load failure with recent redirection

ranj063, 2021-12-20:
**Describe the bug**
Topology fails to load with error:
```

[ 5.849946] sof-audio-pci-intel-cnl 0000:00:1f.3: loading topology:intel/sof-tplg/sof-hda-generic-4ch.tplg
[ 5.853535] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: invalid header size for type 808464432 at offset 0x0 size 0xe76a.
[ 5.853566] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: topology: invalid header: -22
[ 5.853700] sof-audio-pci-intel-cnl 0000:00:1f.3: error: tplg component load failed -22
[ 5.853730] sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed to load DSP topology -22
[ 5.853754] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[ 5.853997] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: failed to instantiate card -22
```

**To Reproduce**
Build topology with SOF main and boot the device

**Reproduction Rate**
100%
**Expected behavior**
Topology should load successfully.

**Impact**
showstopper

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: https://github.com/thesofproject/sof/commit/afac44af5f493db498d9b5bf4fe8a05f8c78b2f2
2) Name of the topology file
    * Topology: sof-hda-generic-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: HDA CNL

Reverting the patch https://github.com/thesofproject/sof/commit/eb344375228b15529838d6e4992bc7572650ab5e fixes the issue.

ranj063, 2021-12-20:
@marc-hb FYI

marc-hb, 2021-12-20:
I really don't see how eb34437 could make any _content_ difference. If it did this would be exposing a very severe bug in `alsatplg`, much more serious than a simple permission issue. I just reverted it locally and permissions aside all the `.tplg` files are byte for byte identical.

lgirdwood, 2021-12-20:
Could we have a race in the build ? Do we need to force clean topology bimaries ? Do we have stale intermediate topology build files anywhere ?

marc-hb, 2021-12-20:
As a coincidence, I tested this commit with very different number of threads because I was fixing an unrelated and proven CMake race condition with `development/` topologies (#5067). While testing #5067 I never noticed any other race. I tested again with `-j1` and `-j400` and the number of threads makes zero difference to anything.

For the record here's the expected file right now:
```
md5sum tools/build_tools/topology/sof-hda-generic-4ch.tplg 
76ae10de56aec534bf21584eb0b8c2da  tools/build_tools/topology/sof-hda-generic-4ch.tplg

ls -l tools/build_tools/topology/sof-hda-generic-4ch.tplg 
-rw-rw-r--  49855 Dec 20 23:33 tools/build_tools/topology/sof-hda-generic-4ch.tplg
```
@ranj063 told me she got:
```
d86fc7901c51120b95c1d8b06953d2cb /home/ranjani/temp/sof-hda-generic-4ch.tplg.broken # 62195 bytes
```
She also said running the `alsatplg` command manually _with_ the redirection works fine.

@ranj063 can you please:
- run a few times and check whether the (broken) size and checksum stay the same?
- check whether these two files are the same?
```
tools/build_tools/topology/sof-hda-generic-4ch.tplg
tools/build_tools/topology/topology1/production/sof-hda-generic-4ch.tplg
```

ranj063, 2021-12-21:
A few more details about my environment:

Ubuntu 20.04
shell type: bash
alsatplg version 1.2.6
libasound version 1.2.6.1
libatopology version 1.2.6.1

[corrupt-tplg.tar.gz](https://github.com/thesofproject/sof/files/7748696/corrupt-tplg.tar.gz)

And manual redirection with the alsatplg command works fine but redirection with the build-tools.sh script results in a corrupt tplg file. Here's the broken tplg file:

marc-hb, 2021-12-21:
I believe your CMake version was 3.16.3

> @ranj063 can you please:
>
> - run a few times and check whether the (broken) size and checksum stay the same?
> - check whether these two files are the same?

Also: is it affecting just this .tplg file or others or most or all? `diff -qr build_tools.before/ build_tools.after`

aiChaoSONG, 2021-12-22:
There are debug logs from alsatplg in the corrupt topology. The debug option is on with ranjani's alsatplg. in the script, these logs are redirected to the topology file, too.
[image]

marc-hb, 2021-12-22:
Awesome find @aiChaoSONG , you rock! (why didn't I open the attached file mysefl? Maybe it was too late after a long day...)

@ranj063 please use `fprintf(stderr, ...)` for your debug logs, this should avoid the issue.

Closing.

ranj063, 2021-12-24:
I'm sorry @marc-hb , I cannot use stderr for debug logs. That's not the right fix. If debug logs will get redirected too, this patch needs to be reverted. Anyway isn't the original problem for which u submitted this change already been fixed upstream?

marc-hb, 2021-12-24:
> I cannot use stderr for debug logs. 

Why not?

Considering `alsatplg -o -` is supported, how come debug logs go to stdout at the same time? Can you explain how to reproduce this?

> Anyway isn't the original problem for which u submitted this change already been fixed upstream?

Yes it just got fixed days ago. Not everyone is using the latest alsatplg.

marc-hb, 2021-12-24:
> Can you explain how to reproduce this?

OK that was easy. See commit message of revert #5132

PR #5132 (2021-12-25): Revert "topology: redirection to workaround hardcoded alsatplg permissions"

This reverts commit eb344375228b15529838d6e4992bc7572650ab5e.

Fixes #5113

The `alsatplg -o -` feature is incompatible with `-v` and basically
buggy because it happily sends .tplg output and debug statements to
the same place without a care. To reproduce the bug, simply run:
```
   VERBOSE=1 ./scripts/build-tools.sh -T
```
... then observe .tplg output files like this:
```
B0x003d3b/015675 - pcm 'DMIC': write 912 bytes
B0x0040cb/016587 - pcm 'HDMI1': write 912 bytes
B0x00445b/017499 - pcm 'HDMI2': write 912 bytes
B0x [... cut]
commit 8f3f2d6f (2021-12-25): Revert "topology: redirection to workaround hardcoded alsatplg permissions"

This reverts commit eb344375228b15529838d6e4992bc7572650ab5e.

Fixes #5113

The `alsatplg -o -` feature is incompatible with `-v` and basically
buggy because it happily sends .tplg output and debug statements to
the same place without a care. To reproduce the bug, simply run:

   VERBOSE=1 ./scripts/build-tools.sh -T

... then observe .tplg output files like this:

B0x003d3b/015675 - pcm 'DMIC': write 912 bytes
B0x0040cb/016587 - pcm 'HDMI1': write 912 bytes
B0x00445b/017499 - pcm 'HDMI2': write 912 bytes
B0x0047eb/01 [... cut]
closed 2021-12-25 by lgirdwood