thesofproject/sof #5114: [BUG] [v2.0] Error code: -62 0x00000005 0x00000000 0x00004000 occurred in test 17_08_TestMulticoreKdCore1PlaybackD0ix16000Hz16b16b

kkarask, 2021-12-20:
**Describe the bug**
Error occurred in test 17_08_TestMulticoreKdCore1PlaybackD0ix16000Hz16b16b:

Exception SofIpcErrorResponseException occurred at 'C:\\work\\tools\\pyexecnetcache\\cavs_scripts-py\\utilities\\ipc_driver.py':763
FW replied with error.
Error code: -62
0x00000005 0x00000000 0x00004000

Test multi core with keyword detection feature with D0i3 support.
Playback and capture on SSP interface on two internal loopbacks. First playback pipeline on Core 0,
second playback pipeline on Core 1

FW was build with chrome config.

**Topology**
```
  pipe_plb1 core 0
                     +-------------------------+
    +-----------+    | +------+   +----------+ |
    | Host1 plb |----->|Buffer|-->| Dai      |----------------------------------+
    +-----------+    | +------+   +----------+ |                                |
                     +-------------------------+                                |
                                                                                |
        pipe_plb2   core 1                                                      |
                     +-------------------------+                  +-------+     |
    +-----------+    | +------+   +----------+ |                  |       |     |
    | Host2 plb |----->|Buffer|-->| Dai      |------------------> |  SSP1 |     |
    +-----------+    | +------+   +----------+ |                  |       |     |
                     +-------------------------+                  +-------+     |
                                                                                |
                                                                                |
                      pipe_cap core 0                                           v
    +--------------------------------------+                                +-------+
    | +----------+   +------+   +-----+    |                                |       |
    | | KD dummy |<--|Buffer|<--| Dai  |<---------------------------------- |  SSP0 |
    | +----------+   +------+   +-----+    |                                |       |
    +------------------------------^-------+                                +-------+
```

**To Reproduce**
Run test 17_08_TestMulticoreKdCore1PlaybackD0ix16000Hz16b16b with Diagnostic driver.

**Reproduction Rate**
100%

**Environment**
1) Firmware branch name and commit
    Branch: main, tag v2.0
    Last working commit: 689e06d78b6e821dce776deb6ed4abea2f24bc2d, First bad: 92510c938e3107d184d7a7cd0578b42b090cbf9a
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL, ADL

**Screenshots or console output**
```
[     4659840.598958] (       42000.000000) c1 pipe         1.3   ....../pipeline-stream.c:495  WARN pipeline_get_timestamp(): DAI position update failed
[     4681665.312500] (       21824.712891) c0 ipc                  src/ipc/ipc3/handler.c:1576 INFO ipc: new cmd 0x60060000
[     4681680.260417] (          14.947917) c1 idc                ......./intel/cavs/idc.c:58   INFO idc_irq_handler(), IPC_IDCTFC_BUSY
[     4681692.630208] (          12.369792) c1 idc                ......./intel/cavs/idc.c:186  INFO idc_do_cmd()
[     4681703.854167] (          11.223958) c1 pipe         1.3   ....../pipeline-stream.c:261  INFO pipe trigger cmd 2
[     4681835.781250] (         131.927078) c1 dw-dma                 src/drivers/dw/dma.c:371  INFO dw_dma_pause(): dma 0 channel 0 pause
[     4681845.520833] (           9.739583) c1 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:1093 INFO ssp_trigger() cmd 2
[     4681854.192708] (           8.671875) c1 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:1084 INFO ssp_pause(), TX
[     4681866.927083] (          12.734375) c1 ll-schedule        ./schedule/ll_schedule.c:142  INFO task complete 0xbe1f00c0 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3>
[     4681876.302083] (           9.375000) c1 ll-schedule        ./schedule/ll_schedule.c:145  INFO num_tasks 0 total_num_tasks 2
[     4684548.385417] (        2672.083252) c0 ipc                  src/ipc/ipc3/handler.c:1576 INFO ipc: new cmd 0x40080000
[     4684558.854167] (          10.468750) c0 ipc                  src/ipc/ipc3/handler.c:718  INFO ipc: pm gate flags 0x10
[     4897512.421875] (      212953.562500) c0 ll-schedule        ./schedule/ll_schedule.c:451  INFO task add 0x9e1be340 dma-trace-task <2b972272-c5b1-4b7e-926f-0fc5cb4c4690>
[     4897522.005208] (           9.583333) c0 ll-schedule        ./schedule/ll_schedule.c:455  INFO task params pri 4 flags 0 start 500000 period 500000
[     4897535.052083] (          13.046875) c0 ll-schedule        ./schedule/ll_schedule.c:357  INFO new added task->start 209487626 at 209477157
[     4897543.750000] (           8.697917) c0 ll-schedule        ./schedule/ll_schedule.c:360  INFO num_tasks 2 total_num_tasks 2
[     4900447.734375] (        2903.984375) c0 ipc                  src/ipc/ipc3/handler.c:1576 INFO ipc: new cmd 0x80010000
[     4900459.973958] (          12.239583) c0 ipc                  src/ipc/ipc3/handler.c:569  INFO ipc: dai 1.0 -> config 
[     4900469.505208] (           9.531250) c0 dai                           src/lib/dai.c:164  INFO dai_get type 1 index 0 new sref 3
[     4900479.661458] (          10.156250) c0 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:275  INFO ssp_set_config(): Already configured. Ignore config
[     4900489.895833] (          10.234375) c0 dai                           src/lib/dai.c:188  INFO dai_put type 1 index 0 new sref 2
[     4900499.973958] (          10.078125) c0 ipc                      src/ipc/ipc3/dai.c:187  INFO ipc_comp_dai_config() dai type = 1 index = 0
[     4900513.151042] (          13.177083) c0 dai          3.9         src/ipc/ipc3/dai.c:281  INFO dai_config() dai type = 1 index = 0 dd 0x9e13e500
[     4910541.692708] (       10028.541992) c0 idc                           src/idc/idc.c:115  ERROR idc_wait_in_blocking_mode() error: timeout, target_core 1
```
Logs:
[TGL_17_08_TestMulticoreKdCore1PlaybackD0ix16000Hz16b16b.zip](https://github.com/thesofproject/sof/files/7743255/TGL_17_08_TestMulticoreKdCore1PlaybackD0ix16000Hz16b16b.zip)

kv2019i, 2021-12-20:
@lyakh can you take a look?

kkarask, 2021-12-20:
After retest commit: 689e06d78b6e821dce776deb6ed4abea2f24bc2d problem exists too but with lower occurrence rate.

lyakh, 2021-12-20:
@kkarask could you please also attach a full firmware trace when this test isn't failing for comparison? Thanks!

kkarask, 2021-12-20:
@lyakh Logs without fail:
[TGL_pass_without_error_code-62.zip](https://github.com/thesofproject/sof/files/7745341/TGL_pass_without_error_code-62.zip)

Logs with fail:
[TGL_fail_with_error_code-62.zip](https://github.com/thesofproject/sof/files/7745345/TGL_fail_with_error_code-62.zip)

bkokoszx, 2021-12-20:
I think I have fix for this case:
https://github.com/bkokoszx/sof/commit/de4a61d1eb1146db3df31dbb11cb318c9306f4e8

Before @lyakh commit (https://github.com/thesofproject/sof/commit/92510c938e3107d184d7a7cd0578b42b090cbf9a) in above case, ```domain_unregister``` was invoked during ```schedule_task_cancel```

after pause pipeline on core 1. 
```
case COMP_TRIGGER_STOP:
		list_for_item(tlist, &ctx->pipelines) {
			p = container_of(tlist, struct pipeline, list);
			pipeline_schedule_cancel(p);
			p->status = COMP_STATE_PAUSED;
		}
		break;
```
After @lyakh commit we do not hit timer_unregister:
```
static int timer_domain_unregister(struct ll_schedule_domain *domain,
				   struct task *task, uint32_t num_tasks)
{
	struct timer_domain *timer_domain = ll_sch_domain_get_pdata(domain);
	int core = cpu_get_id();

	if (task)
		return 0;

	tr_dbg(&ll_tr, "timer_domain_unregister()");

	/* tasks still registered on this core */
	if (!timer_domain->arg[core] || num_tasks)
		return 0;

	tr_info(&ll_tr, "timer_domain_unregister domain->type %d domain->clk %d",
		domain->type, domain->clk);

	timer_unregister(timer_domain->timer, timer_domain->arg[core]);
```
, because it was invoked with task (not NULL).

lyakh, 2021-12-20:
@bkokoszx thanks for the fix! I posted a comment there, please, have a look

bkokoszx, 2021-12-22:
@lyakh 
I've just created PR #5124 with fix.

PR #5124 (2021-12-23): ll_schedule: unregister domain in free function with IRQ_ONLY_FLAG
scheduler_free_ll() function should invoke domain_unregister()
in case when SOF_SCHEDULER_FREE_IRQ_ONLY flag is used, beacause
in timer domain case, timer_domain_unregister() function
unregisters timer interrupts.

fixes #5114
closed 2021-12-23 by lgirdwood