thesofproject/sof #5120: [BUG] "sof-logger was already dead" happened while testing on some zephyr platforms

keqiaozhang, 2021-12-21:
**Describe the bug**
We observed this error on some zephyr platforms recently. From the console log, it shows that sof-logger was already dead, but there're no obvious errors in dmesg or DMA trace. Since the error trace is not available on zephyr platforms, so we don't know what happened that caused sof-logger to stop working.

the DMA trace shows:
```
warn: log_entry_address        0 is not in dictionary range!
warn: Seeking forward 4 bytes at a time until re-synchronize.

Found valid LDC address after skipping 52 bytes (one line uses 20 + 0 to 16 bytes)
Skipped 0 bytes after the last statement.
error: Invalid filename length 1702256483 or ldc file does not match firmwareerror: read_entry_from_ldc_file(0x2d007) returned -22error: fetch_entry() failed with: -22, aborting
```

**To Reproduce**
eg: on WHL zephyr platform:
$ ./check-playback.sh -d 1 -l 1 -r 50

**Reproduction Rate**
Almost 100%

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
sof-logger stops working

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: ac3b3338-1
SOF Branch: main
SOF Commit:afac44af5f49-2
Zephyr Commit: fef2e30b7f83

**Screenshots or console output**
```
2021-12-20 23:13:27 UTC [REMOTE_COMMAND] aplay   -Dhw:0,5 -r 48000 -c 2 -f S16_LE -d 1 /dev/zero -v -q
Hardware PCM card 0 'sof-hda-dsp' device 5 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16384
  period_size  : 4096
  period_time  : 85333
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4096
  period_event : 0
  start_threshold  : 16384
  stop_threshold   : 16384
  silence_threshold: 0
  silence_size : 0
  boundary     : 4611686018427387904
  appl_ptr     : 0
  hw_ptr       : 0
declare -- cmd="journalctl_cmd --since=@1640041583"
2021-12-20 23:13:30 UTC [REMOTE_INFO] pkill -TERM sof-logger
2021-12-20 23:13:30 UTC [REMOTE_ERROR] sof-logger was already dead
2021-12-20 23:13:31 UTC [REMOTE_INFO] nlines=16604 /home/ubuntu/sof-test/logs/check-playback/2021-12-20-23:06:31-26447/slogger.txt
2021-12-20 23:13:32 UTC [REMOTE_INFO] Test Result: FAIL!
```
** dmesg **
```
[ 1472.825236] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x2014001e successful
[... 67 lines cut ...]
[ 1475.780750] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1475.780787] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1475.781709] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1475.781738] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1475.782718] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1475.782746] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1475.804729] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1475.804760] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1475.805704] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1475.805733] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 1476.234981] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x20140000 successful
[ 1476.235007] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x90050000: GLB_TRACE_MSG: DMA_FREE
[ 1476.235164] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x90050000: GLB_TRACE_MSG: DMA_FREE
[ 1476.267007] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[ 1476.267177] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x40010000: GLB_PM_MSG: CTX_SAVE
[ 1476.267210] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0x1010f0f successful
[ 1476.268485] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f successful
[ 1476.268492] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[ 1476.268568] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Debug PCIR: 00000010 at  00000044
[ 1476.268575] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Turning i915 HDAC power 0
[ 1476.268578] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Current DSP power state: D3
[ 1476.268581] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: fw_state change: 6 -> 0
```

Test ID: 9007, Model name:WHL_UPEXT_HDA_ZEPHYR

dbaluta, 2021-12-21:
@marc-hb i'm also looking at to sof zephyr these days. what should is the status of sof logger support with zephyr?

on my side running:

```
root@imx8mpevk:~# ./sof-logger -t -l sof-imx8.ldc 
 TIMESTAMP         (us)              DELTA  C# COMPONENT          LOCATION                      CONTENT ktime=4062.181s  @  2021-03-24 11:32:51 +0000 UTC
```

but no other message.

while running sof-logger without `-t` option i get a similar error as reported in the issue.

```
root@imx8mpevk:~# ./sof-logger -l sof-imx8.ldc 
 TIMESTAMP         (us)              DELTA  C# COMPONENT          LOCATION                      CONTENT ktime=4281.506s  @  2021-03-24 11:36:30 +0000 UTC

Found valid LDC address after skipping 4 bytes (one line uses 20 + 0 to 16 bytes)
error: Invalid filename length 1969303411 or ldc file does not match firmware
error: read_entry_from_ldc_file(0x2000864d) returned -22
error: fetch_entry() failed with: -22, aborting
Skipped 0 bytes after the last statement. Potential mailbox wrap, check the start of the output for later logs.
```

marc-hb, 2021-12-21:
@dbaluta the DMA trace (sof-logger -t) works normally noww, check any recent sof-ci/jenkins result. The shared memory `etrace` requires a different script, we use https://github.com/zephyrproject-rtos/zephyr/commit/9fecab03eeb3

@keqiaozhang, @dbaluta  you're using the wrong. ldc file, always `rsync` both `.ldc and .ri` files at the same time.

> Also any idea how to modify sof config when compiling with zephyr. The equivalent of make menuconfig.

https://docs.zephyrproject.org/2.3.0/guides/kconfig/menuconfig.html

marc-hb, 2021-12-21:
> From the console log, it shows that sof-logger was already dead, but there're no obvious errors in dmesg or DMA trace. Since the error trace is not available on zephyr platforms, so we don't know what happened that caused sof-logger to stop working.

There will never be an error from the kernel or from the firmware when using the wrong .ldc file. 

> error: Invalid filename length 1702256483 or ldc file does not match firmwareerror: read_entry_from_ldc_file(0x2d007) returned -22error: fetch_entry() failed with: -22, aborting

There is no .ldc checksum with Zephyr right now, working on it.

> always rsync both .ldc and .ri files at the same time.

I also recommend never having any .ldc file in` /etc` to avoid mismatches.

marc-hb, 2021-12-21:
@dbaluta for logging on Zephyr please review #5110 but even more urgently zephyrproject-rtos/zephyr/pull/41282 (3 lines long linker script addition)

marc-hb, 2021-12-22:
I took a closer look at the latest daily test results and I found random trace corruption across ALL tests (and all platforms),  even the PASSing test runs! Most of the time the sof-logger can recover from it and the tests PASS, sometimes there is just too much corruption and the test fails.

It seems to happen at random places, not at any particular point in time.

Trying to reproduce now.

```
[     9058880.473366] (          12.656249) c0 ipc                      src/ipc/ipc3/dai.c:187  INFO ipc_comp_dai_config() dai type = 3 index = 6
[     9058893.858782] (          13.385416) c0 dai          9.61        src/ipc/ipc3/dai.c:281  INFO dai_config() dai type = 3 index = 6 dd 0x9e0795c0
[     9059466.462926] (         572.604126) c0 ipc                  src/ipc/ipc3/handler.c:1576 INFO ipc: new cmd 0x60010000
[     9059492.817091] (          26.354166) c0 pipe         9.62  ....../pipeline-params.c:222  INFO pipe params dir 0 frame_fmt 0 buffer_fmt 9 rate 14
warn: log_entry_address        0 is not in dictionary range!
warn: Seeking forward 4 bytes at a time until re-synchronize.

Found valid LDC address after skipping 64 bytes (one line uses 20 + 0 to 16 bytes)
[     9059639.379586] (         146.562500) c0 dai          9.61        src/ipc/ipc3/dai.c:103  INFO dai_data_config() dai type = 3 index = 6 dd 0x9e0795c0
[     9059653.650418] (          14.270833) c0 hda-dai      3.6   /drivers/intel/hda/hda.c:53   INFO hda_get_hw_params(): channels 2 rate 48000
[     9059702.140000] (          48.489582) c0 dai          9.61           src/audio/dai.c:347  INFO dai_playback_params() dest_dev = 0 stream_id = 0 src_width = 4 dest_width = 4
[     9059715.629583] (          13.489583) c0 dai          9.61           src/audio/dai.c:353  INFO dai_playback_params() fifo 0x0
[     9059745.629581] (          29.999998) c0 pipe         9.62  ....../pipeline-params.c:308  INFO pipe prepare
```

marc-hb, 2021-12-22:
I can finally reproduce but only with XCC. No reproduction with the Zephyr SDK 0.13.2

marc-hb, 2021-12-22:
I had a look at check-playback-100s on APL Zephyr in many recent daily test results. They all have "periodic DMA trace corruption" from the moment I enabled sof-logger in Zephyr tests (daily 8417, November 23rd). What's recent is only how worse it became, until now the sof-logger was merely skipping the corruption.

Every time I saw periodic DMA trace corruption with Zephyr, I checked the logs for XTOS on the same APL with the same versions of everything. There was never any DMA trace corruption with XTOS, only with Zephyr.

marc-hb, 2021-12-22:
> I took a closer look at the latest daily test results and I found random trace corruption across ALL tests (and all platforms), even in the PASSing test runs! 

Actually, not all platforms. It seems to affect only WHL and APL but not TGL. There are some logging failures with TGL but no periodic DMA trace corruption like this. It seemed to have appeared a bit more recently on WHL.

marc-hb, 2021-12-23:
@lyakh got the right intuition (thx! the reason why the corruption is periodic in "stable" tests is because it happens exactly  at the **end of the ring buffer** / wrapping time, every time. In my testing the corruption was either never happening (e.g.: with the Zephyr SDK) or always happening (e.g.: with XCC) at the end of the ring buffer.

Instead of the end of the ring buffer, some corrupted data is sent instead. Interestingly:

- It's not just the last, split and 36 bytes-long (in my testing) log statement that is corrupted: the _previous_ one is corrupted too! About 48 bytes seem corrupted every time.
- The corruption seems deterministic! It always looks like this:

skipping  0x000e0009, skipping  0x06480230, skipping  0x00000000, skipping  0x00000000, skipping  0x00000000, skipping  0x00000000, skipping  0x00000000, skipping  0x00000000, skipping  0x00000000, skipping  0x00000000, skipping  0x00000000, skipping  0x00000000, skipping  0x00000000, skipping  0x00000000, skipping  0x08030007, skipping  0x00000000, 

For reference
```
struct log_entry_header {
	uint32_t uid;
	uint32_t id_0 : TRACE_ID_LENGTH; /* e.g. Pipeline ID */
	uint32_t id_1 : TRACE_ID_LENGTH; /* e.g. Component ID */
	uint32_t core_id : 8;		 /* Reporting core's id */

	uint64_t timestamp;		 /* Timestamp (in dsp ticks) */
	uint32_t log_entry_address;	 /* Address of log entry in ELF */ offsetof=18
} __attribute__((packed));
```
sizeof(header)=20     =  5 words

4 arguments = 16 bytes = 4 words

marc-hb, 2021-12-23:
Some data mining  in the test results (i.e., large `grep -r`) showed that WHL UPX was never experiencing this corruption until daily test 8866 on Dec 12. Starting from the next daily test run 8897, it experienced the same end-of-DMA-ring-buffer corruption systematically. This allowed me to bisect the regression to this particular commit from @andyross (PR zephyrproject-rtos/zephyr/pull/40609)

```
commit 923277a84288ff42e4191520ec4b178623a0e1ac
Author: Andy Ross <andrew.j.ross@intel.com> 
Date:   Thu Nov 18 19:23:43 2021 -0800

    soc/intel_adsp_cavs18: Use new IDC driver
    
    Disable the use of the legacy IDC driver and IPM-based sched_ipi()
    implementation.
    
    Signed-off-by: Andy Ross <andrew.j.ross@intel.com>

 .../intel_adsp/cavs_v18/Kconfig.defconfig.series   | 29 ++++++----------------
 1 file changed, 7 insertions(+), 22 deletions(-)
```

In case you wondered, it's not possible to revert just this commit:

```
zephyr/drivers/ipm/ipm_cavs_idc.c:233: multiple definition of `cavs_idc_smp_init'
zephyr/soc/xtensa/intel_adsp/common/soc_mp.c:286: first defined here
```

It is possible to revert the entire series but that does make the corruption go away. I think this commit is just a trigger/messenger. Of which subtle caching issue I have no idea.

marc-hb, 2021-12-23:
The toolchain situation with WHL UPX is almost "reversed" compared to APL UP2: while the corruption on WHL UPX appeared only recently with XCC (see bisect in previous comment), it's been happening since forever with the Zephyr SDK 0.12.3 on WHL UPX! More precisely, "forever" = zephyr commit cf0c5e2a1ce5 `lib/os: Add sys_heap_usable_size()` which is the oldest zephyr commit  compatible with today's SOF.

Every configuration I tested so far (and there have been many) has behaved deterministically: always corrupting or never corrupting. However predicting which hardware+toolchain+firmware configuration will fail and which will not seems completely random.

marc-hb, 2021-12-24:
Test run https://sof-ci.01.org/sofpr/PR5130/build11462/devicetest/ shows XTOS can also reproduce this but only on CML and especially on one particular CML product (CML_SKU0955_HDA), much more than on the other CML. Interestingly, no problem on TGL+Zephyr. All this run is using XCC AFAIK.

There are also some failures in suspend/resume tests on other platforms but my gut feeling is not to worry too much about ring buffer corruption on suspend/resume.

andyross, 2021-12-24:
Just seeing this the first time: yes, this definitely looks like cache interaction to me.  Can you dump the address and size of the DMA buffer on the offending systems?  Maybe one edge is misaligned or incorrectly flushed?  Another possibility is linker interaction, given the recent churn.  This may have always been potentially overlapping, but the recent tightening up of boundaries might have pushed something important into the gap.

andyross, 2021-12-24:
Looking quickly, it looks like the DMA buffer is just the trace buffer in low memory, at SRAM_TRACE_BASE.  This should be below anything Zephyr touches by definition.  Maybe there's a flushing issue?  The code is leaving dirty cache blocks on one core that get flushed out on top of correct data from another?  That could be sensitive to SMP timing issues, and that IDC switchover you bisected to absolutely improved latency of scheduler IPIs.

andyross, 2021-12-24:
Also, given the application: one thing you could try would be to use an uncached address (e.g. z_soc_uncached_ptr() at runtime, or equivalent math on SRAM_TRACE_BASE) for all access to the buffer.  The local core, almost by definition, will never be reading back any data it writes, the only value the cache provides is a little bit of write combining optimization.

kv2019i, 2022-01-05:
Routing to @ujfalusi and @jsarha for investigation.

lgirdwood, 2022-01-05:
IIRC, the DMA minimum block size may be 2 * cache line size on some platforms. Can we force the DMA block copy size to be multiples of 128 bytes and re test.

marc-hb, 2022-01-19:
> Interestingly, no problem on TGL+Zephyr

It's been happening more on TGL recently:
- https://github.com/thesofproject/sof/pull/5224#issuecomment-1015836091
- In daily tests 9447 , TGLU_UP_HDA_ZEPHYR failed many tests but  WHL_UPEXT_HDA_ZEPHYR only one. 

It really comes and goes as seemingly unrelated code changes come and go. However once a well defined configuration reproduces then it reproduces consistently. "Randomly deterministic".

Also: keep in mind the sof-logger skips a lot of corruption ("green failure")

andyross, 2022-01-19:
Did you ever get a chance to dump the buffer extents (start pointer and length) on a failing configuration?

Also, I'd be really curious whether the problem simply goes away entirely if you use an uncached pointer for the buffer and remove the cache control steps completely.   I'll be honest: I think that code is hurting you more than helping.  The nature of this buffer being streaming and write only means that the **only** value you are getting from the L1 data cache is some write combining (i.e. writing a single word is quicker than a round trip to HP-SRAM).  And for that you pay with a HAL[1] invalidate and flush sequence for every update.  I think it would be faster uncached.  It would surely be smaller.

[1] Another thing you could try on Zephyr builds is to use our z_xtensa_cache_*() API instead of the HAL.   I really doubt the HAL has actual bugs, but I will promise that ours is smaller and better.

jsarha, 2022-01-20:
I was able to produce the issue also on gcc built zephyr firmware on up-i11:

`[    94204324.589990] (     1997820.000000) c0 zll-schedule       src/schedule/zephyr_ll.c:141  INFO ll task 0x9 <bad uid ptr 0x0000000e> avg 560, max 1608
error: log_entry_address          0 is not in dictionary range!
warn: Seeking forward 4 bytes at a time until re-synchronize.

Found valid LDC address after skipping 32 bytes (one line uses 20 + 0 to 16 bytes)
[458599188777.999268] (                nan) c0 unknown      0.0   src/schedule/zephyr_ll.c:141  INFO ll task 0x9e056980 dma-trace-task <2b972272-c5b1-4b7e-926f-0fc5cb4c4690> avg 39, max 490

                 --- negative DELTA = -458504934273.255 us: wrap, IPC_TRACE, other? ---

[   295981042.665853] (           0.000000) c0 zll-schedule       src/schedule/zephyr_ll.c:141  INFO ll task 0xbe05d4c0 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3> avg 3388, max 3422
[     1997819.972697] (     1997820.000000) c0 zll-schedule       src/schedule/zephyr_ll.c:141  INFO ll task 0x9e056fd8 agent-work <c63c4e75-8f61-4420-9319-1395932efa9e> avg 54, max 58
`

jsarha, 2022-01-20:
I also think I found a race in src/trace/dma-trace.c trace_work(). The function first copies the log entries with dma_copy_to_host_nowait() and then sends SOF_IPC_TRACE_DMA_POSITION with ipc_msg_send() right after that. At least in theory the Linux side may be faster in reading the buffer than the DSP is in writing it.  However, implementing a dma_copy_to_host_wait() by adding DMA_COPY_BLOCKING to dma_copy() flags did not fix the issue in my case.

jsarha, 2022-01-20:
Also if my findings - based on some hackish debug code - are correct, the log entries sometimes get corrupted right in the middle of the circular buffer between the DSP and Linux and not (only) when the circular buffer wraps from the end to the beginning. However, this finding should be taken with a grain of salt since it looks like my debug code changes are somewhat changing the behavior of the tracing system.

marc-hb, 2022-01-20:
> the log entries sometimes get corrupted right in the middle of the circular buffer between the DSP and Linux and not (only) when the circular buffer wraps from the end to the beginning. 

Across all my testing I've never seen that. I also analyzed many test logs and the message `error: log_entry_address X is not in dictionary range!` always appeared at very regular intervals.

lgirdwood, 2022-01-31:
@jsarha any update here ?

jsarha, 2022-01-31:
Not yet. Still working on a new angle to tackle the issue.

marc-hb, 2022-02-03:
I'm afraid this has been by far the most frequent test failure in recent daily runs, especially when looking at Zephyr :-(

jsarha, 2022-02-04:
Here is what I have established so far with my upx-i11.

I changed the log record size to 64 bytes (= cache line length) always to make the issue more manageable. On the DSP side the dma buffer is 8192 bytes long and on the Linux host side the dma buffer is is 8 times the size of the DSP side buffer, e.g. 64k.

With 64-byte log entries every 128th entry is always broken. This happens 7 of 8 in the middle of the Linux side buffer, but the entry is always transferred from the last 64bytes of the DSP side buffer. The wrongly transferred entry has always exactly the same content on the Linux side. When reading that same 64-byte entry from DSP side, before the dma_copy() is called for it, the memory content is correct. So somehow the dma_copy() does not copy the last cache line length of the dma buffer correctly (or at all) to the Linux host side.

The DMA_GW version used on the setup does not pass any other parameters to dma_copy() but size, so the error must be on the DMA initialization code. I have been scrutinizing the the DMA initialization code for a while now, but it is a bit task to fully understand what is going on there.

Feel free to ping me for details, if you are interested.

miRoox, 2022-02-11:
In today's daily test (10105), this issue still happened on TGLH_RVP_NOCODEC_ZEPHYR and APL_UP2_NOCODEC_ZEPHYR.

marc-hb, 2022-02-11:
@miRoox I looked at these test results. The sof-logger dies too but with immediately and with a different error message. In this bug the sof-loggers was dying only after many `error: log_entry_address 0x.... is not in dictionary range!` warnings which I don't see in 10105

I will spend more time scanning the logs to make sure but so far I think this is a different issue.

miRoox, 2022-02-11:
@marc-hb thanks for your correction, now the DMA trace shows:
```
Found valid LDC address after skipping 12 bytes (one line uses 20 + 0 to 16 bytes)
error: Invalid filename length 31304 or ldc file does not match firmware
error: read_entry_from_ldc_file(0x23006) returned -22
error: fetch_entry() failed with: -22, aborting
Skipped 0 bytes after the last statement.
```

Perhaps this is a different issue.

marc-hb, 2022-02-11:
I filed follow-up issue #5345

marc-hb, 2022-02-11:
@miRoox could you please file yet another issue for the immediate failures on TGL?

closed 2022-02-11 by marc-hb

marc-hb, 2022-02-12:
>  could you please file yet another issue for the immediate failures on TGL?

I did it in new
- #5352