thesofproject/sof #5125: [BUG] sof-coredump-reader gdb script has a fixed reset vector address

kv2019i, 2021-12-22:
**Describe the bug**
The sof/tools/coredumper/sof-coredump-reader.py script parses a DSP oops and converts this to a gdb command script that can be used with xtensa xt-gdb for analysis of the coredump.

The script sets a breakpoint to reset vector and this address is fixed in the output. This is problematic as the reset vector address is not 
same for all targets. The coredump reader should support a way to adjust the reset vector address and/or discover that from environment (e.g. Xtensa core config for xtensa systems).

**To Reproduce**
Cause a DSP panic to occur, capture the exception and run coredumper for the exception. 
Start xt-gdb with XTENSA_CORE set to a Intel cAVS core config.
A xtensa ISS memory allocation failure is hit and no backtrace is printed.

**Reproduction Rate**
100%

**Expected behavior**
Backtrace should be shown correctly.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: n/a
    * SOF: v2.0
2) Name of the topology file
    * Topology: n/a
3) Name of the platform(s) on which the bug is observed.
    * Platform: Intel TGL

**Screenshots or console output**

kv2019i, 2021-12-22:
On Intel TGL platforms, this problem can be worked around by doing:
```
sed -i -es/0xbefe0000/0x9f180000/ dsp-coredump.gdb
```
... to the coredump-reader output (see https://github.com/thesofproject/sof-docs/pull/391 for the recently added documentation on how to use this).

cujomalainey, 2021-12-23:
@lkoenig @bzhg FYI

kv2019i, 2022-01-11:
This is fixed with #5147 , closing.

closed 2022-01-11 by kv2019i