thesofproject/sof #5135: [BUG] Cannot load firmware on i7-1165G7

lenghonglin, 2021-12-26:
**Describe the bug**
cannot load signed firmware v1.8
[image]

**To Reproduce**
follow this [part](https://thesofproject.github.io/latest/getting_started/build-guide/build-from-scratch.html#step-3-build-toolchains-from-source) to build toolchian and update sof firmware.
[image]

build success.

and i replace `sof-cnl.ri` and `sof-cnl.ldc` to `/lib/firmware/intel/sof/intel-signed`, then reboot

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.11.0-43
    * SOF: 9e7a8f921
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: i7-1165G7

lgirdwood, 2021-12-26:
@lenghonglin the otc_private key only works with hardware that has the otc key programmed into the BIOS. This key update can only usually be done by OEMs. Can you confirm if the Intel signed v1.8 or Intel signed v1.9 works on your HW.

lenghonglin, 2021-12-27:
> 

https://github.com/thesofproject/sof-bin  i can update sof with this repo, Intel signed v1.8 , Intel signed v1.9 , Intel signed v2.0 all work fine.

But using the SOF version I compiled myself was problematic.

lenghonglin, 2021-12-27:
what difference between [community](https://github.com/thesofproject/sof-bin/tree/main/v1.8.x/sof-v1.8/community) and [intel-signed](https://github.com/thesofproject/sof-bin/tree/main/v1.8.x/sof-v1.8/intel-signed).

the `community version` show the same error.

lenghonglin, 2021-12-27:
update kernel to 5.13 now

lgirdwood, 2021-12-27:
@lenghonglin the community version is signed with a *public* signing key. This means anyone can sign it (so it's not secure), but it is very useful for developers using developer platforms like Chromebooks and UP boards (which all support the public "otc" key).
The Intel key is secret to Intel only and will work on end user devices (like laptops and PCs) that run Linux.

lenghonglin, 2021-12-28:
> @lenghonglin the community version is signed with a _public_ signing key. This means anyone can sign it (so it's not secure), but it is very useful for developers using developer platforms like [...]

It mean i can't build sof by myself?

marc-hb, 2021-12-28:
> It mean i can't build sof by myself?

It depends which audio keys have been configured in your UEFI firmware image. It's product specific.
https://thesofproject.github.io/latest/getting_started/intel_debug/introduction.html#firmware-binary

> Platform: i7-1165G7

This information does not really matter, this Intel processor will use any key found in the UEFI image.

Most system vendors configure their products to allow only official releases signed by Intel, Chromebooks allow any image signed with the not secret, pseudo-private "community" key, UP products allow both keys after flashing the UEFI image in a certain way: https://github.com/thesofproject/sof-bin/issues/32,... anything is possible.

lenghonglin, 2021-12-28:
> 

`byt` `bdw` `cht` don't need to be signed.  but how can I find the specific model for the platform?

lgirdwood, 2021-12-28:
> > 
> 
> `byt` `bdw` `cht` don't need to be signed. but how can I find the specific model for the platform?

By default most Intel based devices will use the Intel key, except for Chromebooks and UP boards (and possibly other devices from other manufacturers). There is no easy way to check this except by trying to boot the FW.

closed 2021-12-29 by lenghonglin

lenghonglin, 2021-12-30:
Is there a way for the dsp to output `dbg` logs?  @lgirdwood

mengdonglin, 2021-12-31:
> Is there a way for the dsp to output `dbg` logs which don't need build new firmware? @lgirdwood

@lenghonglin I think NO. Because the firmware binaries from https://github.com/thesofproject/sof-bin are for release, not for FW development.  Since you're deep diving in FW, I suggest you no longer use OEM x86 laptops which always requires SOF FW signed by Intel production key. Please consider using a open x86 platform which can run SOF FW signed by the community key (OTC key). e.g. Tiger Lake UP Xtreme i11 board https://up-shop.org/up-xtreme-i11-boards-series.html  They also have Celeron and i3 options which are cheaper than i5.  or Whiskey Lake UP Xtreme board  https://up-shop.org/up-xtreme-series.html.   You will be in using the same HW as us running Zephyr + SOF. Personally I would suggest Tiger Lake UP Xtreme i11 board as its newer and faster.

lgirdwood, 2021-12-31:
@lenghonglin you can also use a Chromebook for SOF development or certain development platforms from NXP, Mediatek or AMD. I will create a feature request so we can list all the updated and new HW that can be used on sof-docs