thesofproject/sof #5140: No card sound deteted TECLAST F15 Intel Celeron/Pentium Silver Processor High Definition Audio

wisbit, 2022-01-01:
Hi,
bought a new laptop, Teclast F15+.
Intel Celeron/Pentium Silver Processor High Definition Audio
Audio card chip ID : 8086:3198

Only HDMI sound detected

Inxi :
`Audio:
  Device-1: Intel Celeron/Pentium Silver Processor High Definition Audio
    driver: snd_hda_intel v: kernel alternate: snd_soc_skl,snd_sof_pci_intel_apl
    bus-ID: 00:0e.0 chip-ID: 8086:3198 class-ID: 0401
  Sound Server-1: ALSA v: k5.15.7-1-MANJARO running: yes
  Sound Server-2: JACK v: 1.9.19 running: no
  Sound Server-3: PulseAudio v: 15.0 running: yes
  Sound Server-4: PipeWire v: 0.3.40 running: yes
`

`[wisbix@wisbix ~]$ pactl list cards
Card #0
        Name: alsa_card.pci-0000_00_0e.0
        Driver: module-alsa-card.c
        Owner Module: 6
        Properties:
                alsa.card = "0"
                alsa.card_name = "HDA Intel PCH"
                alsa.long_card_name = "HDA Intel PCH at 0xa1210000 irq 128"
                alsa.driver_name = "snd_hda_intel"
                device.bus_path = "pci-0000:00:0e.0"
                sysfs.path = "/devices/pci0000:00/0000:00:0e.0/sound/card0"
                device.bus = "pci"
                device.vendor.id = "8086"
                device.vendor.name = "Intel Corporation"
                device.product.id = "3198"
                device.product.name = "Celeron/Pentium Silver Processor High Definition Audio"
                device.form_factor = "internal"
                device.string = "0"
                device.description = "Built-in Audio"
                module-udev-detect.discovered = "1"
                device.icon_name = "audio-card-pci"
        Profiles:
                output:hdmi-stereo: Digital Stereo (HDMI) Output (sinks: 1, sources: 0, priority: 5900, available: no)
                output:hdmi-surround: Digital Surround 5.1 (HDMI) Output (sinks: 1, sources: 0, priority: 800, available: no)
                output:hdmi-surround71: Digital Surround 7.1 (HDMI) Output (sinks: 1, sources: 0, priority: 800, available: no)
                output:hdmi-stereo-extra1: Digital Stereo (HDMI 2) Output (sinks: 1, sources: 0, priority: 5700, available: no)
                output:hdmi-surround-extra1: Digital Surround 5.1 (HDMI 2) Output (sinks: 1, sources: 0, priority: 600, available: no)
                output:hdmi-surround71-extra1: Digital Surround 7.1 (HDMI 2) Output (sinks: 1, sources: 0, priority: 600, available: no)
                output:hdmi-stereo-extra2: Digital Stereo (HDMI 3) Output (sinks: 1, sources: 0, priority: 5700, available: no)
                output:hdmi-surround-extra2: Digital Surround 5.1 (HDMI 3) Output (sinks: 1, sources: 0, priority: 600, available: no)
                output:hdmi-surround71-extra2: Digital Surround 7.1 (HDMI 3) Output (sinks: 1, sources: 0, priority: 600, available: no)
                output:hdmi-stereo-extra3: Digital Stereo (HDMI 4) Output (sinks: 1, sources: 0, priority: 5700, available: no)
                output:hdmi-surround-extra3: Digital Surround 5.1 (HDMI 4) Output (sinks: 1, sources: 0, priority: 600, available: no)
                output:hdmi-surround71-extra3: Digital Surround 7.1 (HDMI 4) Output (sinks: 1, sources: 0, priority: 600, available: no)
                output:hdmi-stereo-extra4: Digital Stereo (HDMI 5) Output (sinks: 1, sources: 0, priority: 5700, available: no)
                output:hdmi-surround-extra4: Digital Surround 5.1 (HDMI 5) Output (sinks: 1, sources: 0, priority: 600, available: no)
                output:hdmi-surround71-extra4: Digital Surround 7.1 (HDMI 5) Output (sinks: 1, sources: 0, priority: 600, available: no)
                off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
        Active Profile: off
        Ports:
                hdmi-output-0: HDMI / DisplayPort (type: HDMI, priority: 5900, latency offset: 0 usec, not available)
                        Properties:
                                device.icon_name = "video-display"
                        Part of profile(s): output:hdmi-stereo, output:hdmi-surround, output:hdmi-surround71
                hdmi-output-1: HDMI / DisplayPort 2 (type: HDMI, priority: 5800, latency offset: 0 usec, not available)
                        Properties:
                                device.icon_name = "video-display"
                        Part of profile(s): output:hdmi-stereo-extra1, output:hdmi-surround-extra1, output:hdmi-surround71-extra1
                hdmi-output-2: HDMI / DisplayPort 3 (type: HDMI, priority: 5700, latency offset: 0 usec, not available)
                        Properties:
                                device.icon_name = "video-display"
                        Part of profile(s): output:hdmi-stereo-extra2, output:hdmi-surround-extra2, output:hdmi-surround71-extra2
                hdmi-output-3: HDMI / DisplayPort 4 (type: HDMI, priority: 5600, latency offset: 0 usec, not available)
                        Properties:
                                device.icon_name = "video-display"
                        Part of profile(s): output:hdmi-stereo-extra3, output:hdmi-surround-extra3, output:hdmi-surround71-extra3
                hdmi-output-4: HDMI / DisplayPort 5 (type: HDMI, priority: 5500, latency offset: 0 usec, not available)
                        Properties:
                                device.icon_name = "video-display"
                        Part of profile(s): output:hdmi-stereo-extra4, output:hdmi-surround-extra4, output:hdmi-surround71-extra4
`

aplay -l
`[wisbix@wisbix ~]$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 9: HDMI 3 [HDMI 3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 10: HDMI 4 [HDMI 4]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
`

No luck with sof firmware.
Nothing shows up, no sound, no card.

lgirdwood, 2022-01-02:
@wisbit can you attach your dmesg.

wisbit, 2022-01-02:
[wisbix@wisbix ~]$ sudo dmesg | grep audio
[    3.926772] snd_hda_intel 0000:00:0e.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])

is that enough or do you wish me to add the whole thing ?

lgirdwood, 2022-01-02:
> @lgirdwood [wisbix@wisbix ~]$ sudo dmesg | grep audio [ 3.926772] snd_hda_intel 0000:00:0e.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
> 
> is that enough or do you wish me to add the whole thing ?

Best to grep for "snd" and "sof" as context is missing above.

wisbit, 2022-01-03:
@lgirdwood 

[wisbix@wisbix ~]$ sudo dmesg | grep sof
[    0.677920] PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
[    0.677924] software IO TLB: mapped [mem 0x000000006fba6000-0x0000000073ba6000] (64MB)

[wisbix@wisbix ~]$ sudo dmesg | grep snd
[    3.926772] snd_hda_intel 0000:00:0e.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])

there. is there anything else I can provide, I'll be happy to.

lgirdwood, 2022-01-03:
@wisbit it does not look like the sof driver is being loaded as this device could be a HDA device only (i.e. with no DSP enabled).  You could check your BIOS menu for a DSP option.
@kv2019i is there a way we can read back the HDA PCI DSP caps to show DSP support ? This device looks like a GLK refresh,

kv2019i, 2022-01-04:
This line reveals DSP is present:
> bus-ID: 00:0e.0 chip-ID: 8086:3198 **class-ID: 0401**

@wisbit Thing to confirm:
1) you have SOF support for GLK, e.g. CONFIG_SND_SOC_SOF_GEMINILAKE is enabled in kernel config
    - please attach the kernel config you used so we can verify
    - alternatively please try a recent Fedora or Ubuntu live image and see if the card is found
2) does the BIOS declare a PCH-connected DMIC  (this will trigger loading SOF)
    - assuming (1) is ok, you should see "Digital mics found on Skylake+ platform, using SOF driver" in kernel dmesg
    - we'd need to see the full dmesg to verify this
    - alternatively, add "options snd_intel_dspcfg dsp_driver=3" to some file in /etc/modprobe.d/ and reboot -- assuming (1) is ok, this should force SOF to be loaded even if BIOS doesn't report correct data

wisbit, 2022-01-04:
@kv2019i 
by kernel config, you mean what is is /etc/default/grub ? if not, where is that ?
I've tried a recent Ubuntu (21.?) live image but card wasn't detected.

you mention CONFIG_SND_SOC_SOF_GEMINILAKE, gemini lake is the name of my graphics card, my audio card is Intel Celeron/Pentium Silver Processor High Definition Audio, does that change anything ?

cheers

kv2019i, 2022-01-04:
@wisbit With kernel config I mean this (should work on Ubuntu systems at least)
cat /boot/config-`uname -r`

E.g. on my Ubuntu machine  I can confirm the kernel is built with GLK support:
cat /boot/config-`uname -r` |grep SOF_GEMIN
CONFIG_SND_SOC_SOF_GEMINILAKE_SUPPORT=y
CONFIG_SND_SOC_SOF_GEMINILAKE=m

Gemini Lake is a Intel code name for the CPU on your system:
https://ark.intel.com/content/www/us/en/ark/products/codename/83915/products-formerly-gemini-lake.html

From your PCI id (8086:3198) we can conclude your system has a Gemini Lake (GLK) CPU and the SOF GLK driver should work on your system.

wisbit, 2022-01-04:
this is my kernel configuration in /etc/default/grub
https://bin.disroot.org/?4fe84ef09578b7d1#4mxqbNR8b4Qc2jPcTcSx7Jr9pC4JFnAgxe752TKk4WyN

this is full dmesg at boot
https://bin.disroot.org/?0d04a31e1dd0df45#8uwe4e6zMjfY6TfdRjhnK95GmFysaJSqHdZ2wGScTarK

This is the output of sudo /usr/bin/dmidecode
https://bin.disroot.org/?5066a293545c6dbb#hEuxn6MuMtzHJcgkSATstcigYricP7j93238PZbw89q

kv2019i, 2022-01-05:
Thanks @wisbit ! There are no hints about the audio codec. It does not seem to be a HDA codec at least. So maybe this is another system with a I2S codec like these Teclast devices, https://github.com/thesofproject/linux/issues/3336 . 
To figure out whether this is the case, can you do this:

- install acpica-tools and attach the NHLT information (dump.bin.gz)

sudo acpidump > dump.bin
gzip dump.bin

You can see discussion in https://github.com/thesofproject/linux/issues/3336 for how this was debugged. Moving this to kernel as there is no evidence this is a SOF FW issue.