thesofproject/sof #5142: [FEATURE] Open usecase/pipeline without being triggered by host

sanzazacuro, 2022-01-03:
**Describe the solution you'd like**
Two items:

- How can a source pipeline be triggered on DSP without being triggered by pcm stream from host? For instance, run a source pipeline (pipeline with a module generating a signal) as always on after boot.

- How can a sink pipeline be set to always on, and constantly consume dmic data? For instance, run `intel-generic-dmic` pipeline as always on after boot.

Thanks.

dbaluta, 2022-01-03:
@ranj063 i remember you said you have some code for hostless components?

lgirdwood, 2022-01-03:
The host id intended to control pipelines, but there should be nothing preventing some init code calling the IPC functions directly to create pipelines and start/stop them. This is probably a bit of extra effort atm, but @kv2019i was looking at a simplified audio API to be used for host less implementations.

hongshui3000, 2022-01-05:
Looking forward to hostless implementation

sanzazacuro, 2022-01-06:
@kv2019i Would the simplified hostless API lgirdwood mentioned above be a wrapper for ipc4/helper.c and ipc4/handler.c?

lgirdwood, 2022-01-10:
@sanzazacuro we are thinking something higher level API, the IPC is low level tightly coupled to the host drivers (and will change over time). @kv2019i did you an example  API in mind ? Were you thinking something like ALSA lib or smaller/simpler ?

plbossart, 2022-01-10:
What about the codec configuration part? You still need to do the hw_params/prepare/trigger for the codec driver, and only the kernel can do this. I don't think a firmware-only solution would help at all, would it?

lgirdwood, 2022-01-10:
> What about the codec configuration part? You still need to do the hw_params/prepare/trigger for the codec driver, and only the kernel can do this. I don't think a firmware-only solution would help at all, would it?

I think the use case here is no host, i.e. a stand alone board like a smart device running an rtos like zephyr.

kv2019i, 2022-01-11:
I'm afraid no ready code yet. When we looked at this for early Zephyr/SOF work, we ended up using the SOF Linux driver and existing SOF topology files in the end still. While more complex to get something new up and running, there is still benefits to having the ability to reuse all the graphs and most important (for our use-case back then), having the ability to modify the graph without touching any code.

But before this, we explored host-free alternatives. The idea we had was to embed a static pipeline as part of the FW image, captured as a sequence of IPCs (set of commands a host would sent to create the described pipelines).  At FW init, you'd feed the fixed IPCs to construct the runtime audio pipelines and start the streaming. This has been tracked as https://github.com/thesofproject/sof/issues/1556
The tool to convert a topology file into a IPC command blob, was not developed, so the issue remains open.

Nothing prevents from using the audio component interfaces directly, e.g. build the pipeline(s) as is done in sof/src/ipc/ipc3/helper.c:ipc_pipeline_new(). This is basicly skipping the IPC part.

ranj063, 2022-01-11:
> @ranj063 i remember you said you have some code for hostless components?

@dbaluta yes, I have code from 3 years ago rotting somewhere to make the tone pipeline work. But it needs to revisited and possibly re-written.

sanzazacuro, 2022-01-12:
Thanks for info @kv2019i , I will experiment a little more with the IPC handlers.

lgirdwood, 2022-01-12:
@sanzazacuro also please see the testebench. It does use teh IPC code and the also calls the pipeline APIs directly. We could have a higher level "stream" API where streams can be configured and controlled and another higher level "topology" API where topologies can be constructed (both these APIs could bypass the IPC calls and use the underlying logic called by IPC to do the same thing). This would guarantee a stable API over time immune to IPC updates.