thesofproject/sof #5150: how to trigger dummy_dma_copy?

lenghonglin, 2022-01-05:
Hello , everyone, I am adapting SOF to the new platform.

I wanna use `dummy_dma` to copy host buffer to dsp buffer, of course, dsp can directly access
all of the host physical memory.

The current progress is that sof already complete the pipeline and can be TRIG_START.

If i use hardware DMA  to copy buffer, then dsp receive `TRIG_START` , the will call `dma_start()` within `host_trigger`, the begin to copy buffer. 

when finished dam copy, will  triger dma irq callback, and  call `pipe_copy`——>`host_dma_cb()`, so i can get the data in `host_dma_cb()`.

dbaluta, 2022-01-05:
dumma_dma is a dummy DMA not associated with any DMA controller. We needed this in order to fit into SOF having Host copy the date from/to AP using DMA.

@paulstelian97 can you offer more details?

lenghonglin, 2022-01-06:
And i found `IMX8` don't use `zephyr_ll_scheduler` rather than `schedule_task_init_ll`.
[image]

Why?

paulstelian97, 2022-01-06:
`dummy_dma` just does its copies inside the `copy()` callback itself, which is what is called by the IRQ handler (which was at least originally due to using DMA scheduling; I'm unsure of the state of the transition to Zephyr-compatible scheduling as of now).

The IRQ from the DAI DMA triggers `pipeline_copy()` which in turn will call `host_copy()` which itself calls `dummy_dma_copy()`. Maybe a few other intermediate calls but this is it in a nutshell.

If `pipeline_copy()` is called from a timer interrupt or other scheduling, then `dummy_dma_copy()` will run inside that.

dbaluta, 2022-01-06:
@lenghonglin I think we are still using schedule_task_init_ll because the interrupt support in Zephyr for imx8 is not complete. Full switch to zephyr is planned for this year for all imx8 supported platforms.

lenghonglin, 2022-01-07:
> @lenghonglin I think we are still using schedule_task_init_ll because the interrupt support in zephyr (西风) for imx8 is not complete. Full switch to zephyr (西风) is planned for this year for all imx8 supported platforms.

Got it, I'm just very curious,Thanks to answer~

lenghonglin, 2022-01-07:
Slove it! Thanks everyone, the audio stream already run on my platform

closed 2022-01-07 by lenghonglin