thesofproject/sof #5164: [BUG]DSP panic when playback with signal stop-start on BDW_WSB_RT286

miRoox, 2022-01-07:
**Describe the bug**
In the internal daily test 9209, DSP panic is observed on BDW_WSB_RT286 when playback with signal stop-start.

**To Reproduce**
```
TPLG=/lib/firmware/intel/sof-tplg/sof-bdw-rt286.tplg MODEL=BDW_WSB_RT286 ~/sof-test/test-case/check-signal-stop-start.sh -m playback -c 50
```

**Reproduction Rate**
Only once in the daily test.

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: https://github.com/thesofproject/linux/commit/7446c50ebb128c85d61294a6c8000cf88e986fd8
SOF Branch: main
SOF Commit: https://github.com/thesofproject/sof/commit/308a24a92b4460708a09c1a485cb5c28fcb418dc
Topology: /lib/firmware/intel/sof-tplg/sof-bdw-rt286.tplg
Platform: BDW_WSB_RT286

**Screenshots or console output**
*dmesg*
```
[ 1213.156960] kernel: sof-audio-acpi-intel-bdw INT3438:00: pcm: prepare stream 1 dir 0
[ 1213.156982] kernel: sof-audio-acpi-intel-bdw INT3438:00: pcm: hw params stream 1 dir 0
[ 1213.156997] kernel: sof-audio-acpi-intel-bdw INT3438:00: stream_tag 0
[ 1213.157017] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 1213.157706] kernel: sof-audio-acpi-intel-bdw INT3438:00: ------------[ DSP dump start ]------------
[ 1213.157731] kernel: sof-audio-acpi-intel-bdw INT3438:00: DSP panic!
[ 1213.157748] kernel: sof-audio-acpi-intel-bdw INT3438:00: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 1213.157843] kernel: sof-audio-acpi-intel-bdw INT3438:00: reason: runtime exception (0x6)
[ 1213.157860] kernel: sof-audio-acpi-intel-bdw INT3438:00: trace point: 0x00000800
[ 1213.157877] kernel: sof-audio-acpi-intel-bdw INT3438:00: panic at :0
[ 1213.157893] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: DSP Firmware Oops
[ 1213.157908] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: Exception Cause: LoadStoreErrorCause, Processor internal physical address or data error during load or store
[ 1213.157925] kernel: sof-audio-acpi-intel-bdw INT3438:00: EXCCAUSE 0x00000003 EXCVADDR 0x00000004 PS       0x00060525 SAR     0x00000000
[ 1213.157945] kernel: sof-audio-acpi-intel-bdw INT3438:00: EPC1     0x00000e63 EPC2     0x00000000 EPC3     0x0001bd3d EPC4    0x00000000
[ 1213.157966] kernel: sof-audio-acpi-intel-bdw INT3438:00: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[ 1213.157984] kernel: sof-audio-acpi-intel-bdw INT3438:00: EPS2     0x00000000 EPS3     0x00060520 EPS4     0x00000000 EPS5    0x00000000
[ 1213.158002] kernel: sof-audio-acpi-intel-bdw INT3438:00: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00002040
[ 1213.158021] kernel: sof-audio-acpi-intel-bdw INT3438:00: stack dump from 0x0041f200
[ 1213.158043] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f200: 00000000 ffffffd8 0041f220 0041ec00
[ 1213.158063] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f210: c001c2c4 0041f250 0041f280 00000003
[ 1213.158160] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f220: 0041f200 00000000 00000000 00000000
[ 1213.158181] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f230: 0041f250 00000001 00000004 004024e0
[ 1213.158201] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f240: 4001e13f 0041f280 00000000 20009438
[ 1213.158221] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f250: 00000e63 0041f2e0 0040d42c 0001bda4
[ 1213.158240] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f260: 00414d2c 0040d438 0040d438 0040d438
[ 1213.158259] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f270: 80019b38 0041f480 1fffa300 c001e13f
[ 1213.158278] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: ipc host -> DSP: pending no complete no raw 0x00000800
[ 1213.158296] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: mask host: pending yes complete no raw 0x7fff0002
[ 1213.158314] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: ipc DSP -> host: pending yes complete no raw 0x8dead006
[ 1213.158332] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: mask DSP: pending no complete no raw 0x7f9e0000
[ 1213.158349] kernel: sof-audio-acpi-intel-bdw INT3438:00: ------------[ DSP dump end ]------------
[ 1213.158365] kernel: sof-audio-acpi-intel-bdw INT3438:00: fw_state change: 6 -> 7
[ 1213.158454] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: trace IO error
[ 1213.663226] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx timed out for 0x60010000 (msg/reply size: 108/20)
[ 1213.663250] kernel: sof-audio-acpi-intel-bdw INT3438:00: preventing DSP entering D3 state to preserve context
[ 1213.663257] kernel: sof-audio-acpi-intel-bdw INT3438:00: ------------[ DSP dump start ]------------
[ 1213.663265] kernel: sof-audio-acpi-intel-bdw INT3438:00: Firmware exception
[ 1213.663274] kernel: sof-audio-acpi-intel-bdw INT3438:00: fw_state: SOF_FW_CRASHED (7)
[ 1213.663356] kernel: sof-audio-acpi-intel-bdw INT3438:00: reason: runtime exception (0x6)
[ 1213.663365] kernel: sof-audio-acpi-intel-bdw INT3438:00: trace point: 0x00000800
[ 1213.663374] kernel: sof-audio-acpi-intel-bdw INT3438:00: panic at :0
[ 1213.663382] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: DSP Firmware Oops
[ 1213.663390] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: Exception Cause: LoadStoreErrorCause, Processor internal physical address or data error during load or store
[ 1213.663399] kernel: sof-audio-acpi-intel-bdw INT3438:00: EXCCAUSE 0x00000003 EXCVADDR 0x00000004 PS       0x00060525 SAR     0x00000000
[ 1213.663411] kernel: sof-audio-acpi-intel-bdw INT3438:00: EPC1     0x00000e63 EPC2     0x00000000 EPC3     0x0001bd3d EPC4    0x00000000
[ 1213.663421] kernel: sof-audio-acpi-intel-bdw INT3438:00: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[ 1213.663431] kernel: sof-audio-acpi-intel-bdw INT3438:00: EPS2     0x00000000 EPS3     0x00060520 EPS4     0x00000000 EPS5    0x00000000
[ 1213.663440] kernel: sof-audio-acpi-intel-bdw INT3438:00: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00002040
[ 1213.663449] kernel: sof-audio-acpi-intel-bdw INT3438:00: stack dump from 0x0041f200
[ 1213.663461] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f200: 00000000 ffffffd8 0041f220 0041ec00
[ 1213.663472] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f210: c001c2c4 0041f250 0041f280 00000003
[ 1213.663482] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f220: 0041f200 00000000 00000000 00000000
[ 1213.663491] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f230: 0041f250 00000001 00000004 004024e0
[ 1213.663500] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f240: 4001e13f 0041f280 00000000 20009438
[ 1213.663510] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f250: 00000e63 0041f2e0 0040d42c 0001bda4
[ 1213.663519] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f260: 00414d2c 0040d438 0040d438 0040d438
[ 1213.663529] kernel: sof-audio-acpi-intel-bdw INT3438:00: 0x0041f270: 80019b38 0041f480 1fffa300 c001e13f
[ 1213.663538] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: ipc host -> DSP: pending no complete no raw 0x00000800
[ 1213.663548] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: mask host: pending no complete no raw 0x7fff0000
[ 1213.663557] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: ipc DSP -> host: pending no complete yes raw 0x4dead006
[ 1213.663566] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: mask DSP: pending no complete no raw 0x7f9e0000
[ 1213.663575] kernel: sof-audio-acpi-intel-bdw INT3438:00: ------------[ DSP dump end ]------------
[ 1213.663585] kernel: sof-audio-acpi-intel-bdw INT3438:00: HW params ipc failed for stream 0
[ 1213.663594] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: set pcm hw_params after resume
[ 1213.663602] kernel: sof-audio-acpi-intel-bdw INT3438:00: ASoC: error at snd_soc_pcm_component_prepare on INT3438:00: -110
[ 1213.663613] kernel:  Media Playback 1: ASoC: __soc_pcm_prepare() failed (-110)
[ 1213.663625] kernel:  Media Playback 1: ASoC: dpcm_fe_dai_prepare() failed (-110)
[ 1213.664035] kernel: sof-audio-acpi-intel-bdw INT3438:00: pcm: free stream 1 dir 0
[ 1213.664135] kernel: sof-audio-acpi-intel-bdw INT3438:00: pcm: close stream 1 dir 0
[ 1216.162476] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: fail in sof_ipc_trace_update_filter -19
[ 1216.174445] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: fail in sof_ipc_trace_update_filter -19
```

[full_dmesg.txt](https://github.com/thesofproject/sof/files/7825980/full_dmesg.txt)
[console_log.txt](https://github.com/thesofproject/sof/files/7825981/console_log.txt)
[sof-logger.txt](https://github.com/thesofproject/sof/files/7825986/sof-logger.txt)

lgirdwood, 2022-01-07:
@miRoox are you able to run the stack trace script with the output ?

miRoox, 2022-01-08:
@lgirdwood sorry, this error only appeared once in daily tests and I can't reproduce it manually with even 500 stop-start iterations (`-c 500`).

XiaoyunWu6666, 2022-01-24:
haven't seen it in CI after its first occurence

closed 2022-01-24 by XiaoyunWu6666