thesofproject/sof #5168: [BUG] ipc tx error on suspend/resume

specious, 2022-01-10:
Sound is not restored on suspend/resume with the recent 5.16-rc7 kernel.

This is the kernel log on wake:

```
[54581.864703] Restarting tasks ... 
[54581.865400] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx error for 0x80010000 (msg/reply size: 216/12): -19
[54581.865416] sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed resetting DAI config for HDA0.OUT
[54581.867204] done.
[54581.871672] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops 0xffffffff822ad480)
[54581.872203] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx error for 0x30010000 (msg/reply size: 96/20): -22
[54581.872213] sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed to load widget HDA0.OUT
[54581.873490] sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed widget list set up for pcm 0 dir 0
[54581.873499] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[54581.873505]  HDA Analog: ASoC: soc_pcm_hw_params() failed (-22)
[54581.873537]  HDA Analog: ASoC: dpcm_fe_dai_hw_params failed (-22)
[54581.873831] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx error for 0x30010000 (msg/reply size: 96/20): -22
[54581.873839] sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed to load widget HDA0.OUT
[54581.873843] sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed setting up DAI widget HDA0.OUT
[54581.873847] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on Analog CPU DAI: -22
[54581.873864]  Analog Playback and Capture: ASoC: soc_pcm_hw_params() failed (-22)
[54581.873868]  HDA Analog: ASoC: dpcm_fe_dai_hw_params failed (-22)
[54581.874014] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx error for 0x30010000 (msg/reply size: 96/20): -22
[54581.874023] sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed to load widget HDA0.OUT
[54581.874025] sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed setting up DAI widget HDA0.OUT
[54581.874028] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on Analog CPU DAI: -22
[54581.874038]  Analog Playback and Capture: ASoC: soc_pcm_hw_params() failed (-22)
[54581.874041]  HDA Analog: ASoC: dpcm_fe_dai_hw_params failed (-22)
[54581.874278] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx error for 0x30010000 (msg/reply size: 96/20): -22
[54581.874294] sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed to load widget HDA0.OUT
[54581.874297] sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed setting up DAI widget HDA0.OUT
[54581.874299] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on Analog CPU DAI: -22
[54581.874309]  Analog Playback and Capture: ASoC: soc_pcm_hw_params() failed (-22)
[54581.874316]  HDA Analog: ASoC: dpcm_fe_dai_hw_params failed (-22)
[54581.874606] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx error for 0x30010000 (msg/reply size: 96/20): -22
[54581.874614] sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed to load widget HDA0.OUT
[54581.874618] sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed setting up DAI widget HDA0.OUT
[54581.874621] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on Analog CPU DAI: -22
[54581.874633]  Analog Playback and Capture: ASoC: soc_pcm_hw_params() failed (-22)
[54581.874636]  HDA Analog: ASoC: dpcm_fe_dai_hw_params failed (-22)
```

It works perfectly with the 5.15.13 kernel.

Perhaps it has something to do with the [recent changes](https://lore.kernel.org/alsa-devel/20211125101520.291581-4-kai.vehmanen@linux.intel.com/T/).

**Impact**

Annoyance (need to reboot to restore sound)

**Environment**

This is on a Thinkpad X1 Carbon gen 8 with Intel Comet Lake PCH-LP  cAVS audio.

* Kernel: 5.16-rc7 mainline
* SOF: (with kernel)
* Platform: Arch x86_64 with custom built kernel

**Log output**

[dmesg.txt](https://github.com/thesofproject/sof/files/7837116/dmesg.txt)

lgirdwood, 2022-01-10:
@specious how often does this happen ? Fwiw, the changes https://lore.kernel.org/alsa-devel/20211125101520.291581-4-kai.vehmanen@linux.intel.com/T/ were a fix for a similar issue on CML hardware so it may be something is still not right (and this fix was stress tested on a lot of HW).
Btw, are you able to run sof-logger next time this happens and attache the FW log. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

lgirdwood, 2022-01-10:
@kv2019i fyi.

specious, 2022-01-11:
It happens consistently on wake. I can boot and have audio, but after the first suspend/resume, it will require a reboot to get it back.

I'd love to help diagnose what is occurring. I can boot into the 5.16-rc7 kernel and test it out. I just installed the `sof-tools` package and it's got the logger and two other utilities.

How exactly do I run the logger? I'm not sure what the `ldc` file is and where it comes from. It also looks like I need to boot with `snd_sof.sof_debug=1` (I have it compiled into the kernel).

plbossart, 2022-01-11:
@specious here are some ideas to help find the root cause

a) downgrade to SOF version 1.9.3 . I see that you are using 2:0:0-b678a so we need to rule out a firmware issue in the last release

b) prevent dynamic pipelines for HDaudio. this can be done with the quirk

options snd_sof sof_debug=0xF

````c
/* Flag definitions used in sof_core_debug (sof_debug module parameter) */
#define SOF_DBG_ENABLE_TRACE	BIT(0)
#define SOF_DBG_RETAIN_CTX	BIT(1)	/* prevent DSP D3 on FW exception */
#define SOF_DBG_VERIFY_TPLG	BIT(2) /* verify topology during load */
#define SOF_DBG_DYNAMIC_PIPELINES_OVERRIDE	BIT(3) /* 0: use topology token
							* 1: override topology
							*/
#define SOF_DBG_DYNAMIC_PIPELINES_ENABLE	BIT(4) /* 0: use static pipelines
							* 1: use dynamic pipelines
							*/
#define SOF_DBG_DISABLE_MULTICORE		BIT(5) /* schedule all pipelines/widgets
							* on primary core
							*/
#define SOF_DBG_PRINT_ALL_DUMPS		BIT(6) /* Print all ipc and dsp dumps */
````

lgirdwood, 2022-01-14:
@specious ```/lib/firmware/intel/sof``` and ```/lib/firmware/intel/sof-tplg``` are directories e.g. on my PC
```
drwxr-xr-x 6 root root    4096 Dec 21 09:11 sof
drwxr-xr-x 4 root root   32768 Dec 21 09:11 sof-tplg
```
These are then populated with the binaries and links.

xxxbxxx, 2022-01-17:
got the same issue!

workaround to avoid reboot:
```bash
sudo rmmod snd_sof_pci_intel_cnl
sudo modprobe snd_sof_pci_intel_cnl
```

specious, 2022-01-17:
If the module is built into the kernel, is there also a way to reinitialize it?

specious, 2022-01-18:
My kernel configuration: [config-linux-5.16-20220117.txt](https://github.com/thesofproject/sof/files/7886114/config-linux-5.16-20220117.txt)

specious, 2022-01-18:
My kernel configuration: [config-linux-5.16-20220118.txt](https://github.com/thesofproject/sof/files/7886592/config-linux-5.16-20220118.txt)

Although all drivers are built in, everything works perfectly with the 5.15.13 kernel, and until a sleep/wake also on 5.16.0.

I was simply wondering if there's any way to get the driver to reinitialize without doing an `rmmod`.

dumprop, 2022-01-23:
I have same problem on opensuse - https://bugzilla.suse.com/process_bug.cgi
alsa-info.sh with broken audio https://alsa-project.org/db/?f=9aa7cd2010b1474ccc835b55d807077bc69018d6

carmilso, 2022-01-24:
Hi! I have exactly the same issue but with `snd_sof_pci_intel_tgl`. Thanks for the `rmmod` workaround, I'll keep subscribed to the issue

tiwai, 2022-01-24:
> I was simply wondering if there's any way to get the driver to reinitialize without doing an `rmmod`.

You can unbind and rebind via sysfs. It's a hard-way, equivalent with the device hot-unplug/plug the PCI device, so it might hit other bugs, though.

lgirdwood, 2022-01-24:
@dumprop @carmilso do you guys see the same line in your dmesg ?
```
[54581.871672] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops 0xffffffff822ad480)
[54581.872203] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx error for 0x30010000 (msg/reply size: 96/20): -22
[54581.872213] sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed to load widget HDA0.OUT
[54581.873490] sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed widget list set up for pcm 0 dir 0
[54581.873499] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
```
That shows the HDA0.OUT widget loading message. This may point to some IP programming flow in FW that is failing when we try and use that HDA stream.

dumprop, 2022-01-24:
> @dumprop @carmilso do you guys see the same line in your dmesg ?
>
> ```
> [54581.871672] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops 0xffffffff822ad480) [...]

Yes, unless I can't see `mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops 0xffffffff822ad480`. I have:
```
[   10.777128] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
```
(on boot, if it's make sense)
```
[ 4787.005478] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x80010000 (msg/reply size: 216/12): -19
[ 4787.005496] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed resetting DAI config for HDA0.OUT
[ 4787.035273] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x30010000 (msg/reply size: 96/20): -22
[ 4787.035289] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load widget HDA0.OUT
[ 4787.036568] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed widget list set up for pcm 0 dir 0
[ 4787.036580] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[ 4787.036589]  HDA Analog: ASoC: soc_pcm_hw_params() failed (-22)
[ 4787.036644]  HDA Analog: ASoC: dpcm_fe_dai_hw_params failed (-22)
[ 4787.036897] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x30010000 (msg/reply size: 96/20): -22
[ 4787.036904] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load widget HDA0.OUT
[ 4787.036908] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed setting up DAI widget HDA0.OUT
[ 4787.036912] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on Analog CPU DAI: -22
[ 4787.036922]  Analog Playback and Capture: ASoC: soc_pcm_hw_params() failed (-22)
[ 4787.036926]  HDA Analog: ASoC: dpcm_fe_dai_hw_params failed (-22)
```
(after resume, so there are no anything related to mei_hdcp)

tmoesi, 2022-01-27:
Hi,
I have a very similar issue and have to do
```
sudo rmmod snd_sof_pci_intel_tgl
sudo modprobe snd_sof_pci_intel_tgl
```
to get my audio to work after wakeup from suspend.

This is my dmesg:
```
sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x80010000 (msg/reply size: 216/12): -19
sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed resetting DAI config for HDA0.OUT
sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x30010000 (msg/reply size: 96/20): -22
sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load widget HDA0.OUT
done.
sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed widget list set up for pcm 0 dir 0
sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
 HDA Analog: ASoC: soc_pcm_hw_params() failed (-22)
mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
 HDA Analog: ASoC: dpcm_fe_dai_hw_params failed (-22)
sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x30010000 (msg/reply size: 96/20): -22
sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load widget HDA0.OUT
sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed setting up DAI widget HDA0.OUT
sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on Analog CPU DAI: -22
 Analog Playback and Capture: ASoC: soc_pcm_hw_params() failed (-22)
 HDA Analog: ASoC: dpcm_fe_dai_hw_params failed (-22)
sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x30010000 (msg/reply size: 96/20): -22
sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load widget HDA0.OUT
sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed setting up DAI widget HDA0.OUT
sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on Analog CPU DAI: -22
 Analog Playback and Capture: ASoC: soc_pcm_hw_params() failed (-22)
 HDA Analog: ASoC: dpcm_fe_dai_hw_params failed (-22)
sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x30010000 (msg/reply size: 96/20): -22
sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load widget HDA0.OUT
sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed setting up DAI widget HDA0.OUT
sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on Analog CPU DAI: -22
 Analog Playback and Capture: ASoC: soc_pcm_hw_params() failed (-22)
 HDA Analog: ASoC: dpcm_fe_dai_hw_params failed (-22)
sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x30010000 (msg/reply size: 96/20): -22
sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load widget HDA0.OUT
sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed setting up DAI widget HDA0.OUT
sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on Analog CPU DAI: -22
 Analog Playback and Capture: ASoC: soc_pcm_hw_params() failed (-22)
 HDA Analog: ASoC: dpcm_fe_dai_hw_params failed (-22)
```
**Environment**

I am running a Lenovo Yoga 7 slim with kernel `5.16.2-1-MANJARO`.
Information around the hardware can also be found at https://linux-hardware.org/?probe=896b348390.

tiwai, 2022-01-27:
A similar bug report on bugzilla.kernel.org https://bugzilla.kernel.org/show_bug.cgi?id=215535

It suggests that the first bad commit is 5fcdbb2d45df6afb654674379546996b0027aa3e
("ASoC: SOF: Add support for dynamic pipelines")

lgirdwood, 2022-01-27:
@plbossart @ranj063 is there a kernel config or module option that can disable dynamic pipelines so we can root cause and workaround atm ?

ujfalusi, 2022-01-27:
Hi,

While trying to reproduce a different issue I think I did hit this issue:
https://github.com/thesofproject/sof/issues/5134#issuecomment-1022943267

In short: 5.16.1-artix1-1 kernel
suspend while aplay is running and on resume I see
```
[ 2330.127230] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x80010000
[ 2330.127389] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x80010000 (msg/reply size: 216/12): -19
[ 2330.127394] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed resetting DAI config for HDA0.OUT
```
stopping aplay and starting it again will fail and I have this in the dmesg:
```
[   47.869300] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x60010000 (msg/reply size: 108/20): -22
[   47.869307] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hw params ipc failed for stream 1
[   47.869310] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[   47.869320]  HDA Analog: ASoC: soc_pcm_hw_params() failed (-22)
[   47.869392] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[   47.869587] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[   47.869601]  HDA Analog: ASoC: dpcm_fe_dai_hw_params failed (-22)
```

Checked that vanilla 5.16 and 5.16.1 also have the same issue, but 5.17-rc1 does not. With some bisecting
```
git bisect start '--term-new=fixed' '--term-old=unfixed' '--' 'sound/soc/sof/'
# fixed: [e783362eb54cd99b2cac8b3a9aeac942e6f6ac07] Linux 5.17-rc1
git bisect fixed e783362eb54cd99b2cac8b3a9aeac942e6f6ac07
# unfixed: [df0cc57e057f18e44dac8e6c18aba47ab53202f9] Linux 5.16
git bisect unfixed df0cc57e057f18e44dac8e6c18aba47ab53202f9
# fixed: [d9a7246534753efa383ad8d05ab3691df846c4b4] ASoC: SOF: Add a helper for freeing PCM stream
git bisect fixed d9a7246534753efa383ad8d05ab3691df846c4b4
# unfixed: [8253aa4700b37cef1ca3bbda0d986349357608d3] ASoC: SOF: imx: Add code to manage DSP related clocks
git bisect unfixed 8253aa4700b37cef1ca3bbda0d986349357608d3
# unfixed: [65c16dd2942f4476a3f96a2625b1475c6137c09a] ASoC: SOF: Add PM support for i.MX8/i.MX8X/i.MX8M
git bisect unfixed 65c16dd2942f4476a3f96a2625b1475c6137c09a
# fixed: [cd2f33e93d578e9e1c47ef8981ec69298da9cb38] ASoC: SOF: Intel: power optimizations with HDaudio SPIB register
git bisect fixed cd2f33e93d578e9e1c47ef8981ec69298da9cb38
# fixed: [96da174024b9c63bd5d3358668d0bc12677be877] ASoC: SOF: handle paused streams during system suspend
git bisect fixed 96da174024b9c63bd5d3358668d0bc12677be877
# unfixed: [fb71d03b29bcbd8c03798d36e7b2a2297b6dea45] ASoC: SOF: topology: don't use list_for_each_entry_reverse()
git bisect unfixed fb71d03b29bcbd8c03798d36e7b2a2297b6dea45
# first fixed commit: [96da174024b9c63bd5d3358668d0bc12677be877] ASoC: SOF: handle paused streams during system suspend
```

I ended up backporting the attached patches to 5.16.1 from 5.17-rc1 and the resume issue is resolved:
[backports_from_5.17-rc1.tar.gz](https://github.com/thesofproject/sof/files/7951589/backports_from_5.17-rc1.tar.gz)

@ranj063, @plbossart, @kv2019i 

I'll try to find a simpler way to fix it, but this is a good start.

dmesg from the start of the first playback to resume: [dmesg.txt](https://github.com/thesofproject/sof/files/7951691/dmesg.txt)
firmware log for the same window: [firmware-log.txt](https://github.com/thesofproject/sof/files/7951699/firmware-log.txt)

dumprop, 2022-02-02:
> this **96da174024b9c63bd5d3358668d0bc12677be877** (ASoC: SOF: handle paused streams during system suspend) commit added in **5.16.3** - https://cdn.kernel.org/pub/linux/kernel/v5.x/ChangeLog-5.16.3

Yes, now I don't notice such a problem

lgirdwood, 2022-02-04:
Closing since resolved with kernel update.

closed 2022-02-04 by lgirdwood