thesofproject/sof #5172: Dummy output and no sound with KUU Xbook laptop (ES8336)

hav4k, 2022-01-10:
Hi.

Just install OpenSUSE Tumbleweed with last kernel 5.16 and SOF-bin, fixed dummy output but persist no sound. What can i do? Bluetooth works fine, speakers and headphones not. 

sudo dmesg | grep -i sound shows something of quirk 0x30... i will edit later with output.

Regards and thanks for help.

lgirdwood, 2022-01-10:
@hav4k are you able to attach the full dmesg. It initially looks like the sound server (can you confirm pulseaudio or pipewire) is not able to use the SOF audio card.

hav4k, 2022-01-10:
inxi:

Audio:     Device-1: Intel Celeron/Pentium Silver Processor High Definition Audio driver: sof-audio-pci-intel-apl
           Sound Server-1: ALSA v: k5.16.0-2-default running: yes
           Sound Server-2: PipeWire v: 0.3.42 running: yes
           
          [dmesg.txt](https://github.com/thesofproject/sof/files/7839131/dmesg.txt)

Do you need more details? Thanks for reply, @lgirdwood

lgirdwood, 2022-01-10:
> inxi:
>
> Audio: Device-1: Intel Celeron/Pentium Silver Processor High Definition Audio driver: sof-audio-pci-intel-apl Sound Server-1: ALSA v: k5.16.0-2-default running: yes Sound Server-2: PipeWire [...]

Sorry, not able to read this dmesg. Can you attach to your reply by drag and drop.

hav4k, 2022-01-10:
[dmesg.txt](https://github.com/thesofproject/sof/files/7841285/dmesg.txt)
@lgirdwood

plbossart, 2022-01-10:
like the same issue as https://github.com/thesofproject/linux/issues/3336