thesofproject/sof #5193: [BUG] [Topology1] Macro "PIPELINE_FORMAT" not expanded

aiChaoSONG, 2022-01-12:
**Describe the bug**
The macro `PIPELINE_FORMAT` is not replaced with its value, for example, s32le, in the generated conf by m4.

**To Reproduce**
Run script to build topology: `./script/build-tools.sh -T`
Search the keyword `PIPELINE_FORMAT` from sof/tools/build-tools/topology/\*.conf

Issue introduced by c687815185948d49350e4b58d59e786ac16339e1, confirmed.

aiChaoSONG, 2022-01-12:
@lkoenig FYI

lgirdwood, 2022-01-12:
That's good to know PIPELINE_FORMAT is being set by accident. @lkoenig are you able to fix and set to s32le ?

lkoenig, 2022-01-12:
@lgirdwood I took the option to define it as the DAI_FORMAT. Should I instead define it as s32le ?

lgirdwood, 2022-01-12:
> @lgirdwood I took the option to define it as the DAI_FORMAT. Should I instead define it as s32le ?

No, I think the fix is fine. Lets wait for @aiChaoSONG to validate and close.

lkoenig, 2022-01-13:
PCM_ID is also reported to not be defined.
The issue  PCM_ID is caused by detect pipeline being created with PIPELINE_ADD instead of PIPELINE_PCM_ADD.
detect pipeline expect to be attached to a PCM.
I wonder if the correct fix would be to use PIPELINE_PCM_ADD in those case  ? @lgirdwood what do you think ?

lkoenig, 2022-01-17:
PR #5194 contains CL to address issues shown in that bug report.
The .conf files after those change differs from the ones prior c687815 
A quick fix could be also to revert  c687815

closed 2022-01-17 by mwasko

lkoenig, 2022-01-17:
@mwasko just wanted to make sure you saw the comment I made [above](https://github.com/thesofproject/sof/issues/5193#issuecomment-1014274876)

PR #5194 (2022-01-17): topology: define PIPELINE_FORMAT from DAI_ADD
Pipeline extects PIPELINE_FORMAT to be defined. When including a
pipeline using DAI_ADD, DAI_FORMAT is defined but no PIPELINE_FORMAT.
Prior c687815, the last previously defined PIPELINE_FORMAT was used.
This change ensure the requested DAI_FORMAT is used in the pipeline.

This resolve #5193.