thesofproject/sof #5195: [BUG][Zephyr] FW not suspended after dmic test

kv2019i, 2022-01-12:
**Describe the bug**
 check-runtime-pm-status-15 fails on SOF Zephyr build on UP WHL board
See Intel internal daily plan #9293

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-hda-generic-4ch.tplg MODEL=WHL_UPEXT_HDA_ZEPHYR ~/sof-test/test-case/check-runtime-pm-status.sh -l 15 

**Reproduction Rate**

**Expected behavior**
Test should pass (system should go to runtime suspend)

**Impact**
Increased power consumption. 

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 55744127
    * SOF Commit: 9c002f37e0f0 
    * Zephyr Commit: 768c5d42c0bd 
2) Name of the topology file
    * Topology: sof-hda-generic-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**

kv2019i, 2022-01-12:
Can you @marc-hb take a look (you have the board)? Tagging P1 as this is blocking CI.

XiaoyunWu6666, 2022-01-13:
This may not be a single issue .

Below are some findings observed from inner CI daily testresults and our issue identification records :
Since  UP WHL board was added to CI (late Nov ,2021)  , it had the issue of"ipc timed out when TRIG_STOP with multiple-pipeline test" . 
check-runtime-pm-status-15 is the case that goes after this ,  and it got affected and always failed at the first round (so it appears to fail to suspend after dmic arecord)

XiaoyunWu6666, 2022-01-13:
I add a 'reset' after multiple-pipeline-all-50 testcase , then the runtime-pm-status doesn't get affected 
please check inner planresult 9342

miRoox, 2022-01-18:
This failure is caused by the previous failure about multiple pipelines, which is already reported in https://github.com/thesofproject/sof/issues/4680. As @XiaoyunWu6666 said, if we reboot system after multiple-pipeline-all-50, the subsequent tests won't fail. Therefore, this is not a real bug but just DSP state had been corrupted at this point.

closed 2022-01-18 by miRoox