thesofproject/sof #5238: x5-e8000 sound eventually locks to a screech

dlarue, 2022-01-21:
I have a dozen Vulkan 10" tablets running Kubuntu 21.10 and it was sof-bin which allowed sound to start working but myself and some others have experience the sound system failing with a constant screech sound after a few minutes of playback. It takes logging out to stop it.
I've installed sof-bin version 2.0.x and was wondering if you thought another version might be better suited for this hardware?

Attaching dmesg and lspci output.
[outfile.dmesg.txt](https://github.com/thesofproject/sof-bin/files/7910273/outfile.dmesg.txt)
[outfile.lspci.txt](https://github.com/thesofproject/sof-bin/files/7910274/outfile.lspci.txt)

plbossart, 2022-01-21:
It's likely the same problem we've seen on Baytrail for ages. 
see https://github.com/thesofproject/sof/issues/3868

please try the same work-around with the trace enabled:

add this to /etc/modprobe.d/alsa-base.conf

options snd_sof sof_debug=1

dlarue, 2022-01-21:
Still get the screech with "option snd_sof sof_debug=1" in the modprob.d config file.
Trying the sof-cht.ri which was posted in a zip file on one of the machines now but that didn't solve the problem for one on that thread.

plbossart, 2022-01-21:
@dlarue just to confirm, you need to reboot for the option to take effect.

If this doesn't work, the only option is to switch back to the legacy driver:
options snd-intel-dspcfg dsp_driver=2

This issue has been known for about a year and all the development team is busy on newer products, I am skeptical on this ever being fixed.

@lgirdwood FYI.

dlarue, 2022-01-21:
Yes, I've been rebooting just to make darn sure things grabbed changes.
setting debug=1 didn't work but it seems "options snd-intel-dspcfg dsp_driver=3" is lasting a long time so far.
I've seen this mentioned in the thread and it was not set by default in Kubuntu 21.10.
What is driver option 3?  options snd-intel-dspcfg dsp_driver=3

plbossart, 2022-01-21:
dsp_driver = 0 -> autodetection, this will dependent on Kconfig options 
CONFIG_SND_INTEL_BYT_PREFER_SOF=y -> SOF driver and open-source firmware
CONFIG_SND_INTEL_BYT_PREFER_SOF=n -> SST legacy driver and closed-source firmware

forcing a non-zero value overrides the autodetection and Kconfig options
dsp_driver = 1 -> irrelevant in this case.
dsp_driver = 2 -> SST legacy driver and closed-source firmware.
dsp_driver = 3 -> SOF driver and open-source firmware

dlarue, 2022-01-21:
ok, thanks for explaining.
It's been running on one machine for about an hour so far with dsp_driver=3 option, the debug option and the posted sof-cht.ri
Now it's been looping on a long audio file for over 2 hours on two machines without failure.

dlarue, 2022-01-23:
I'm starting to believe it does work on the x5-e8000 media processor in these tablets after it ran for a day looping an audio file on one tablet and another played a 2 hour video twice without failure.  

It did not work with just the sof_debug=1 entry and required including specifying the dsp_drive to use the open-source SOF driver.

options snd-intel-dspcfg dsp_driver=3
options snd_sof sof_debug=1

I've asked a couple of others to also verify this works and I'm waiting for confirmation before closing this issue.

lgirdwood, 2022-01-24:
also related/duplicate of #4662

plbossart, 2022-01-24:
Thanks for testing and sharing your results @dlarue 

Can I ask you to try to hibernate and see if audio works when you resume the device? That would help with other reports we're seeing on CHT devices. Thanks!

dlarue, 2022-01-24:
I don't currently have a swap partition on the tablets and so it only sleeps right now. It survives sleep just fine.
I'll boot one tablet from USB and create a 4GB swap partition( 4GB of RAM ) and see if I can enable hibernate to test this.

dlarue, 2022-01-24:
On one tested system sound still works on restart from hibernate.
One guy said the fix did not help him but I've not validated what he really tested and he can't work with me until next week.
I've put out a script to a group with dozens of these Vulcan tablets running Linux so hopefully I'll have more feedback later this week.

dlarue, 2022-01-26:
well crap, I did a ffresh clonezilla image of Kubuntu on another tablet, applied my sound patching script and sound played for some time but a few hours later I checked and it was screeching....

lgirdwood, 2022-01-27:
This sounds like it could be a race between DMA and the pipeline logic, we do have some optimisation in development for HiFi2 DSPs that can increase pipeline performance by 20 - 40%. I hope this could help here and reduce occurrence. @singalsu another user for the frag API update which should land in the next few weeks. @dlarue would you be able to help test this update when ready ?

dlarue, 2022-01-27:
@lgirdwood yes, I can test on these tablets.

dlarue, 2022-03-16:
Any movement on this yet?

plbossart, 2022-03-16:
no progress, we haven't had any other reports of issues with the work-around @dlarue

dlarue, 2022-03-16:
@plbossart   I wonder what I'm doing wrong then since I still get the buzzing sound after a bit of working audio? I will go back and see if I included all which was required for the work-around but I'm usually good about documenting my system changes into scripts. Script posted below. 

maker@PB-Robotics-Club:~$ cat /proc/version
Linux version 5.13.0-35-generic (buildd@ubuntu) (gcc (Ubuntu 11.2.0-7ubuntu2) 11.2.0, GNU ld (GNU Binutils for Ubuntu) 2.37) #40-Ubuntu SMP Mon Mar 7 08:03:10 UTC 2022

#!/bin/bash

sudo apt install -y git
HERE=`pwd`
cd
git clone https://github.com/thesofproject/sof-bin
mkdir -p ~/GetSoundWorking/backup && sudo mv /lib/firmware/intel/sof* ~/GetSoundWorking/backup
cd sof-bin/
sudo ./install.sh v2.0.x/v2.0
sync
cd
rm -rf sof-bin

grep "snd-intel-dspcfg" /etc/modprobe.d/alsa-base.conf > /dev/null 2>&1
if [ $? != 0 ] ; then
    echo "options snd-intel-dspcfg dsp_driver=3" | sudo tee -a /etc/modprobe.d/alsa-base.conf
fi
grep "sof_debug=1" /etc/modprobe.d/alsa-base.conf > /dev/null 2>&1
if [ $? != 0 ] ; then
    echo "option snd_sof sof_debug=1" | sudo tee -a /etc/modprobe.d/alsa-base.conf
fi
tail /etc/modprobe.d/alsa-base.conf
echo "---------------------------------------------------------------"
echo "you should see the last 2 lines above showing sof_debug=1' and dsp_driver=3'"
echo "reboot for changes to take effect"
cd $HERE

mengdonglin, 2022-06-21:
Move BYT and CHT platform to maintenance mode and SOF v2.2 will be the last upstream release with support for this platform. There will be no more bug fixing effort on this platform.

closed 2022-06-21 by mengdonglin