thesofproject/sof #5248: [BUG]

insiderser, 2022-01-23:
**Describe the bug**
Only the left microphone is used for audio capture on Dell XPS 17 9700.

When I physically touch the left microphone, the captured audio spikes with a lot of noise (as expected). But when I touch the right microphone, the captured audio is the same, e.g. there are no noise spikes (not expected).

If I mute the left microphone completely in alsamixer (leaving only the right one active), it still uses the left microphone only, albeit at half the volume.

**Reproduction Rate**
All the time.

**Expected behavior**
Both left and right microphones are used for capture.

**Impact**
- Worse raw recorded audio quality (?).
- Cannot use multiple microphones for better noise suppression.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.16.2-arch1-1
    * SOF: 2.0-1
2) Name of the topology file
    * Topology: What's this? This is not explained at https://thesofproject.github.io/latest/contribute/process/bug-tracking.html
3) Name of the platform(s) on which the bug is observed.
    * Platform: Dell XPS 17 9700

**Screenshots or console output**
[amixer.txt](https://github.com/thesofproject/sof/files/7919045/amixer.txt)
[dmesg.txt](https://github.com/thesofproject/sof/files/7919535/dmesg.txt)

What is ldc file, and how do I generate it? This is not explained at https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

lgirdwood, 2022-01-24:
@singalsu @plbossart  any ideas ? It looks like the Mics are being mixed into the left channel. I suspect this is happening in the codec (RT715) but amixer looks ok AFAICT

singalsu, 2022-01-24:
The dmesg.txt does not reveal which topology is used. It's unlikely that SOF causes this (a demux?) but to confirm adding this file to module options would make a more verbose kernel log. Example from my device:

```
$ cat /etc/modprobe.d/sof-dyndbg.conf 
# Kernel debug prints from SOF
options snd_sof dyndbg=+p
options snd_sof_pci dyndbg=+p

```

plbossart, 2022-01-24:
We don't play with microphone selection in the SOF firmware, we get two channels over SoundWire from the RT715 codec, so it's likely a bad setup of alsa mixers.

@insiderser do you use UCM for setting up the mixers? Can you provide the values used for rt715 controls?

Adding Realtek team: @jack-cy-yu  @shumingfan in case they can help with mixer setups.

insiderser, 2022-01-24:
@plbossart Yes, I use alsa-ucm. Should I open an issue with them instead?

@singalsu Topology is intel/sof-tplg/sof-cml-rt711-rt1308-rt715.tplg [[dmesg.txt](https://github.com/thesofproject/sof/files/7926403/dmesg.txt)]

plbossart, 2022-01-24:
@insiderser can you attach the results of 'alsa-info', so that we can double check which SKU this device is and check if we have one available for tests.

singalsu, 2022-01-24:
The topology is currently like this in git main. There's two capture PCMs (PCM1C and PCM4C) and there's no mux component in them. PGA and EQIIR do not have channel data duplication capability.

[image]

plbossart, 2022-01-24:
the two capture paths are for headset mic (RT711) and local mics (RT715). They are completely independent can be used concurrently.

insiderser, 2022-01-24:
@plbossart Sure.
[alsa-info.txt](https://github.com/thesofproject/sof/files/7928234/alsa-info.txt)

mengdonglin, 2022-01-25:
@shumingfan  Can you help check ALC715 mixer settings in the attached alsa-info?

jack-cy-yu, 2022-01-25:
@mengdonglin The mixer setting looks good to me. Is the same setting works on other platform?

jack-cy-yu, 2022-01-25:
@insiderser Could you help to dump the register during recording?

insiderser, 2022-01-25:
@jack-cy-yu I can try, but please send me the steps to take to do this, as I'm not familiar with how to debug SOF.

jack-cy-yu, 2022-01-25:
> @jack-cy-yu I can try, but please send me the steps to take to do this, as I'm not familiar with how to debug SOF.

During recording, please go to "/sys/kernel/debug/regmap/<sdw addr>/ ", then you can do "cat registers" to dump register.

jack-cy-yu, 2022-01-25:
> @shumingfan Can you help check ALC715 mixer settings in the attached alsa-i

insiderser, 2022-01-26:
> > @jack-cy-yu I can try, but please send me the steps to take to do this, as I'm not familiar with how to debug SOF.
> 
> During recording, please go to "/sys/kernel/debug/regmap// ", then you can do "cat registers" to dump register.

Thanks. Here are the dumps:

[sdw:0:025d:0711:00-sdw.txt](https://github.com/thesofproject/sof/files/7945375/sdw.0.025d.0711.00-sdw.txt)
[sdw:0:025d:0711:00.txt](https://github.com/thesofproject/sof/files/7945376/sdw.0.025d.0711.00.txt)
[sdw:1:025d:1308:00.txt](https://github.com/thesofproject/sof/files/7945377/sdw.1.025d.1308.00.txt)
[sdw:2:025d:1308:00.txt](https://github.com/thesofproject/sof/files/7945378/sdw.2.025d.1308.00.txt)
[sdw:3:025d:0715:00-sdw.txt](https://github.com/thesofproject/sof/files/7945379/sdw.3.025d.0715.00-sdw.txt)
[sdw:3:025d:0715:00.txt](https://github.com/thesofproject/sof/files/7945380/sdw.3.025d.0715.00.txt)

Also, this issue doesn't happen on Windows 11 running on that same laptop, e.g. both microphones are used on Windows.

jack-cy-yu, 2022-01-27:
@insiderser Could you help try attached patch? The purpose for this patch is to check whether DMIC is normal or not
[0001-ASoC-rt715-add-hidden-register-for-debug.zip](https://github.com/thesofproject/sof/files/7949245/0001-ASoC-rt715-add-hidden-register-for-debug.zip)
.

insiderser, 2022-02-01:
> @insiderser Could you help try attached patch? The purpose for this patch is to check whether DMIC is normal or not [0001-ASoC-rt715-add-hidden-register-for-debug.zip](https://github.com/thesofproject/sof/files/7949245/0001-ASoC-rt715-add-hidden-register-for-debug.zip) .

Sorry for taking time. Here are the dumps:
[sdw:0:025d:0711:00.txt](https://github.com/thesofproject/sof/files/7982314/sdw.0.025d.0711.00.txt)
[sdw:1:025d:1308:00.txt](https://github.com/thesofproject/sof/files/7982315/sdw.1.025d.1308.00.txt)
[sdw:2:025d:1308:00.txt](https://github.com/thesofproject/sof/files/7982316/sdw.2.025d.1308.00.txt)
[sdw:3:025d:0715:00.txt](https://github.com/thesofproject/sof/files/7982318/sdw.3.025d.0715.00.txt)

Kernel version: 5.16.4

jack-cy-yu, 2022-02-03:
@insiderser Do you have recorded files for three different hidden register setting? We'll need to analyze the audio files based on different settings.

insiderser, 2022-02-03:
> @insiderser Do you have recorded files for three different hidden register setting? We'll need to analyze the audio files based on different settings.

Here's the recording I did while dumping the registers the last time. You can also hear me "touching" the microphones (the last 10 seconds).
[record.zip](https://github.com/thesofproject/sof/files/7997455/record.zip)

I don't understand what you mean under "three different hidden register setting." I'm not familiar with how to debug Linux kernel or its modules, or how SOF works.

jack-cy-yu, 2022-02-04:
@insiderser I mean to add the register setting in rt715_io_init in rt715.c . Please check the "0001-ASoC-rt715-add-hidden-register-for-debug.patch", there is a register setting specific for debug only,

+	/*  For debug only */
+	rt715_index_write(rt715->regmap, 0x1a, 0x0); // all rising latch
+

with this setting, all recording data is supposed to get from Left channel, and you can get the recording file.
After above steps, please replace the "rt715_index_write(rt715->regmap, 0x1a, 0x0);" in rt715_io_init to "rt715_index_write(rt715->regmap, 0x1a, 0xff00);", and please do the recording again, you'll get 2nd file. Check whether the recordrd data is silence or not?
In the last step, please replace the "rt715_index_write(rt715->regmap, 0x1a, 0xff00);" in rt715_io_init to "rt715_index_write(rt715->regmap, 0x1a, 5500);", this is the default setting, which we suppose to get normal result (recorded data from both L and R ch). With this setting, you can do the recording again and get 3rd file.

insiderser, 2022-02-07:
@jack-cy-yu Thanks for the explanation!

Only the left microphone was used in all cases (0x0, 0xff00, 5500). No recordings were silent.

jack-cy-yu, 2022-02-08:
@insiderser I add another debug control to check the Sdw dataport status (see the attached patch), please help to do one more experiment, thanks!
This control is added for dmic issue debug
1. Please use amixer cset to set "Debug Ctrl" to 1
   after dmic start to record, our AE team need to know the
   initial status of Sdw dataport.
2. (During recording) Please use amixer cset to set "Debug Ctrl" to 2
   to disable L channel of dataport.
3. (During recording) Please use amixer cset to set "Debug Ctrl" to 3
   to disable R channel of dataport.
4. (During recording) Please use amixer cset to set "Debug Ctrl" to 4
   to disable all channels of dataport.
5. (During recording) Please use amixer cset to set "Debug Ctrl" to 5
   to enable all channels.

Please follow above five steps to record, keep 3-5 seconds for each step and let us know the result. Thanks.

[0001-ASoC-rt715-add-debug-control-for-dmic-issue.zip](https://github.com/thesofproject/sof/files/8022487/0001-ASoC-rt715-add-debug-control-for-dmic-issue.zip)

@mengdonglin Can we borrow one device from Intel for internal verification?

insiderser, 2022-02-10:
@jack-cy-yu Still only the left microphone is used after doing any of steps 1-3 or 5. Step 4 mutes the recorded audio completely. To repeat, step 2 does not mute the left microphone.

[dmesg.txt](https://github.com/thesofproject/sof/files/8043552/dmesg.txt)

jack-cy-yu, 2022-02-11:
@insiderser I've got the device and did the same experiment, I got the same result, step 2 doesn't mute the left microphone, and that's the strange point because if the device ONLY uses left mic, audio should be muted after step 2. I can get voice even after step 2 or step 3, which means left and right mics are used. This experiment is suggested by our AE team, they'd like to know that recorded file is silent or not if the dataport is muted. 
@plbossart @singalsu Is there any process after SoC get audio from codec?

lgirdwood, 2022-02-11:
@insiderser can you confirm which capture PCM is being used for the test. In the topology diagram above we have 2 capture PCMs (one has EQ) but both should NOT mute one channel.

insiderser, 2022-02-11:
@lgirdwood I guess it's PCM4C. When I mute PGA5.0 in amixer, all microphones are muted as well.

lgirdwood, 2022-02-14:
@singalsu is there anyway that the EQ on pipeline 5 above could mute only 1 channel ? I'm assuming that the same processing is done on both channels ?

insiderser, 2022-02-14:
@lgirdwood For context, when I completely mute the left channel of PGA5.0 via amixer (e.g. leaving only the right channel to be not muted), still only the left microphone is used.

lgirdwood, 2022-02-14:
@insiderser thanks for info, fwiw I dont think it's likely for EQ to mute/copy channels or for volume to do this but @singalsu can confirm.

insiderser, 2022-03-31:
This issue still exists with Linux kernel v5.17

lgirdwood, 2022-04-06:
> This issue still exists with Linux kernel v5.17

@insiderser thanks for letting us know, I dont think this is SOF FW or topology related as the same capture pipelines are used on other devices and work as expected for L/R channels.
@plbossart @bardliao do you know if this soundwire HW topology is used on any other platforms, could we be missing a workaround for this device that's in Windows driver for soundwire (e.g. override of default SDW slot/channel mappings - just guessing) ?

bardliao, 2022-04-06:
@lgirdwood As I know, Realtek is working on it. @jack-cy-yu Do you have any update?

jack-cy-yu, 2022-04-07:
> @insiderser I've got the device and did the same experiment, I got the same result, step 2 doesn't mute the left microphone, and that's the strange point because if the device ONLY uses left mic, [...]

@bardliao from the experiment on 11 Feb, I can get voice even after step 2 or step 3, which means left and right mics are used. The debug kcontrol tries to mute L ch and R ch separately to see if the audio really muted after that, but the result shows we can still get voice even after L ch is muted or R ch is muted.

Attached file is the amixer script and the file I recorded today, I touched and rubbed left mic and right mic (in the file, I touched left side when I said 1 2 3, I touched right side when I said 4 5 6), it seems the voice can be captured normally.
[2022-04-07.zip](https://github.com/thesofproject/sof/files/8439144/2022-04-07.zip)
[rt715_mixer.zip](https://github.com/thesofproject/sof/files/8439145/rt715_mixer.zip)

bardliao, 2022-04-07:
@jack-cy-yu I think there was some misunderstanding. What @insiderser reported is that the right mic captured nothing. but your  previous experiment was to check left and right channel. As far as I know, both left and right have left and right channels. In other words, the left and right channels of @insiderser's recording could be both from the left mic.
However, according to @jack-cy-yu's last experiment both left and right mics works on his Dell XPS 17 9700. I start to suspect that could it be a HW issue on @insiderser's laptop?
@insiderser Do you have a chance to install Windows on your Dell XPS 17 9700 and test it again?

insiderser, 2022-04-12:
Thanks everyone for looking into this issue.

I did some more investigation with Windows, and it looks like this is a hardware issue after all.

What misled me was that when I open the Windows' volume mixer (which shows the amplitude of microphone data), I can strangely see big amplitude spikes when I touch the right microphone. But when I record a video using the Camera app, the video doesn't have any audio spikes when the right microphone is touched.

I'm going to close this issue. Sorry for taking that much of your time. I have certainly learned a lot from you all.

closed 2022-04-12 by insiderser