thesofproject/sof #5257: [BUG][APL] pause-resume tests failing on UP2 with pcm512x HiFiBerry cards / topology

lyakh, 2022-01-25:
**Describe the bug**
The pcm512x topology hasn't been tested by CI for some time. Having re-added it to daily tests, pause-resume tests showed a regression. It is supposedly related to that topology's non-traditional mixer placement in the pipeline and to recent changes in trigger processing.

**To Reproduce**
See e.g. https://sof-ci.sh.intel.com/#/result/planresultdetail/9641?model=APL_UP2_PCM512X&testcase=check-pause-resume-playback-100

**Reproduction Rate**
Seems easy enough to reproduce. E.g. at the above link it happened on 18th  / 19th iteration

**Expected behavior**
pipelines with embedded forking components like mixers should work too.

**Impact**
This is also relevant for QuickBuild CI tests

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev
    * SOF: main
2) Name of the topology file
    * Topology: sof-apl-pcm512x.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: UP2

XiaoyunWu6666, 2022-01-28:
happened in inner daily 9800
model APL_UP2_PCM512X
testcase check-pause-resume-playback-100

With IO error in console

[etrace]
```
[      395074.619718] (           0.000000) c0 dma-trace              src/drivers/dw/dma.c:1128 ERROR dw_dma_get_data_size(): xrun detected
[          10.416666] (          10.416666) c0 dma-trace                   src/audio/dai.c:881  ERROR dai_report_xrun(): underrun due to no data available
[          17.812499] (           7.395833) c0 dma-trace                   src/audio/dai.c:748  ERROR comp_underrun(): dev->comp.id = 8, source->avail = 192, copy_bytes = 0
[          45.312498] (          27.499998) c0 dma-trace          ....../pipeline-stream.c:134  ERROR pipeline_copy(): ret = -61, start->comp.id = 8, dir = 1
[          53.072915] (           7.760417) c0 dma-trace          ..../pipeline-schedule.c:183  ERROR pipeline_task(): xrun recovery failed! pipeline is stopped.
```

lgirdwood, 2022-01-31:
@lyakh @XiaoyunWu6666 Does this happen on Zephyr ?

marc-hb, 2022-02-05:
Still observed with today's daily run 9986. P1 because it's affecting daily test results. If it's actually low severity explain why and I will SKIP this test until the bug is fixed.

Kernel Branch: topic/sof-dev
Kernel Commit: https://github.com/thesofproject/linux/commit/03175849
SOF Branch: main
SOF Commit: cf04ada3236b

lyakh, 2022-02-22:
fixed by #5374

closed 2022-02-22 by lyakh