thesofproject/sof #5273: [BUG] Remove HiFi code duplication in HiFi buffer management

cujomalainey, 2022-01-26:
Tracking bug for https://github.com/thesofproject/sof/pull/5169#discussion_r790016690

cujomalainey, 2022-01-26:
Turns out github links don't point to resolved comments properly. This bug is track the deduplication of *_setup_circular(source) functions in HiFi sections.

lgirdwood, 2022-01-27:
@cujomalainey fyi @singalsu is creating a spreadsheet for all the SIMD optimizations/cleanups. Will be able to share it soon.

singalsu, 2022-02-01:
I've noticed that audio_stream_bytes_without_wrap() performs better in many cases than using same circular start-end registers for several circular buffers in the hot loop.  I will test the most important components with such change and change code if my test devices show processing time improvement.

lgirdwood, 2022-02-16:
50% complete, should be fully complete for v2.2

kv2019i, 2023-02-17:
@singalsu @andrula-song Can you update on status? I put milestone tentatively to v2.6, but we can pull in to v2.5 if we are ready.

cujomalainey, 2023-02-21:
@singalsu any chance you can share that doc?

andrula-song, 2023-02-22:
I checked our code, there are still two components to can remove the duplication of the circular buffer setting, one is eq_fir and another is tdfb.

singalsu, 2023-03-03:
@kv2019i @cujomalainey Please keep 2.6 as target even if this would be a small change. We have other bugs fixing going on, better to not risk those for 2.5.

andrula-song, 2023-05-11:
PR https://github.com/thesofproject/sof/pull/7499 is under review.

alex-cri, 2023-05-25:
@andrula-song, @singalsu, with PR #7499  merged can we close this?

andrula-song, 2023-05-26:
I have test several method to remove that duplication in TDFB HiFi part, but unfortunately, I found there is only about 0.2MCPS improvement but it needs about 19+ MCPS, So I did not create a PR for TDFB since such improvement seems meaningless. So I think we can close this issue now.

cujomalainey, 2023-05-26:
Can you share the data in a draft PR for us to see the diff, MCPS isn't the only metric of importance here

andrula-song, 2023-05-29:
here is the draft https://github.com/thesofproject/sof/pull/7700

alex-cri, 2023-05-30:
@cujomalainey, is there further feedback that you feel is relevant for this issue?

commit 7d98a0b5 (2023-06-27): Audio: TDFB: Optimize the circular buffer usage

Optimize the tdfb_fir_sx functions by remove the duplicated
usage of circular buffer.

Fixes: #5273.
closed 2023-06-27 by kv2019i