thesofproject/sof #5281: [BUG] ipc4 set pipeline 3 status error on TGLU_RVP_NOCODEC_IPC4

XiaoyunWu6666, 2022-01-28:
**Describe the bug**
ipc error happened when setting status for pipeline 3 when aplay or arecord with PCM: Port1 [hw:0,1]
on TGLU_RVP_NOCODEC_IPC4

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/cavs-tgl-nocodec.tplg MODEL=TGLU_RVP_NOCODEC_IPC4 ~/sof-test/test-case/check-capture.sh -d 1 -l 100 -r 1

**Reproduction Rate**
almost 100%

**Environment**
    * Kernel: inner source linux topic/sof-cavs-dev f81ca4668a
    * SOF: main branch plus PR 5231 5280
    * Platform: TGLU_RVP_NOCODEC_IPC4
    * TOPOLOGY: cavs-tgl-nocodec.tplg
[image]

**Logs**
[dmesg]
```
[  404.995577] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 bind module 4 -0 to module 4 -1
[  404.995714] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: route copier.SSP.2.1 -> copier.host.2.1 setup complete
[  404.995719] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: stream dir 1, posn mailbox offset is 0
[  404.996004] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 1
[  404.996029] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x14001e successful
[  404.996033] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 2 status 3
[  404.996648] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 2 status 4
[  405.499325] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 0
[  405.499337] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 2 status 3
[  405.502458] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: Unknown error while processing the request
[  405.502536] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc error for msg 0x13020003 : 0x0
[  405.502551] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to set pipeline 2 status 3
[  405.502558] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to set pipeline 2 to state 3
[  405.502578] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x20140000 successful
[  405.502600] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -5
[  405.502608] kernel:  Port0: ASoC: trigger FE cmd: 0 failed: -5
[  405.502656] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: free stream 0 dir 1
[  405.502708] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 unbind module 4 -0 to module 4 -1
[  405.502892] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: route copier.SSP.2.1 -> copier.host.2.1 unbind complete
[  405.502903] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget copier.host.2.1 instance 1 freed
[  405.502910] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget copier.SSP.2.1 instance 0 freed
[  405.502913] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 delete pipeline 2
[  405.503161] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget pipeline.2 instance 0 freed
[  405.503256] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: close stream 0 dir 1reenshots or console output**
```
[sof-logger]
```

 TIMESTAMP         (us)              DELTA  C# COMPONENT          LOCATION                      CONTENT ktime=3213.741s  @  2022-01-28 06:10:40 +0000 UTC
[      192924.940251] (           0.000000) c0 dma-trace          /drivers/intel/ssp/ssp.c:807  ERROR ssp_set_config(), ssioc = 0x00000020, ssrsa = 0x00000003, sstsa = 0x00000003
[     2058558.459867] (     2058558.500000) c0 dma-trace             src/audio/component.c:102  ERROR comp_set_state(): wrong state = 4, COMP_TRIGGER_RESET
[     2058671.220279] (         112.760414) c0 dma-trace             src/audio/component.c:102  ERROR comp_set_state(): wrong state = 4, COMP_TRIGGER_RESET
[     2058692.001528] (          20.781250) c0 dma-trace             src/audio/component.c:102  ERROR comp_set_state(): wrong state = 4, COMP_TRIGGER_RESET
[     2059432.157749] (         740.156250) c0 dma-trace            src/ipc/ipc4/handler.c:873  ERROR ipc4: 1 failed err 7
[     4150379.835079] (     2090947.625000) c0 dma-trace             src/audio/component.c:102  ERROR comp_set_state(): wrong state = 4, COMP_TRIGGER_RESET
[     4150491.970491] (         112.135414) c0 dma-trace             src/audio/component.c:102  ERROR comp_set_state(): wrong state = 4, COMP_TRIGGER_RESET
[     4150512.178823] (          20.208332) c0 dma-trace             src/audio/component.c:102  ERROR comp_set_state(): wrong state = 4, COMP_TRIGGER_RESET
[     4151236.814211] (         724.635376) c0 dma-trace            src/ipc/ipc4/handler.c:873  ERROR ipc4: 1 failed err 7
[     4175812.334068] (       24575.519531) c0 dma-trace              src/drivers/dw/dma.c:285  ERROR dw_dma_start(): dma 0 channel 0 not ready ena 0x1 status 0x3
[     4175834.104901] (          21.770832) c0 dma-trace          ....../pipeline-stream.c:433  ERROR pipeline_trigger_run(): ret = -16, host->comp.id = 262145, cmd = 7
[     4175851.396567] (          17.291666) c0 dma-trace          ..../pipeline-schedule.c:64   ERROR pipeline_task_cmd(): failed to trigger components: -16
[     5184515.262736] (     1008663.875000) c0 dma-trace            src/ipc/ipc4/handler.c:357  ERROR failed to wait schedule thread
[     5185441.460616] (         926.197876) c0 dma-trace            src/ipc/ipc4/handler.c:873  ERROR ipc4: 1 failed err 7
[     5185759.064770] (         317.604156) c0 dma-trace             src/ipc/ipc4/helper.c:189  ERROR ipc_pipeline_free(): module free () failed
```

RanderWang, 2022-02-10:
I will check it

RanderWang, 2022-02-14:
I can't reproduce it on old ipc4 kernel branch, so it should be caused by kernel

ujfalusi, 2022-02-14:
@RanderWang, what is the 'old ipc4 kernel'? Can you reproduce with 'new ipc4 kernel' (which one is the new?)?
Can it be reproduced with the really new `integration/ipc4-wip` branch?
I can not make the `cavs-tgl-nocodec.tplg` work on my up-xtreme i11 board to test.

bardliao, 2022-02-14:
I would suggest that we focus on the HDA test for now. As I know, currently, https://github.com/thesofproject/linux/tree/integration/ipc4-wip supports HDA only.

lgirdwood, 2022-02-15:
Pushing to v2.2 since it's a WIP.

XiaoyunWu6666, 2022-03-14:
update the issue status:
In early March we update the IPC4 test according to the validation plan . Now we no longer run sof+IPC4 on TGLU RVP NOCODEC.
Instead , we run sof zephyr +IPC4 on adl nocodec.

However , since we have blocker issue that  the UUIDs of BRNGUP and BASEFW are incompatible with cAVS 1.8+ platforms.
potential fix :https://github.com/zephyrproject-rtos/zephyr/pull/43571
 so that this issue cannot be observed

XiaoyunWu6666, 2022-03-18:
newest test result with rander's sof PR #5497 and another sof PR for adding tgl IPC4 config to platforms/tgl/ipc4_overlay.conf #5566 

**Environment**
Kernel Branch: integration/ipc4-wip
Kernel Commit: [c95392aa](https://github.com/thesofproject/linux/commit/c95392aa)
SOF Branch: main + PR #5497  + PR #5566 
[dmesg]
```
[12427.698408] sof-audio-pci-intel-tgl 0000:00:1f.3: widget copier.HDA.6.1 setup complete
[12427.698419] sof-audio-pci-intel-tgl 0000:00:1f.3: sof_ipc4_route_setup: bind copier.host.5.1 -> copier.HDA.6.1
[12427.698429] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x45000004|0x10004: MOD_BIND
[12427.698562] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x65000000|0x10004: MOD_BIND
[12427.698586] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x45000004|0x10004: MOD_BIND
[12427.708350] sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 3 dir 0 cmd 1
[12427.708377] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x14001e successful
[12427.708386] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 5 state 3
[12427.708396] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13050003|0x0: GLB_SET_PIPELINE_STATE
[12427.708536] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000000|0x0: GLB_SET_PIPELINE_STATE
[12427.708566] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x13050003|0x0: GLB_SET_PIPELINE_STATE
[12427.708576] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 5 state 4
[12427.708584] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13050004|0x0: GLB_SET_PIPELINE_STATE
[12427.708869] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000000|0x0: GLB_SET_PIPELINE_STATE
[12427.709874] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x13050004|0x0: GLB_SET_PIPELINE_STATE
[12427.709885] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 6 state 3
[12427.709895] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13060003|0x0: GLB_SET_PIPELINE_STATE
[12428.215780] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13060003|0x0
[12428.215804] sof-audio-pci-intel-tgl 0000:00:1f.3: failed to pause pipeline 6
[12428.215817] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[12428.215831]  HDMI1: ASoC: trigger FE cmd: 1 failed: -110
[12428.215850] sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 3 dir 0 cmd 0
[12428.215860] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 5 state 3
[12428.215872] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13050003|0x0: GLB_SET_PIPELINE_STATE
[12428.719799] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13050003|0x0
[12428.719821] sof-audio-pci-intel-tgl 0000:00:1f.3: failed to pause pipeline 5
[12428.719842] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x20140000 successful
[12428.719861] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[12428.719875]  HDMI1: ASoC: trigger FE cmd: 0 failed: -110
[12429.226127] sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: free stream 3 dir 0
[12429.226146] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 5 state 3
[12429.226158] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13050003|0x0: GLB_SET_PIPELINE_STATE
[12429.727855] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13050003|0x0
[12429.727877] sof-audio-pci-intel-tgl 0000:00:1f.3: failed to pause pipeline 5
[12429.727915] sof-audio-pci-intel-tgl 0000:00:1f.3: sof_ipc4_route_free: unbind modules copier.host.5.1 -> copier.HDA.6.1
[12429.727928] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x46000004|0x10004: MOD_UNBIND
[12430.231882] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x46000004|0x10004
[12430.231903] sof-audio-pci-intel-tgl 0000:00:1f.3: failed to unbind modules copier.host.5.1 instance 0 -> copier.HDA.6.1 instance 1
[12430.231918] sof-audio-pci-intel-tgl 0000:00:1f.3: widget copier.host.5.1 instance 0 freed
[12430.231934] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x12050000|0x0: GLB_DELETE_PIPELINE
[12430.735911] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x12050000|0x0
[12430.735933] sof-audio-pci-intel-tgl 0000:00:1f.3: failed to free pipeline widget pipeline.5
[12430.735948] sof-audio-pci-intel-tgl 0000:00:1f.3: widget copier.HDA.6.1 instance 1 freed
[12430.735961] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x12060000|0x0: GLB_DELETE_PIPELINE
[12431.239934] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x12060000|0x0
[12431.239957] sof-audio-pci-intel-tgl 0000:00:1f.3: failed to free pipeline widget pipeline.6
[12431.239971] sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to free connected widgets
[12431.240106] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_free on 0000:00:1f.3: -110
[12431.240549] sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: close stream 3 dir 0
[12433.738145] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0x9010f0f successful
[12433.738691] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f successful
[12433.738710] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1

```

lgirdwood, 2022-03-18:
@RanderWang @juimonen any ideas ?

RanderWang, 2022-03-21:
I will check it.

XiaoyunWu6666, 2022-03-28:
As the newest daily IPC4 testresult shows, this can be reproduced 100% by 
```
TPLG=/lib/firmware/intel/avs-tplg/cavs-mixin-mixout-hda.tplg MODEL=TGLU_UP_HDA_IPC4 ~/sof-test/test-case/check-playback.sh -d 3 -l 1 -r 1 -F
```

XiaoyunWu6666, 2022-04-07:
close and updated it to #5661 since the testplan configuration has been changed

closed 2022-04-07 by XiaoyunWu6666