thesofproject/sof #5293: Unable to enable Zephyr+IPC4: tplg2 component load failed

XiaoyunWu6666, 2022-01-31:
**Describe the bug**
Unable to load topology on TGLU_UP_HDA_IPC4  which is used to enable zephyr+ipc4 in CI.

To enable IPC4 ,intel_adsp_cavs25.conf is modified as stated in Environment section
Not sure whether this is merely a topology problem or a kernel side problem . Any advice , thanks!

**Environment**
* Kernel: inner source linux topic/sof-cavs-dev f81ca4668a
* SOF: main branch plus PR 5231 , and add below contents to the  
 zephyrproject/zephyr/samples/subsys/audio/sof/boards/intel_adsp_cavs25.conf
```
CONFIG_RIMAGE_SIGNING_SCHEMA="tgl-cavs"
CONFIG_IPC_MAJOR_4=y
```

* Platform: TGLU_UP_HDA_IPC4  which is used to enable zephyr+ipc4 in CI
* TOPOLOGY: cavs-passthrough-hda    

**Screenshots or console output**
[dmesg]

```
[ 2654.274245] sof-audio-pci-intel-tgl 0000:00:1f.3: firmware boot complete
[ 2654.274247] sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state change: 5 -> 6
[ 2654.274305] sof-audio-pci-intel-tgl 0000:00:1f.3: snd_sof_init_trace: dma_trace_pages: 16
[ 2654.274324] sof-audio-pci-intel-tgl 0000:00:1f.3: bard: ipc4 doesn't support trace_ipc
[ 2654.274473] sof-audio-pci-intel-tgl 0000:00:1f.3: created machine skl_hda_dsp_generic
[ 2654.274642] sof-audio-pci-intel-tgl 0000:00:1f.3: Completing SOF PCI probe
[ 2654.294193] skl_hda_dsp_generic skl_hda_dsp_generic: skl_hda_audio_probe: entry
[ 2654.294270] skl_hda_dsp_generic skl_hda_dsp_generic: skl_hda_add_dai_link: dai link name - iDisp1
[ 2654.294316] skl_hda_dsp_generic skl_hda_dsp_generic: skl_hda_add_dai_link: dai link name - iDisp2
[ 2654.294352] skl_hda_dsp_generic skl_hda_dsp_generic: skl_hda_add_dai_link: dai link name - iDisp3
[ 2654.294386] skl_hda_dsp_generic skl_hda_dsp_generic: skl_hda_add_dai_link: dai link name - Analog Playback and Capture
[ 2654.294419] skl_hda_dsp_generic skl_hda_dsp_generic: skl_hda_add_dai_link: dai link name - Digital Playback and Capture
[ 2654.294456] skl_hda_dsp_generic skl_hda_dsp_generic: skl_hda_add_dai_link: dai link name - dmic01
[ 2654.296524] skl_hda_dsp_generic skl_hda_dsp_generic: skl_hda_add_dai_link: dai link name - dmic16k
[ 2654.296739] sof-audio-pci-intel-tgl 0000:00:1f.3: loading topology2:intel/sof-tplg/sof-hda-generic.tplg
[ 2654.296820] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:20:1 Kernel ABI 3:18:0
[ 2654.296829] sof-audio-pci-intel-tgl 0000:00:1f.3: tplg2: ready widget pipe 0  comp 0 type 7 name : iDisp3 Tx stream none
[ 2654.296833] sof-audio-pci-intel-tgl 0000:00:1f.3: widget type 7 name iDisp3 Tx not handled
[ 2654.296839] sof-audio-pci-intel-tgl 0000:00:1f.3: tplg2: ready widget pipe 1  comp 1 type 27 name : copier.HDA.1.0 stream Analog Playback and Capture
[ 2654.296843] sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to find module ID for widget copier.HDA.1.0 with uuid 9BA00C83-CA12-4A83-943C-1FA2E82F9DDA
[ 2654.296850] sof-audio-pci-intel-tgl 0000:00:1f.3: error: DSP failed to add widget id 0 type 27 name : copier.HDA.1.0 stream Analog Playback and Capture
[ 2654.296855] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: failed to load widget copier.HDA.1.0
[ 2654.296859] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: topology: could not load header: -12
[ 2654.296862] sof-audio-pci-intel-tgl 0000:00:1f.3: error: tplg2 component load failed -12
[ 2654.296871] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP topology -22
[ 2654.296875] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[ 2654.296910] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: failed to instantiate card -22
[ 2654.297307] skl_hda_dsp_generic: probe of skl_hda_dsp_generic failed with error -22

```

lgirdwood, 2022-01-31:
@lyakh can you take a look at this, thanks !

lyakh, 2022-02-01:
@lgirdwood @mengdonglin @XiaoyunWu6666 sorry, I don't see a reference to the CI test result nor do I see firmware trace here. But the error is verry telling: -12 is `-ENOMEM`... I'm guessing some topology value got corrupted and there's a large number somewhere where memory size is expected.

lgirdwood, 2022-02-01:
Looks like this will need @RanderWang to fix when back from vacation.

RanderWang, 2022-02-08:
Last time I found that the uuid of copier was not built into FW manifest, so it can't be found. @XiaoyunWu6666 can you decode FW to check uuid ? thanks

XiaoyunWu6666, 2022-02-08:
hi @RanderWang , I decoded FW  and got info below.
However, I could not get any relevant hints in it ... 

```
sof git:(main) ✗ ./tools/sof_ri_info/sof_ri_info.py zephyrproject/build-sof-staging/sof/community/sof-tgl.ri

SOF Binary zephyrproject/build-sof-staging/sof/community/sof-tgl.ri size 0x662e0

  Extended Manifest ver 1.0.0 length 736

  CSE Manifest ver 0x102 checksum 0x0 partition name ADSP

    ADSP.man (CSS Manifest) type 0x4 file offset 0x33c hdr_len 900 ver 0x21000 date 2022/02/08
      Rsvd0 0x0
      Modulus size (dwords) 96
        6b 75 ed 58 20 08 85 95 ... 55 d1 7d c6 0d 79 12 a9 (Community 3k key)
      Exponent size (dwords) 1
        01 00 01 00
      Signature (file offset 0x540, length 0x180)
        45 6b d7 6a 33 29 1e c5 ... 5d 19 ca 52 d4 4f 6b 57

      Plat Fw Auth Extension type 0xf file offset 0x6c0 length 0x78
       name ADSP vcn 0x0 bitmap 00 00 00 00 08 00 00 00 00 00 00 00 00 00 00 00 svn 0x0

      Other Extension type 0x16 file offset 0x738 length 0x68

    cavs0015.met (ADSP Metadata File Extension) type 0x11 file offset 0x7a0 length 0x70
     ver 0x0 base offset 0x40c69510 limit offset 0x7cd98c12
      IMR type 0x3
      Attributes
        43 02 a3 58 4c 78 6f 8c 00 20 00 00 c0 7a 06 00

    cavs0015

  cavs0015 (ADSP Manifest) file offset 0x22e0 name ADSPFW build ver 0.0.0.0 feature mask 0xffff image flags 0x0
    HW buffers base address 0x0 length 0x0
    Load offset 0x30000

    BRNGUP    0d7b48cc-1ea9-470a-a8c1-533424528a17
      entry point 0xb0038000 type 0x21 ( loadable LL )
      cfg offset 0 count 0 affinity 0x3 instance max count 1 stack size 0x1
      .text   0xb0038000 file offset 0x8000 flags 0x1001f ( contents alloc load readonly code type=0 pages=1 )
      .rodata 0xffffffff file offset 0x9000 flags 0x1012f ( contents alloc load readonly data type=1 pages=1 )
      .bss    0x0 file offset 0x0 flags 0xf00 ( type=15 pages=0 )

    BASEFW    fc869e2e-45f8-4045-a416-89880ae320a9
      entry point 0xbe010400 type 0x21 ( loadable LL )
      cfg offset 0 count 0 affinity 0x3 instance max count 1 stack size 0x1
      .text   0xbe010000 file offset 0x9000 flags 0x2c001f ( contents alloc load readonly code type=0 pages=44 )
      .rodata 0xbe03c000 file offset 0x35000 flags 0x31012f ( contents alloc load readonly data type=1 pages=49 )
      .bss    0xbe06d000 file offset 0x0 flags 0x20202 ( alloc type=2 pages=2 )

Intel Tigerlake-LP
  imr                                 0xb0000000 (4096 + 16773120  0.02% used)
    BRNGUP.text                         0xb0038000 (4096)
  l2 hpsram                           0xbe000000 (389120 + 2625536  12.91% used)
    BASEFW.text                         0xbe010000 (180224)
    BASEFW.rodata                       0xbe03c000 (200704)
    BASEFW.bss                          0xbe06d000 (8192)
  l2 lpsram                           0xbe800000 (65536)

```

RanderWang, 2022-02-09:
thanks. according to your info,   there are only two modules and copier is missed. 
please check my info.  These info are added by rimage with config of tgl-cavs
````
    BRNGUP    2b79e4f3-4675-f649-89df-3bc194a91aeb
      entry point 0xb0038000 type 0x21 ( loadable LL )
      cfg offset 0 count 0 affinity 0x3 instance max count 1 stack size 0x1
      .text   0xb0038000 file offset 0x8000 flags 0x1001f ( contents alloc load readonly code type=0 pages=1 )
      .rodata 0xb0039000 file offset 0x9000 flags 0x1012f ( contents alloc load readonly data type=1 pages=1 )
      .bss    0x0 file offset 0x0 flags 0xf00 ( type=15 pages=0 )

    BASEFW    0e398c32-5ade-ba4b-93b1-c50432280ee4
      entry point 0xbe02c400 type 0x21 ( loadable LL )
      cfg offset 0 count 0 affinity 0x3 instance max count 1 stack size 0x1
      .text   0xbe02c000 file offset 0xa000 flags 0x1f001f ( contents alloc load readonly code type=0 pages=31 )
      .rodata 0xbe04b000 file offset 0x29000 flags 0x2d012f ( contents alloc load readonly data type=1 pages=45 )
      .bss    0xbe078000 file offset 0x0 flags 0x2680202 ( alloc type=2 pages=616 )

    MIXIN     39656eb2-3b71-4049-8d3f-f92cd5c43c09
      entry point 0x0 type 0x20 ( builtin LL )
      cfg offset 0 count 10 affinity 0x1 instance max count 48 stack size 0x0
      .text   0x0 file offset 0x0 flags 0x0 ( type=0 pages=0 )
      .rodata 0x0 file offset 0x0 flags 0x0 ( type=0 pages=0 )
      .bss    0x0 file offset 0x0 flags 0x0 ( type=0 pages=0 )

    MIXOUT    3c56505a-24d7-418f-bddc-c1f5a3ac2ae0
      entry point 0x0 type 0x20 ( builtin LL )
      cfg offset 10 count 8 affinity 0x1 instance max count 48 stack size 0x0
      .text   0x0 file offset 0x0 flags 0x0 ( type=0 pages=0 )
      .rodata 0x0 file offset 0x0 flags 0x0 ( type=0 pages=0 )
      .bss    0x0 file offset 0x0 flags 0x0 ( type=0 pages=0 )

    COPIER    9ba00c83-ca12-4a83-943c-1fa2e82f9dda
      entry point 0x0 type 0x20 ( builtin LL )
      cfg offset 18 count 10 affinity 0x1 instance max count 50 stack size 0x0
      .text   0x0 file offset 0x0 flags 0x0 ( type=0 pages=0 )
      .rodata 0x0 file offset 0x0 flags 0x0 ( type=0 pages=0 )
      .bss    0x0 file offset 0x0 flags 0x0 ( type=0 pages=0 )
````

XiaoyunWu6666, 2022-02-09:
hi @rander , It seems that the firmware I built  was different from you , maybe there is a gap between ways we built.
 Below is mine :
Step 1 : fetch the newest sof code 
Step 2  : apply PR 5231 
Step 3 : zephyrproject/zephyr/samples/subsys/audio/sof/boards/intel_adsp_cavs25.conf
 Add below to 
```CONFIG_RIMAGE_SIGNING_SCHEMA="tgl-cavs"
CONFIG_IPC_MAJOR_4=y
```
Step 4 : build firmware with script : ./scripts/xtensa-build-zephyr.sh tgl

RanderWang, 2022-02-09:
I only built it for xtos, not zephyr.  The FW structure should be the same for both cases

XiaoyunWu6666, 2022-02-09:
@aiChaoSONG found some IPC4 files were not specified in sof/zephyr/CMakeLists.txt, so they are not built when building with zephyr. 
However after adding inclusion of src/audio/copier.c and src/audio/base_fw.c , the decode output of zephyr-build firmware still lacks copier component.
@lyakh @RanderWang  Are there any other dependencies for enabling copier when building zephyr firmware?
Many thanks~

XiaoyunWu6666, 2022-02-09:
Just synced with @aiChaoSONG , he found some mismatch in zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs25/board.cmake
```
board_set_rimage_target(tgl)

```
However , we need  board_set_rimage_target(tgl-cavs) instead to enable using tgl-cavs.toml when signing with tool rimage

lyakh, 2022-02-10:
> @lyakh @RanderWang Are there any other dependencies for enabling copier when building zephyr firmware?

@XiaoyunWu6666 yes, I think I know why: there are no references to the copier, so it isn't linked into the image. You need to add a prototype of and a call to `sys_comp_copier_init()` to sof/zephyr/wrapper.c similar to, e.g. `sys_comp_tone_init()`

lyakh, 2022-02-10:
> @XiaoyunWu6666 yes, I think I know why: there are no references to the copier, so it isn't linked into the image. You need to add a prototype of and a call to `sys_comp_copier_init()` to [...]

...and yes, I presume a call to `sys_comp_basefw_init()` has to be added too, possibly other modules as well, if you're adding those too

RanderWang, 2022-02-10:
> ...and yes, I presume a call to `sys_comp_basefw_init()` has to be added too, possibly other modules as well, if you're adding those too

@lyakh it is another issue. Currently rimage adds module UUID to extended fw manifest according to tgl-cavs.toml and driver will check it. Now the UUID is not added to FW extended manifest since it use tgl.toml

aiChaoSONG, 2022-02-11:
@lgirdwood For 1),  the patch is here: https://github.com/thesofproject/sof/pull/5344, For 2) Kai has reported an issue, https://github.com/thesofproject/sof/issues/5335, I I don't have a good solution to un-hardcode the rimage sign schema, or provide an override mechanism.

XiaoyunWu6666, 2022-02-28:
For two problems in this issue ,
1) no references to the copier and base_fw, already fixed by #5344 
2) wrong image signing schema used when building for IPC4 #5335 can be fixed by #5412  and zephyr https://github.com/zephyrproject-rtos/zephyr/pull/42993

This can be closed but let's wait for some days until zephyr provides fix for the xcc build so that I can make a double check

keqiaozhang, 2022-03-07:
This issue is fixed, closing this bug.

closed 2022-03-07 by keqiaozhang