thesofproject/sof #5294: [BUG] 19_01_TestAsrcCapture Exception SofIpcErrorResponseException occured -22 error code

wszypelt, 2022-01-31:
**Describe the bug**
Bug was found in:
19_01_TestAsrcCapture8000Hz16b16b2ch
19_01_TestAsrcCapture48000Hz32b32b2ch

the content of the error:
Exception SofIpcErrorResponseException occurred at 'C:\\work\\tools\\pyexecnetcache\\cavs_scripts-py\\utilities\\ipc_driver.py':763 FW replied with error. Error code: -22 0x00000005 0x00000000 0x00004000

Last working build: SOF_FW_MASTER_01.01.00.3043 hash: 7d05389b63bab99a9c6b4ac6755766749f9b4983

**Topology**
```

                +---------------------------------+
    +------+    | +---+  +----+  +---+  +-------+ |
    | Host |----->|Buf|->|Asrc|->|Buf|->|SSP Dai|---+
    +------+    | +---+  +----+  +---+  +-------+ | |
                +---------------------------------+ |
                                                    |
                +------------------+                |
    +------+    | +---+  +-------+ |                |
    | Host |<-----|Buf|<-|SSP Dai|<-----------------+
    +------+    | +---+  +-------+ |
                +------------------+

```
Description:
Simultaneous playback and capture on external loopback (same SSP) with signal verification.
Playback pipeline contains ASRC. In the test topology is built and destroyed several times to check different source
sampling rates.

**To Reproduce**
Run tests with diagnostic driver
19_01_TestAsrcCapture8000Hz16b16b2ch
19_01_TestAsrcCapture48000Hz32b32b2ch

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF BRANCH: main HASH: 0bf7b73dc8c65b1f3603d4c5455c51e493394012
2) Name of the platform(s) on which the bug is observed.
    * Platform: ICL RVP, WHL RVP, TGL RVP, APL RVP
3) Test environment:
BRANCH: master HASH: 352801b9abd9f30f8cc563cafe91723a55823ca2

**Screenshots or console output**
```
[     1682928.697917] (           9.322917) c0 ipc                      src/ipc/ipc3/dai.c:187  INFO ipc_comp_dai_config() dai type = 1 index = 0
[     1682939.817708] (          11.119792) c0 dai          1.1         src/ipc/ipc3/dai.c:281  INFO dai_config() dai type = 1 index = 0 dd 0x9e11e100
[     1682960.911458] (          21.093750) c0 dai          2.6         src/ipc/ipc3/dai.c:281  INFO dai_config() dai type = 1 index = 0 dd 0x9e11e400
[     1693345.859375] (       10384.948242) c0 ipc                  src/ipc/ipc3/handler.c:1573 INFO ipc: new cmd 0x60010000
[     1693359.375000] (          13.515625) c0 pipe         1.3   ....../pipeline-params.c:222  INFO pipe params dir 0 frame_fmt 0 buffer_fmt 0 rate 8000
[     1693368.567708] (           9.192708) c0 pipe         1.3   ....../pipeline-params.c:226  INFO pipe params stream_tag 1 channels 2 sample_valid_bytes 2 sample_container_bytes 2
[     1693417.291667] (          48.723957) c0 dai          1.1         src/ipc/ipc3/dai.c:103  INFO dai_data_config() dai type = 1 index = 0 dd 0x9e11e100
[     1693463.281250] (          45.989582) c0 dai          1.1            src/audio/dai.c:347  INFO dai_playback_params() dest_dev = 2 stream_id = 0 src_width = 2 dest_width = 2
[     1693473.229167] (           9.947917) c0 dai          1.1            src/audio/dai.c:353  INFO dai_playback_params() fifo 0x77010
[     1693486.979167] (          13.750000) c0 pipe         1.3   ....../pipeline-params.c:308  INFO pipe prepare
[     1693508.645833] (          21.666666) c0 dai          1.1            src/audio/dai.c:622  INFO dai_prepare()
[     1693518.489583] (           9.843750) c0 dai          1.1            src/audio/dai.c:587  INFO dai_config_prepare(), channel = 0
[     1693527.786458] (           9.296875) c0 dw-dma                 src/drivers/dw/dma.c:192  INFO dw_dma_channel_get(): dma 0 request channel 0
[     1693537.656250] (           9.869792) c0 dai          1.1            src/audio/dai.c:606  INFO dai_config_prepare(): new configured dma channel index 0
[     1706912.838542] (       13375.182617) c0 ipc                  src/ipc/ipc3/handler.c:1573 INFO ipc: new cmd 0x60010000
[     1706926.484375] (          13.645833) c0 pipe         2.9   ....../pipeline-params.c:222  INFO pipe params dir 1 frame_fmt 0 buffer_fmt 0 rate 8000
[     1706935.625000] (           9.140625) c0 pipe         2.9   ....../pipeline-params.c:226  INFO pipe params stream_tag 1 channels 2 sample_valid_bytes 2 sample_container_bytes 2
[     1706982.942708] (          47.317707) c0 asrc         2.4      src/audio/asrc/asrc.c:520  INFO asrc_params()
[     1706998.125000] (          15.182292) c0 asrc         2.4      src/audio/asrc/asrc.c:539  ERROR asrc_params(), zero sink rate
[     1707009.088542] (          10.963542) c0 pipe         2.9   ....../pipeline-params.c:242  ERROR pipeline_params(): ret = -22, host->comp.id = 5
[     1707019.817708] (          10.729167) c0 ipc                  src/ipc/ipc3/handler.c:283  ERROR ipc: pipe 2 comp 5 params failed -22
[     1707028.697917] (           8.880208) c0 pipe         2.9   ......./pipeline-graph.c:341  INFO pipe reset
[     1707072.656250] (          43.958332) c0 asrc         2.4      src/audio/asrc/asrc.c:1032 INFO asrc_reset()
[     1707081.588542] (           8.932292) c0 asrc         2.4      src/audio/asrc/asrc.c:1034 INFO asrc_reset(), skew_min=0, skew_max=0
[     1707091.770833] (          10.182292) c0 asrc         2.4      src/audio/component.c:62   INFO comp_set_state(), state already set to 1
[     1707102.447917] (          10.677083) c0 dai          2.6            src/audio/dai.c:671  INFO dai_reset()
```
**Logs**

[19_01_TestAsrcCapture8000Hz16b16b2ch.zip](https://github.com/thesofproject/sof/files/7970626/19_01_TestAsrcCapture8000Hz16b16b2ch.zip)

lgirdwood, 2022-01-31:
@singalsu the ASRC sink rate should be coming from the DAI here right ?

singalsu, 2022-02-01:
> @singalsu the ASRC sink rate should be coming from the DAI here right ?

Yes that's correct, the HW params get from SSP DAI should propagate it to pipeline.

abonislawski, 2022-02-08:
@softwarecki any update?

softwarecki, 2022-02-08:
I have it on my list, I will post a fix soon

closed 2022-02-15 by lgirdwood