thesofproject/sof #5308: [BUG] Custom IPC message data invalid when fetched from kernel

sanzazacuro, 2022-02-03:
**Describe the bug**
I want to send a custom ipc message from DSP to host. The ipc message size and data pointer is invalid when fetched from kernel. This is how its setup:

_on dsp:_
`
#define SOF_IPC_CUSTOM_MSG SOF_CMD_TYPE(0x123)

struct ipc_msg *ipc_msg;
int16_t test_data = 123;

ipc_msg = ipc_msg_init(SOF_IPC_CUSTOM_MSG, sizeof(testdata));

ipc_msg_send(ipc_msg, (void *)&test_data, true);
`

_in_kernel, ipc driver (ipc.c):_
`
callback in snd_sof_ipc_msgs_rx(struct snd_sof_dev *sdev) => ipc_trace_message(struct snd_sof_dev *sdev, u32 msg_id)

in switch case SOF_IPC_TRACE_DMA_POSITION:
after position has been fetched from DMA and the ipc_msg struct reads ipc_complete = true, the msg_size and msg_data is invalid.
`

Is there anything I have done wrong in order to fetch the message content correctly from the DSP?

**Reproduction Rate**
100% 

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.15
    * SOF: 5ef33e6310b7921cb2fdae4bf2ae798d26cb5438
2) Name of the topology file
    * Topology: sof-tgl-max98357a-rt5682.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: tigerlake

lgirdwood, 2022-02-04:
@sanzazacuro can you look at how host.c does this for unsolicited position update messages. I think you need to define a struct
```c
struct my_message {
    struct sof_ipc_reply rhdr; <<<< the reply type is used here.
    int test;
};
```

Best to grep host.c for "ipc_msg" as it has full flows for msg usage.

lgirdwood, 2022-02-09:
@sanzazacuro all good now - can we close ?

closed 2022-02-10 by sanzazacuro