thesofproject/sof #5309: [BUG] Sound not wokring (no sinks) on Matebook D15

TronFortyTwo, 2022-02-03:
**Describe the bug**
I have installed Arch on a Huawei MateBook D15, and I get no sound from the speaker. As far as I understand ALSA can't see my speakers. Alsa only sees the HDMI outputs but no analog output (more details below). I know it's not an hardware issue since it works on windows. Even with a jack headphone sound doesn't work. I can also reproduce in other distribution such as ubuntu.
I attach useful logs and info, hoping you may find some clue.

**To Reproduce**
No audio is ever played.

**Reproduction Rate**
All the time

**Expected behavior**
Audio is played

**Impact**
I am forced to use blueetooth speaker to hear audio from my laptop.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: Linux emanuele 5.16.4-zen1-1-zen #1 ZEN SMP PREEMPT Sat, 29 Jan 2022 19:08:11 +0000 x86_64 GNU/Linux
    * SOF: sof-firmware-2.0-1
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: Arch linux, ubuntu

**Screenshots or console output**

I proceed to add more datailed logs and outputs:

First of all to get pacmd list-cards to see my card, I added this to driver options:

```
$ cat /etc/modprobe.d/intel_dspfix.conf 
options snd_intel_dspcfg dsp_driver=1
This is lspci. As you can see I got a Multimedia audio controller, but I don't see any Audio device or speaker (not sure it should be anyway)
```

If I remove that option, my card is not seen at all:
```
$ pacmd list-cards // without modeprobe fix above                                                                                                
0 card(s) available.
```

lspci:

```
$ lspci -vv
00:00.0 Host bridge: Intel Corporation Device 9b71 (rev 0c)
        Subsystem: QUANTA Computer Inc Device 125d
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B+ ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Capabilities: <access denied>
        Kernel driver in use: skl_uncore

00:02.0 VGA compatible controller: Intel Corporation CometLake-U GT2 [UHD Graphics] (rev 02) (prog-if 00 [VGA controller])
        Subsystem: QUANTA Computer Inc Device 125d
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 139
        Region 0: Memory at b0000000 (64-bit, non-prefetchable) [size=16M]
        Region 2: Memory at a0000000 (64-bit, prefetchable) [size=256M]
        Region 4: I/O ports at 3000 [size=64]
        Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
        Capabilities: <access denied>
        Kernel driver in use: i915
        Kernel modules: i915

00:04.0 Signal processing controller: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem (rev 0c)
[... 329 lines cut ...]
Playback device is hw:0,3                                                                                                             
Stream parameters are 48000Hz, S16_LE, 2 channels                                                                                     
Using 16 octaves of pink noise                                                                                                        
Rate set to 48000Hz (requested 48000Hz)                                                                                               
Buffer size range from 64 to 1048576                                                                                                  
Period size range from 32 to 524288                                                                                                   
Using max buffer size 1048576                                                                                                         
Periods = 4                                                                                                                           
was set period_size = 262144                                                                                                          
was set buffer_size = 1048576                                                                                                         
 0 - Front Left                                                                                                                       
 1 - Front Right                                                                                                                      
Time per period = 10.951082                                                                                                           
 0 - Front Left                                                                                                                       
 1 - Front Right                                                                                                                      
Time per period = 10.950650                                                                                                           
 0 - Front Left                                                                                                                       
 1 - Front Right                                                                                                                      
Time per period = 10.936318                                                                                                           
 0 - Front Left                                                                                                                       
^C 1 - Front Right                                                                                                                    
Time per period = 2.777345
```

Lastly, the output of alsa-info.sh is uploaded at [http://alsa-project.org/db/?f=4c968c4fd … 0b0e592d92](http://alsa-project.org/db/?f=4c968c4fded32268bd70a8e1cc23e80b0e592d92)

eworm-de, 2022-02-03:
This is a duplicate of https://github.com/thesofproject/linux/issues/3248

plbossart, 2022-02-03:
yes, it's a duplicate. 

You need to remove this 
options snd_intel_dspcfg dsp_driver=1

Please try PR https://github.com/thesofproject/linux/pull/3338 

and install topology files mentioned in https://github.com/thesofproject/linux/pull/3338#issuecomment-1013246294

TronFortyTwo, 2022-02-19:
Thanks for your answers! If I get to test as you suggested I will report the results

plbossart, 2023-01-05:
no update in 9+ months, closing.

closed 2023-01-05 by plbossart