thesofproject/sof #5325: [BUG][TGL-U][audio Capture] DSP panic after long time recording test for about 19 hours

CBala21, 2022-02-08:
**Describe the bug**
A DSP panic error was reported after a long time recording test for about 19 hours. basically audio capture through I2S in slave mode configuration. 

[62976.411003] sof-audio-pci-intel-tgl 0000:00:1f.3: error : DSP panic!

[62976.411686] sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005

[62976.412138] sof-audio-pci-intel-tgl 0000:00:1f.3: error: can't enter idle

[62976.412610] sof-audio-pci-intel-tgl 0000:00:1f.3: error: trace point 00004000

[62976.413104] sof-audio-pci-intel-tgl 0000:00:1f.3: error: panic at src/lib/agent.c:64

[62976.413591] sof-audio-pci-intel-tgl 0000:00:1f.3: error: DSP Firmware Oops

[62976.414006] sof-audio-pci-intel-tgl 0000:00:1f.3: EXCCAUSE 0x0000003f EXCVADDR 0x00000000 PS       0x00060725 SAR     0x00000000

[62976.414556] sof-audio-pci-intel-tgl 0000:00:1f.3: EPC1     0x00000000 EPC2     0xbe02cfc6 EPC3     0x00000000 EPC4    0x00000000

[62976.415135] sof-audio-pci-intel-tgl 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000

[62976.415824] sof-audio-pci-intel-tgl 0000:00:1f.3: EPS2     0x00060d20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000

[62976.416486] sof-audio-pci-intel-tgl 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222

[62976.417002] sof-audio-pci-intel-tgl 0000:00:1f.3: stack dump from 0xbe0790f0

[62976.417706] sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe0790f0: be079120 00000000 be032b74 00000001

[62976.418586] sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe0790f4: 00000000 00000000 00000040 be084a00

[62976.419417] sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe0790f8: c62be900 a50b04ec 000c0800 00000000

[62976.420373] sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe0790fc: c0c49e15 ffffffff cda6a018 ffff8920

[62976.421062] sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe079100: c2838500 ffff8920 ce644400 ffff8920

[62976.421858] sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe079104: 00000000 00000000 00000000 00000000

[62976.422825] sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe079108: 8ab2c7b0 ffffffff c2838500 ffff8920

[62976.423867] sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe07910c: 00000000 00000000 00000000 0000000

**To Reproduce**
run the below test for longer duration.
arecord -vD hw:0,0 -r 48000 -c 2 -fS16_LE -t raw /dev/null

**Reproduction Rate**
Always during long run.

**Expected behavior**
No DSP panic during long duration audio recording 

**Impact**
TGL-U customer feature impacted and stability issue

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: Linux version 5.10.65-intel-ese-standard-lts
    * SOF: Firmware info: version 1:5:0-58b36
2) Name of the topology file
    * Topology: sof-tgl_rt1308.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL-U RVP 

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See [](https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html)
[TGL_DSP_panic.txt](https://github.com/thesofproject/sof/files/8024392/TGL_DSP_panic.txt)

lgirdwood, 2022-02-09:
@CBala21 can you update to SOF v2.0 as v1.5 is quite old. v2.0 has many fixes for TGL.

plbossart, 2022-12-12:
no update in 10 months and obsolete firmware -> closing

closed 2022-12-12 by plbossart