thesofproject/sof #5335: [BUG] [Zephyr] Wrong rimage signing schema used when building for IPC4

kv2019i, 2022-02-09:
**Describe the bug**
When SOF is built for IPC4, a different signing schema (e.g. "tgl-cavs") must be used. 

When building for Zephyr, signing schema is not change and resulting image is not correct. 

Zephyr signing tool (west zephyr/scripts/west_commands/sign.py) takes the target from RIMAGE_TARGET.

SOF scripts to build with IPC4 should specify an alternative target RIMAGE_TARGET when building for Zephyr&IPC4.

In XTOS build, matching logic is in src/arch/xtensa/CMakeLists.txt and set via config override:
sof/src/arch/xtensa/configs/override/tigerlake_ipc4.config

Original report from @XiaoyunWu6666 and @aiChaoSONG

lgirdwood, 2022-02-10:
@marc-hb IIUC, you have a few WIP PRs to support Zephyr. Do any cover this ?

aiChaoSONG, 2022-02-11:
I filed a patch on zephyr side to add sign schema overriding support https://github.com/zephyrproject-rtos/zephyr/pull/42724. @lgirdwood @kv2019i @lyakh not sure they will accept it.

lgirdwood, 2022-02-16:
Moving to v2.2 since this is related to IPC4

marc-hb, 2022-02-18:
> @marc-hb IIUC, you have a few WIP PRs to support Zephyr. Do any cover this ?

Not sure what you're referring to sorry. In any case I don't have any WIP related to this.

marc-hb, 2022-02-19:
> In any case I don't have any WIP related to this.

Well now I have. A bit:
- https://github.com/zephyrproject-rtos/zephyr/pull/42993

aiChaoSONG, 2022-02-28:
Correct rimage sign schema on sof side: https://github.com/thesofproject/sof/pull/5412

closed 2022-03-01 by abonislawski