thesofproject/sof #5345: [CML] error: log_entry_address 0x___ is not in dictionary range

marc-hb, 2022-02-11:
This is a follow-up to #5120.

#5120 was describing a very regular DMA corruption type: every time the end of the ring buffer was reached when it was misaligned.

The frequency of `error: log_entry_address 0x____ is not in dictionary range` errors has massively gone down now ([even before the alignment fix was merged](https://github.com/thesofproject/sof/pull/5329#issuecomment-1034565091)). But it's still not completely zero, this sof-logger warning is still happening in some specific tests on some specific platforms. Note this warning is NOT fatal, most of the time the tests PASS with it. It now seems rare enough not make the sof-logger crash (it was taking #5120 _very many_ corruption events to take down the sof-logger eventually) and it does not seem to happen at regular intervals anymore but it still happens.

Looking for instance at daily run ID 10105, I found it happening more than 10 times in these tests (and a few others):

Start Time: 2022-02-10 22:27:29 UTC
Kernel Branch: topic/sof-dev
Kernel Commit: 936898a6
SOF Branch: main
SOF Commit: 443b21de4b3f
Zephyr Commit: 5094a6e08cee

Found more than 10 occurences in:
10105?model=CML_RVP_SDW&testcase=volume-basic-test-50
10105?model=CML_SKU0983_SDW&testcase=multiple-pipeline-all-50

It's also happening in suspend/resume and pause/resume tests but I'm guessing it's more or less expected in those? So I ignored them. See thesofproject/linux/issues/3275 for more background.

After excluding all the "expected" cases, it seems to happen only on CML.

In 10105?model=CML_RVP_NOCODEC&testcase=check-playback-100times
I found the message 155 times and it seems to be very near `src/lib/alloc.c:593  INFO heap: system status` every time.

marc-hb, 2022-02-11:
Test PR #5340 makes the sof-logger stricter and demonstrates an easy way to reproduce.

mengdonglin, 2023-01-03:
Won't fix IPC3 issues not observed on stable-v2.2 branch

closed 2023-01-03 by mengdonglin