thesofproject/sof #5355: [BUG][TGL] xrun with ipc error when COMP_CONNECT when aplay and arecord simultaneously on TGLU_RVP_NOCODEC

XiaoyunWu6666, 2022-02-14:
**Describe the bug**
when running simultaneous playback and capture , xrun happened with ipc error when COMP_CONNECT
since inner daily testresult 10158

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-tgl-nocodec.tplg MODEL=TGLU_RVP_NOCODEC ~/sof-test/test-case/simultaneous-playback-capture.sh -l 25

**Reproduction Rate**
Not sure , haven't reproduced yet 
**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 98119478(https://github.com/thesofproject/linux/commit/98119478)
SOF Branch: main
SOF Commit:b8954754f055(https://github.com/thesofproject/sof/commit/b8954754f055)
Zephyr Commit: v3.0.0-rc3-1-g655b0d21274a

**Screenshots or console output**
[dmesg]
```
[ 5969.588446] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget BUF4.1 setup complete
[ 5969.588459] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[ 5969.588688] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[ 5969.588695] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 5969.588986] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[ 5969.588988] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget SSP1.IN setup complete
[ 5969.588989] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: setting up route BUF4.0 -> PCM1C
[ 5969.588992] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[ 5969.589160] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: reply size (16) exceeds the buffer size (12)
[ 5969.589167] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x30030000 (msg/reply size: 16/12): -22
[ 5969.589169] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: sof_ipc3_route_setup: route BUF4.0 -> PCM1C failed
[ 5969.589173] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 5970.085585] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 5970.085653] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 5970.091546] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x30020000 (msg/reply size: 12/12)
[ 5970.091564] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 5970.091569] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 5970.091597] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000081 rirb 00
[ 5970.091605] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 5970.091630] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 5970.091636] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 5970.091641] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 5970.091645] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware exception
[ 5970.091651] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 5970.091679] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[ 5970.091954] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: unexpected fault 0x00000000 trace 0x00004000
[ 5970.091961] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 5970.091975] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to free widget PCM1C
[ 5970.091993] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30210000: GLB_TPLG_MSG: BUFFER_FREE
[ 5970.092030] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: trace IO error
[ 5970.585612] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 5970.585667] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG: DMA_POSITION
[ 5970.595523] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x30210000 (msg/reply size: 12/12)
[ 5970.595541] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 5970.595545] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 5970.595574] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000081 rirb 00
[ 5970.595581] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 5970.595607] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 5970.595613] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 5970.595618] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 5970.595622] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware exception
[ 5970.595627] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 5970.595655] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[ 5970.595947] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: unexpected fault 0x00000000 trace 0x00004000
[ 5970.595954] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 5970.595962] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to free widget BUF4.0
[ 5970.595997] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 5971.099474] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x30020000 (msg/reply size: 12/12)
[ 5971.099491] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 5971.099495] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 5971.099523] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000081 rirb 00
[ 5971.099531] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 5971.099555] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 5971.099561] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 5971.099566] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 5971.099570] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware exception
[ 5971.099575] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 5971.099602] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[ 5971.099929] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: unexpected fault 0x00000000 trace 0x00004000
[ 5971.099936] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
```

[sof-logger]
```
[   655009903.711828] (       51871.457031) c3 dma-trace          /schedule/edf_schedule.c:116  ERROR schedule_edf_task(), task already queued or running 3
[   666027625.669857] (    11017722.000000) c3 dma-trace              src/drivers/dw/dma.c:1128 ERROR dw_dma_get_data_size(): xrun detected
[   666027647.232356] (          21.562500) c3 dma-trace                   src/audio/dai.c:881  ERROR dai_report_xrun(): underrun due to no data available
[   666027669.576105] (          22.343750) c3 dma-trace                   src/audio/dai.c:748  ERROR comp_underrun(): dev->comp.id = 14, source->avail = 0, copy_bytes = 0
[   666027737.492769] (          67.916664) c3 dma-trace          ....../pipeline-stream.c:134  ERROR pipeline_copy(): ret = -61, start->comp.id = 14, dir = 1
[   666027760.201101] (          22.708332) c3 dma-trace          ..../pipeline-schedule.c:183  ERROR pipeline_task(): xrun recovery failed! pipeline is stopped.
Skipped 5692 bytes after the last statement. 
```
complete dmesg :
[dmesg.txt](https://github.com/thesofproject/sof/files/8057654/dmesg.txt)

lgirdwood, 2022-02-14:
@XiaoyunWu6666 can this be reproduced with Zephyr ?

XiaoyunWu6666, 2022-02-15:
> @XiaoyunWu6666 can this be reproduced with Zephyr ?
haven't reproduced it on both TGLU_RVP_NOCODEC and  TGLU_RVP_NOCODEC_ZEPHYR .
I will post more details after I managed to reproduce it , thanks

mengdonglin, 2022-09-06:
Close as this issue cannot be reproduced in recent daily testing.

closed 2022-09-06 by mengdonglin