thesofproject/sof #5357: [RFC][CI] Build SOF on Zephyr with IPC4

aiChaoSONG, 2022-02-14:
We can already build SOF+Zephyr+IPC4+XCC manually with some hack or workaround, some issues need to be addressed before adding the build support in our build script.

SOF Application Kconfig is here: [intel_adsp_cavs25.conf](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/subsys/audio/sof/boards/intel_adsp_cavs25.conf) it is for IPC3.  We cannot make this one file flexible for both IPC3 and IPC4 support. There are two ways to add kconfig overlay:

- Use board revision: we can add `intel_adsp_cavs25_ipc4.conf` kconfig revision in the same folder on zephyr side, then use `west build --board intel_adsp_cavs25@ipc4` to build firmware. With this, `intel_adsp_cavs25.conf`, `intel_adsp_cavs25_ipc4.conf` and `prj.conf` will be merged together into `.config`
- Use OVERLAY_CONFIG env: we can specified OVERLAY_CONFIG like this: `export OVERLAY_CONFIG=folder-contains-the-overlay/intel_adsp_cavs25_ipc4.conf`, the file suffix has to be `.conf`. If we use this options, where should be put the overlay file, in zephyr tree or sof tree?

Talking about TGL and TGL-H, I think board revision is a good choice, but for IPC4+zephyr, I am not sure.

@lgirdwood  @kv2019i  @lyakh @marc-hb

lgirdwood, 2022-02-14:
@kv2019i @mwasko @nashif @andyross whats the preference here that would work for Zephyr  ? 
My understanding is that the board naming is resolved, but how do we differentiate the same board and carry upstream ?

mwasko, 2022-02-14:
@aiChaoSONG using board revision looks good to me.

lgirdwood, 2022-02-14:
So IIUC, this looks like a Zephyr "application" or Zephyr "module" configuration that could apply to any (or most) boards that run the SOF application / module. @andyross this would point more to a collection of overlays that would exist in the SOF module repo and be combined with the Zephyr config at build time ?

plbossart, 2022-02-14:
I think there's a definition/language issue here @aiChaoSONG @andyross 

intel_adsp_cavs25_ipc4.conf cannot be a board definition, it's an IP description.

A board description would be skl-hda-generic-4mics or adl-max98387-ssp2. The board makes use of the IP along with a subset of interfaces, gpios, external audio codecs, you name it.

andyross, 2022-02-14:
Indeed, it's a jargon collision.  In Zephyrese, device configurations live in a hierarchy of (in order of increasing specificity) "arch", "soc", and "board".   The file referenced above was the board configuration for this particular device in the Zephyr tree.  And the suggestion above was to use a board variant definition (which is certainly possible) to hold the "use IPC4" cookies.  And I'm saying no, that's a bad idea and that software configuration belongs in the application layer somewhere.  (Though I take no position as to where exactly!)

plbossart, 2022-02-14:
intel_adsp_cavs25_ipc4.conf can't be a "board" definition but may not even be a 'soc' definition.

Examples to prove my point: 
JSL is cAVS 2.0 but has fewer cores and memory than ICL which is also cAVS2.0. 
TGL-H is cAVS 2.5 but have less memory than TGL

So yes, the split between arch, soc and board is a very good one, and our configs ought to follow that model.

aiChaoSONG, 2022-02-15:
IPC4 kconfig is SOF application level configuration, so it should stay with SOF, right? @andyross 
@lgirdwood if we are going to put such overlays in SOF tree, should I put them in `arch/xtensa/configs/override`,  or create a new overlay folder special for zephyr?

marc-hb, 2022-02-16:
> intel_adsp_cavs25_ipc4.conf can't be a "board" definition but may not even be a 'soc' definition.

I believe you're commenting on the hardware part of this name: `intel_adsp_cavs25_...` but this issue is about the software part `..._ipc4` which according to @andyross belongs to a totally different place (cause it's software). Do you think we should change the way hardware is named / configured before even looking at the software part?

aborisovich, 2022-02-16:
Guys can we consider using [shields](https://docs.zephyrproject.org/latest/guides/porting/shields.html) for different platforms of the same family (lets say TGL and TGL-H would be both shields of intel_adsp_cavs25 board).
Then in the shield files we can define software features. Yes I know that they are not the part of the hardware configuration but they represent a feature set of 'this specific' platform. In my opinion it does not make sense to describe what we mean when we say 'TGL' only by hardware characteristics. We mostly mean software features too in context of the firmware.

aborisovich, 2022-02-16:
Guys I think we should start with creation of "platform" directory in the SOF.
This config approach is great but we also have some data in script files that in my opinion should not be there but rather bound to CMake. I am talking about: XTENSA_CORE, XTENSA_TOOLS_VERSION, RIMAGE_KEY, name dictionary entries in the xtensa-build-zephyr scripts. Those are well-known values. So I would put prj_<build>.conf files which @andyross mentioned in [Setting initial configuration](https://docs.zephyrproject.org/latest/guides/build/kconfig/setting.html#initial-conf) in the platform directory tree to create something like this:

platforms:

- TGL
** platform.cmake
** prj.conf

- TGL-H
** platform.cmake
** prj.conf

lgirdwood, 2022-02-22:
@kv2019i @mwasko @macchian @mmaka1 @aborisovich @andyross @nashif topic for Wednesday. Need to resolve soon.

XiaoyunWu6666, 2022-02-24:
fix on zephyr side :https://github.com/zephyrproject-rtos/zephyr/pull/42993
potential fix on sof side :https://github.com/thesofproject/sof/pull/5412

lgirdwood, 2022-02-24:
Zephyr fixed now merged, SOF side passing CI.

aiChaoSONG, 2022-02-28:
close this RFC by #5412

closed 2022-02-28 by aiChaoSONG