thesofproject/sof #5362: [BUG] 5.17-rc4, DMA-API: sof-audio-pci-intel-cnl 0000:00:1f.3: mapping sg segment longer than device claims to support

cmurf, 2022-02-14:
**Describe the bug**
Getting a warning in dmesg when running 5.17 series kernels

[   15.267615] DMA-API: sof-audio-pci-intel-cnl 0000:00:1f.3: mapping sg segment longer than device claims to support [len=540672] [max=65536]
[   15.269962] WARNING: CPU: 0 PID: 5 at kernel/dma/debug.c:1162 debug_dma_map_sg+0x328/0x380

**To Reproduce**
Boot

**Reproduction Rate**
Always

**Expected behavior**
Not this

**Impact**
None that I'm immediately aware of

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.17.0-0.rc4.96.fc36.x86_64+debug
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
```
[   15.151384] sof-audio-pci-intel-cnl 0000:00:1f.3: hda codecs found, mask 5
[   15.153777] sof-audio-pci-intel-cnl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   15.156152] sof-audio-pci-intel-cnl 0000:00:1f.3: DMICs detected in NHLT tables: 4
[   15.256677] sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware info: version 2:0:0-b678a
[   15.258721] sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware: ABI 3:20:0 Kernel ABI 3:18:0
[   15.260959] sof-audio-pci-intel-cnl 0000:00:1f.3: warn: FW ABI is more recent than kernel
[   15.263115] sof-audio-pci-intel-cnl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
[   15.265457] ------------[ cut here ]------------
[   15.267615] DMA-API: sof-audio-pci-intel-cnl 0000:00:1f.3: mapping sg segment longer than device claims to support [len=540672] [max=65536]
[   15.269962] WARNING: CPU: 0 PID: 5 at kernel/dma/debug.c:1162 debug_dma_map_sg+0x328/0x380
[   15.271716] Modules linked in: kvm snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic irqbypass rapl intel_cstate intel_uncore snd_sof_pci_intel_cnl snd_sof_intel_hda_common soundwire_intel soundwire_generic_allocation soundwire_cadence snd_sof_intel_hda snd_sof_pci snd_sof_xtensa_dsp snd_sof soundwire_bus iwlmvm snd_soc_skl snd_soc_hdac_hda snd_hda_ext_core snd_soc_sst_ipc snd_soc_sst_dsp snd_soc_acpi_intel_match snd_soc_acpi snd_soc_core mac80211 snd_compress ac97_bus snd_pcm_dmaengine snd_hda_intel snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec libarc4 think_lmi(+) firmware_attributes_class snd_hda_core intel_wmi_thunderbolt wmi_bmof snd_hwdep snd_seq snd_seq_device snd_pcm i2c_i801 btusb i2c_smbus uvcvideo btrtl iwlwifi snd_timer btbcm btintel videobuf2_vmalloc videobuf2_memops btmtk videobuf2_v4l2 thunderbolt videobuf2_common bluetooth mei_me joydev cfg80211 processor_thermal_device_pci_legacy videodev mei processor_thermal_device idma64 processor_thermal_rfim mc
[   15.271907]  processor_thermal_mbox ecdh_generic processor_thermal_rapl intel_rapl_common intel_soc_dts_iosf intel_pch_thermal thinkpad_acpi ledtrig_audio platform_profile rfkill snd soundcore int3403_thermal int340x_thermal_zone int3400_thermal acpi_thermal_rel acpi_pad acpi_tad vfat fat zram i915 hid_multitouch crct10dif_pclmul crc32_pclmul crc32c_intel nvme ucsi_acpi ghash_clmulni_intel e1000e typec_ucsi serio_raw nvme_core ttm typec wmi i2c_hid_acpi i2c_hid video pinctrl_cannonlake ipmi_devintf ipmi_msghandler fuse
[   15.284052] CPU: 0 PID: 5 Comm: kworker/0:0 Not tainted 5.17.0-0.rc4.96.fc36.x86_64+debug #1
[   15.285653] Hardware name: LENOVO 20QDS3E200/20QDS3E200, BIOS N2HET66W (1.49 ) 11/10/2021
[   15.287116] Workqueue: events sof_probe_work [snd_sof]
[   15.288896] RIP: 0010:debug_dma_map_sg+0x328/0x380
[   15.291373] Code: 5c 24 10 8b 4c 24 18 48 8b 54 24 20 48 89 c6 44 8b 44 24 2c 48 c7 c7 78 d8 83 a8 4c 89 5c 24 10 4c 89 4c 24 08 e8 54 7d c6 00 <0f> 0b 4c 8b 5c 24 10 4c 8b 4c 24 08 8b 15 76 05 30 02 85 d2 0f 85
[   15.293651] RSP: 0018:ffffa8d1400afc80 EFLAGS: 00010282
[   15.296773] RAX: 000000000000007f RBX: ffff8d67c2dd30d0 RCX: 0000000000000000
[   15.296777] RDX: 0000000000000001 RSI: ffffffffa889dc38 RDI: 00000000ffffffff
[   15.296780] RBP: ffff8d67c5466040 R08: 0000000000000000 R09: ffffa8d1400afab8
[   15.296782] R10: ffffa8d1400afab0 R11: ffffffffa9165128 R12: 0000000000000000
[   15.296784] R13: 0000000000000001 R14: 0000000000000001 R15: ffff8d67c14e3380
[   15.296786] FS:  0000000000000000(0000) GS:ffff8d6b2ca00000(0000) knlGS:0000000000000000
[   15.310044] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   15.310051] CR2: 00007f5d919a8e48 CR3: 000000017ec98004 CR4: 00000000003706f0
[   15.310055] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[   15.310058] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[   15.310063] Call Trace:
[   15.310066]  <TASK>
[   15.310093]  dma_alloc_noncontiguous+0x14b/0x1a0
[   15.310115]  snd_dma_noncontig_alloc+0x23/0x70 [snd_pcm]
[   15.310139]  snd_dma_alloc_dir_pages+0x50/0x80 [snd_pcm]
[   15.310159]  cl_stream_prepare.constprop.0+0x50/0xd0 [snd_sof_intel_hda_common]
[   15.310179]  hda_dsp_cl_boot_firmware+0x84/0x780 [snd_sof_intel_hda_common]
[   15.327745] zram0: detected capacity change from 0 to 16777216
[   15.329313]  snd_sof_run_firmware+0xb7/0x290 [snd_sof]
[   15.329335]  ? snd_sof_load_firmware_raw+0xaa/0x410 [snd_sof]
[   15.349152]  sof_probe_work+0xdc/0x280 [snd_sof]
[   15.349177]  process_one_work+0x297/0x5f0
[   15.367245]  worker_thread+0x4f/0x390
[   15.370436]  ? process_one_work+0x5f0/0x5f0
[   15.373680]  kthread+0xf2/0x120
[   15.376914]  ? kthread_complete_and_exit+0x20/0x20
[   15.379879]  ret_from_fork+0x1f/0x30
[   15.383177]  </TASK>
[   15.386397] irq event stamp: 328373
[   15.389597] hardirqs last  enabled at (328385): [<ffffffffa717b270>] __up_console_sem+0x60/0x70
[   15.392856] hardirqs last disabled at (328396): [<ffffffffa717b255>] __up_console_sem+0x45/0x70
[   15.395953] softirqs last  enabled at (328318): [<ffffffffa70f2a63>] __irq_exit_rcu+0xf3/0x170
[   15.399156] softirqs last disabled at (328313): [<ffffffffa70f2a63>] __irq_exit_rcu+0xf3/0x170
[   15.402316] ---[ end trace 0000000000000000 ]---
[   15.444250] intel_tcc_cooling: Programmable TCC Offset detected
[   15.513003] Adding 8388604k swap on /dev/zram0.  Priority:100 extents:1 across:8388604k SSDscFS
[   15.520249] sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware info: version 2:0:0-b678a
[   15.520255] sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware: ABI 3:20:0 Kernel ABI 3:18:0
[   15.520258] sof-audio-pci-intel-cnl 0000:00:1f.3: warn: FW ABI is more recent than kernel

```

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

[dmesg.txt](https://github.com/thesofproject/sof/files/8063558/dmesg.txt)

cmurf, 2022-02-14:
ThinkPad X1 Carbon Gen 7 Laptop
[lspci-vvnn.txt](https://github.com/thesofproject/sof/files/8063568/lspci-vvnn.txt)

plbossart, 2022-02-15:
Thanks for reporting this @cmurf, very puzzling problem.

As you can see from the trace, we rely on default helpers from the sound/ tree, the allocation of the DMA is not done inside the SOF driver but in those common helpers.

To check if indeed the problem is with this set of helpers would you mind trying to run the legacy driver by adding this in /etc/modprobe/alsa-base.conf (or any file in that directory with the .conf extension):

options snd_intel_dspcfg dsp_driver=1

reboot and see if you see the same issue with that driver?

cmurf, 2022-02-15:
Added to `/etc/modprobe.d/alsa-base.conf` and the next boot contains:

```
[   21.636691] DMA-API: snd_hda_intel 0000:00:1f.3: mapping sg segment longer than device claims to support [len=73728] [max=65536]
[   21.636703] WARNING: CPU: 5 PID: 1153 at kernel/dma/debug.c:1162 debug_dma_map_sg+0x328/0x380
```
[dmesg2.txt](https://github.com/thesofproject/sof/files/8064931/dmesg2.txt)

plbossart, 2022-02-15:
Great, thanks @cmurf, it's super useful to see that both drivers behave in the same way.

At this point we will have to ping @tiwai to ask for his guidance. It's no longer an SOF bug proper but something going boink in the memory management.