thesofproject/sof #5364: [BUG] Some Kconfig configuration builds fail

lyakh, 2022-02-15:
**Describe the bug**
Some `make menuconfig` configurations cannot be built. The "Audio components" menu contains many entries, that aren't enabled e.g. for TGL configurations and probably in all other default configurations too. Some of them can be enabled, but a build then fails, because they can only be built with xcc, because they depend on HiFi support. Yet others depend on add-on libraries and fail to link without them.

**To Reproduce**
`make menuconfig`, select all entries under "Audio components" and try to build

**Expected behavior**
The compiler is known at menuconfig time. It was configured during the cmake step, so, it should be possible to exclude Kconfig options, that cannot be built with the configured toolchain. As for libraries, this might be more difficult, but maybe we could check if required add-on libraries can be found in library paths. The easiest way would possibly be specifying explicitly in Kconfig paths to respective libraries, and if a path hasn't been specified, don't build the target.

**Impact**
Makes testing of rarely used Kconfig options more difficult

marc-hb, 2022-02-17:
The (orphaned) kconfig.py we use offers unfortunately no "randconfig" option like the C version does.

We could still test allconfig, alldefconfig and others in CI.

marc-hb, 2022-03-08:
We should also fail the build on kconfig warnings. kconfiglib is surprisingly lenient with malformed defconfig files, maybe with other malformed files too. Here's one-line temporary hack that achieves that:

```diff
--- a/scripts/kconfig/kconfiglib.py
+++ b/scripts/kconfig/kconfiglib.py
@@ -3617,6 +3617,8 @@ class Kconfig(object):
             if self._warn_to_stderr:
                 sys.stderr.write(msg + "\n")
 
+        raise BaseException("Werror")
+
     def _warn_undef_assign(self, msg, filename, linenr):
         # See the class documentation
```

Zephyr has a proper kconfiglib -Werror implementation, not sure how.

We have a single kconfig warning in the XTOS build right now:
```
warning: the int symbol CORE_COUNT (defined at src/platform/Kconfig:286) has a non-int default MP_NUM_CPUS (undefined)
```
@kv2019i any way to fix this warning now that some more multicore fixes have been merged?