thesofproject/sof #5371: How to synchronize producers and consumers?

lenghonglin, 2022-02-16:
Hello, everyone , I am porting sof to a new platform.

Now i meet a question with playback
 
[image]

`uds` mean producer, `cnt` means consumer. and `BUFFER_PERIOD_COUNT` is 4.  It show that the producers can produce faster than consumers. when uds - cnt = 4 , the dma_buffer will be coverd by producer.

Recording can also be a problem.

How to synchronize producers and consumers in SOF?

paulstelian97, 2022-02-16:
When production is faster than consumption everything is perfectly fine -- the production is just forced to stall some periods (as the buffer will report insufficient space, and components are supposed to only put in what fits), so long as the available data is sufficient to feed the next period.

On playback, the fact that the consumer is hardware and the producer is faster means the DSP just gets some idle time. On capture (recording) it's more interesting, because we want the consumer to be faster (and the consumer would then synchronize on insufficient input data and stall some cycles).

Stalling processing portions of a pipeline is perfectly fine. The host component is also allowed to stall without an issue.

There is a concept we use called xflow, where the stalling happens for the opposite reason. You have overflows, where the pipeline is slower than a capture device, and underflows, where the pipeline is slower than the consumer AKA the hardware. In all circumstances, we want the pipeline, or rather the processing components of said pipeline, to be faster than the hardware itself. They sync up by meeting buffers that are already full.

lenghonglin, 2022-02-16:
> When production is faster than consumption everything is perfectly fine -- the production is just forced to stall some periods (as the buffer will report insufficient space, and components are supposed to only put in what fits), so long as the available data is sufficient to feed the next period.
> 
> On playback, the fact that the consumer is hardware and the producer is faster means the DSP just gets some idle time. On capture (recording) it's more interesting, because we want the consumer to be faster (and the consumer would then synchronize on insufficient input data and stall some cycles).
> 
> Stalling processing portions of a pipeline is perfectly fine. The host component is also allowed to stall without an issue.
> 
> There is a concept we use called xrun, where the stalling happens for the opposite reason. You have overruns, where the pipeline is slower than a capture device (as consequence we lose portions of a recording), and underruns, where the pipeline is slower than the consumer AKA the hardware (in which case the hardware is forced to pause due to lack of data). In all circumstances, we want the pipeline, or rather the processing components of said pipeline, to be faster than the hardware itself. They sync up by meeting buffers that are already full.

Thanks for your reply.
You mean  production is faster than consumption that is normal. and if met this condition, should call `xrun`， right?

lgirdwood, 2022-02-16:
@lenghonglin playback speed is governed by how fast your DMA (and hence DAI) consumes the data. The producer will block if the downstream pipeline is full until there is buffer space available.
So on Intel platforms, the host side playback DMA (that copies audio from the host to the DSP pipeline) will only copy data if there is enough free space.

lenghonglin, 2022-02-16:
> @lenghonglin playback speed is governed by how fast your DAI DMA (and hence DAI) consumes the data. The producer will block if the downstream pipeline is full until there is buffer space available. So on Intel platforms, the host side playback DMA (that copies audio from the host to the DSP pipeline) will only copy data if there is enough free space.

How to block producer ?

lenghonglin, 2022-02-16:
I try to call `xrun` when  producers faster than consumers.
[image]

lgirdwood, 2022-02-16:
> How to block producer ?

See host.c
```c
/**
 * Calculates bytes to be copied in normal mode.
 * @param dev Host component device.
 * @return Bytes to be copied.
 */
static uint32_t host_get_copy_bytes_normal(struct comp_dev *dev)
{
	struct host_data *hd = comp_get_drvdata(dev);
	struct comp_buffer *buffer = hd->local_buffer;
	uint32_t avail_bytes = 0;
	uint32_t free_bytes = 0;
	uint32_t copy_bytes = 0;
	int ret;

	/* get data sizes from DMA */
	ret = dma_get_data_size(hd->chan, &avail_bytes,
				&free_bytes);
	if (ret < 0) {
		comp_err(dev, "host_get_copy_bytes_normal(): dma_get_data_size() failed, ret = %u",
			 ret);
		/* return 0 copy_bytes in case of error to skip DMA copy */
		return 0;
	}

	buffer = buffer_acquire(buffer);

	/* calculate minimum size to copy */
	if (dev->direction == SOF_IPC_STREAM_PLAYBACK) {
		/* limit bytes per copy to one period for the whole pipeline
		 * in order to avoid high load spike
		 */
		free_bytes = audio_stream_get_free_bytes(&buffer->stream);
		copy_bytes = MIN(hd->period_bytes, MIN(avail_bytes, free_bytes));
		if (!copy_bytes)
			comp_info(dev, "no bytes to copy, %d free in buffer, %d available in DMA",
				  free_bytes, avail_bytes);
	} else {
		avail_bytes = audio_stream_get_avail_bytes(&buffer->stream);
		copy_bytes = MIN(avail_bytes, free_bytes);
		if (!copy_bytes)
			comp_info(dev, "no bytes to copy, %d avail in buffer, %d free in DMA",
				  avail_bytes, free_bytes);
	}

	buffer = buffer_release(buffer);

	/* copy_bytes should be aligned to minimum possible chunk of
	 * data to be copied by dma.
	 */
	return ALIGN_DOWN(copy_bytes, hd->dma_copy_align);
}
```
This will copy 0 bytes (and return NO error) if there is no free space which has the effect of not moving the ALSA position pointer and blocking any any applications which call write().

On the next timer tick the DAI DMA should have consumed data meaning there is free space from the host DMA to copy data (and unblock the producers on the host).

lenghonglin, 2022-02-16:
> > How to block producer ?
> 
> See host.c
> 
> ```c
> /**
>  * Calculates bytes to be copied in normal mode.
>  * @param dev Host component device.
>  * @return Bytes to be copied.
>  */
> static uint32_t host_get_copy_bytes_normal(struct comp_dev *dev)
> {
> 	struct host_data *hd = comp_get_drvdata(dev);
> 	struct comp_buffer *buffer = hd->local_buffer;
> 	uint32_t avail_bytes = 0;
> 	uint32_t free_bytes = 0;
> 	uint32_t copy_bytes = 0;
> 	int ret;
> 
> 	/* get data sizes from DMA */
> 	ret = dma_get_data_size(hd->chan, &avail_bytes,
> 				&free_bytes);
> 	if (ret < 0) {
> 		comp_err(dev, "host_get_copy_bytes_normal(): dma_get_data_size() failed, ret = %u",
> 			 ret);
> 		/* return 0 copy_bytes in case of error to skip DMA copy */
> 		return 0;
> 	}
> 
> 	buffer = buffer_acquire(buffer);
> 
> 	/* calculate minimum size to copy */
> 	if (dev->direction == SOF_IPC_STREAM_PLAYBACK) {
> 		/* limit bytes per copy to one period for the whole pipeline
> 		 * in order to avoid high load spike
> 		 */
> 		free_bytes = audio_stream_get_free_bytes(&buffer->stream);
> 		copy_bytes = MIN(hd->period_bytes, MIN(avail_bytes, free_bytes));
> 		if (!copy_bytes)
> 			comp_info(dev, "no bytes to copy, %d free in buffer, %d available in DMA",
> 				  free_bytes, avail_bytes);
> 	} else {
> 		avail_bytes = audio_stream_get_avail_bytes(&buffer->stream);
> 		copy_bytes = MIN(avail_bytes, free_bytes);
> 		if (!copy_bytes)
> 			comp_info(dev, "no bytes to copy, %d avail in buffer, %d free in DMA",
> 				  avail_bytes, free_bytes);
> 	}
> 
> 	buffer = buffer_release(buffer);
> 
> 	/* copy_bytes should be aligned to minimum possible chunk of
> 	 * data to be copied by dma.
> 	 */
> 	return ALIGN_DOWN(copy_bytes, hd->dma_copy_align);
> }
> ```
> 
> This will copy 0 bytes (and return NO error) if there is no free space which has the effect of not moving the ALSA position pointer and blocking any applications which call write().
> 
> On the next timer tick the DAI DMA should have consumed data meaning there is free space for the host DMA to copy data (and unblock the producers on the host).

I got this part, so it's very important in dai_copy
```c
/* copy and process stream data from source to sink buffers */
static int dai_copy(struct comp_dev *dev)
{
	struct dai_data *dd = comp_get_drvdata(dev);
	uint32_t dma_fmt = dd->dma_buffer->stream.frame_fmt;
	const uint32_t sampling = get_sample_bytes(dma_fmt);
	struct comp_buffer *buf = dd->local_buffer;
	uint32_t avail_bytes = 0;
	uint32_t free_bytes = 0;
	uint32_t copy_bytes = 0;
	uint32_t src_samples;
	uint32_t sink_samples;
	uint32_t samples;
	int ret = 0;
	uint32_t flags = 0;

	comp_dbg(dev, "dai_copy()");

	/* get data sizes from DMA */
	ret = dma_get_data_size(dd->chan, &avail_bytes, &free_bytes);	
	if (ret < 0) {
		dai_report_xrun(dev, 0);
		return ret;
	}

	buffer_lock(buf, &flags);

	/* calculate minimum size to copy */
	if (dev->direction == SOF_IPC_STREAM_PLAYBACK) {
		src_samples = audio_stream_get_avail_samples(&buf->stream);
		sink_samples = free_bytes / sampling;
		samples = MIN(src_samples, sink_samples);
	} else {
		src_samples = avail_bytes / sampling;
		sink_samples = audio_stream_get_free_samples(&buf->stream);
		samples = MIN(src_samples, sink_samples);
	}

	/* limit bytes per copy to one period for the whole pipeline
	 * in order to avoid high load spike
	 */
	samples = MIN(samples, dd->period_bytes / sampling);

	copy_bytes = samples * sampling;

	buffer_unlock(buf, flags);

	comp_dbg(dev, "dai_copy(), dir: %d copy_bytes= 0x%x, frames= %d",
		 dev->direction, copy_bytes,
		 samples / buf->stream.channels);

	/* Check possibility of glitch occurrence */
	if (dev->direction == SOF_IPC_STREAM_PLAYBACK &&
	    copy_bytes + avail_bytes < dd->period_bytes)
		comp_warn(dev, "dai_copy(): Copy_bytes %d + avail bytes %d < period bytes %d, possible glitch",
			  copy_bytes, avail_bytes, dd->period_bytes);
	else if (dev->direction == SOF_IPC_STREAM_CAPTURE &&
		 copy_bytes + free_bytes < dd->period_bytes)
		comp_warn(dev, "dai_copy(): Copy_bytes %d + free bytes %d < period bytes %d, possible glitch",
			  copy_bytes, free_bytes, dd->period_bytes);

	/* return if nothing to copy */
	if (!copy_bytes) {
		comp_warn(dev, "dai_copy(): nothing to copy");
		return 0;
	}

	ret = dma_copy(dd->chan, copy_bytes, 0);
	if (ret < 0) {
		dai_report_xrun(dev, copy_bytes);
		return ret;
	}

	return ret;
}
```

lenghonglin, 2022-02-16:
> I got this part, so it's very important in dai_copy
>
> ```c
> /* copy and process stream data from source to sink buffers */
> static int dai_copy(struct comp_dev *dev)
> {
> 	struct dai_data *dd = [...]

`dma_get_data_size(dd->chan, &avail_bytes, &free_bytes);` this function will get correct `free_bytes` and if no space, will return 0. and next will block by `host_copy()`, right?

lgirdwood, 2022-02-16:
> `dma_get_data_size(dd->chan, &avail_bytes, &free_bytes);` this function will get correct `free_bytes` and if no space, will return 0. and next will block by `host_copy()`, right?

Sorry - I should be clear, it will "block" the Linux userspace (application) producer. The FW pipeline will still keep running on every timer tick and copy data when space permits.

So the host playback DMA copy() will only copy bytes if the pipeline has free space other wise it copies 0 bytes (with NO error). Free space is created in the pipeline by the playback DAI DMA consuming (at a fixed rate).

lenghonglin, 2022-02-16:
> 

Yes, `block by host_copy()` means host copies 0 bytes , I may have misexpressed myself.

Like this: 
[image]

display `buffer covered` mean `producer count` - `consumer count` = 4. ps dai dma elem size is 4.

lgirdwood, 2022-02-16:
DAI elem size 4 seems too small. On Intel platforms we program the DAI DMA copy size to 48 frames per timer tick (for 48kHz) i.e. we copy 48 * 2 * 2 bytes (48k stereo 16bit). This means the DMA updates the pipeline and consumes 48 frames per update.

lgirdwood, 2022-02-16:
> > DAI elem size 4 seems too small. On Intel platforms we program the DAI DMA copy size to ~48 frames per timer tick (for 48kHz) i.e. we copy 48 * 2 * 2 bytes (48k stereo 16bit). This means the DMA updates the pipeline and consumes 48 frames per update.
> 
> sry，DAI elem cnt is 4

oh, you mean the DMA has 4 periods - ack we always have >= 3 DAI DMA periods on Intel to avoid the case where FW is writing and DMA is reading from the same period. Intel DAI period size is ~48 frames.

marc-hb, 2022-02-17:
> the production is just forced to stall some periods (as the buffer will report insufficient space, and components are supposed to only put in what fits)

A short yet precise name for this is "backpressure".

lenghonglin, 2022-03-01:
> > the production is just forced to stall some periods (as the buffer will report insufficient space, and components are supposed to only put in what fits)
> 
> A short yet precise name for this is "[backpressure](https://www.google.com/search?q=data+backpressure)".

Got it~  Learned a new term, Thanks again

closed 2022-03-01 by lenghonglin