thesofproject/sof #5375: [BUG] soc_pcm_hw_params() failed (-22) for PCM7 (BT) on ADLP I2S Chrome device

kv2019i, 2022-02-16:
**Describe the bug**
sof-test failure on the BT PCM node 

Intel internal daily plan reference: #9293

**To Reproduce**
sof-test
 ~/sof-test/test-case/check-playback.sh -d 3 -l 1 -r 1 -F 

Or directly with aplay

aplay   -Dhw:0,7 -r 8000 -c 1 -f S16_LE -d 3 /dev/zero -v -q
`
**Reproduction Rate**
100%

**Expected behavior**
Playback to BT PCM is expected to fail if no Bluetooth audio connection is established in the system, but the FW error shown here is unexpected.

**Impact**
Test error

**Environment**
 * Kernel: latest sof-dev
 * SOF: latest main
 * Topology: adl-max98390-rt5682.tplg
 * Platform: ADLP Chrome Device

**dmesg**
```
[   37.591766] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   37.592079] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x60010000 (msg/reply size: 108/20): -22
[   37.592086] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: HW params ipc failed for stream 1
[   37.592088] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[   37.592094] kernel:  Bluetooth: ASoC: __soc_pcm_hw_params() failed (-22)```

** DMA trace **
```
[      102359.787599] (          26.510416) c0 dai          13.68          src/audio/dai.c:298  ERROR dai_verify_params(): pcm rate parameter 8000 does not match hardware rate 16000
[      102380.725098] (          20.937500) c0 dai          13.68          src/audio/dai.c:460  ERROR dai_params(): pcm params verification failed.
[      102401.662598] (          20.937500) c0 pipe         13.69 ....../pipeline-params.c:242  ERROR pipeline_params(): ret = -22, host->comp.id = 66
[      102424.162597] (          22.500000) c0 ipc                  src/ipc/ipc3/handler.c:283  ERROR ipc: pipe 13 comp 66 params failed -22```

** Notes **
Similar to https://github.com/thesofproject/linux/issues/3359

lgirdwood, 2022-02-16:
@mwasko fyi - can be reproduced with adl-004-drop-stable.

closed 2022-02-16 by yongzhi1

johnylin76, 2022-02-24:
Hi Kai,

Do you think this bug would be detectable and make bad influences for end-users of the customer devices?
Typically "playback to BT PCM while no BT audio connection" is not a reasonable action for audio server (userspace) to take. (Not guaranteed for some error states though.)
Would the FW error cause unrecoverable issues of BT PCM pipeline (or even the whole FW)?

The reason for asking these questions is we are approaching the feature-freeze deadline of ChromeOS milestone M100, which we planned to launch the offload feature on. We would like to know the severity level for the offload feature without the fix.

Thanks.

sathyap-chrome, 2022-02-24:
@johnylin76 the fix for the issue is already part of new adl 004 drop merged last night. The error was detectable on few headsets which support only NBS.

Let me know if you need more details

johnylin76, 2022-02-25:
Thanks Sathya.
>The error was detectable on few headsets which support only NBS.

Do you mean the user may sense something weird for the headset? such as sound artifacts, connection lost/unstable, and anything might bother users.