thesofproject/sof #5380: [BUG]open volume of pipeline,and the audio output is noisy

bawanqin1987, 2022-02-17:
**Describe the bug**
when i use pipe-passthrough-playback.m4 ,the audio data playback ok
when i use pipe-volume-playback.m4,the audio data playback with noise

**To Reproduce**
with pipe-volume-playback.m4,i feed a audio format file,and by dai--dac drv(our codec dac path),There is a noise in the speaker output
and bypass volume ,there is no noise,data is ok

**Reproduction Rate**
10/10

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

bawanqin1987, 2022-02-17:
I initially suspect that the output of the digital field exceeds 0dB after the gameplay of volume's volume curve and the volume adjustment of our DAI DAC driver combined, which is distorted
So could you please tell me how to use volume and what should be paid attention to

plbossart, 2022-02-17:
@bawanqin1987 would you mind adding more details:

Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
Kernel: {SHA}
SOF: {SHA}
Name of the topology file
Topology: {FILE}
Name of the platform(s) on which the bug is observed.
Platform: {PLATFORM}

???

bawanqin1987, 2022-02-17:
Kernel: {zephyr}
SOF: {1.5.0}
Name of the topology file
Topology: {pipe-volume-playback.m4}
Name of the platform(s) on which the bug is observed.
Platform: {CSK6}with my new add

lgirdwood, 2022-02-18:
@bawanqin1987 check your total pipeline gain is <= 0dB. Greater than 0dB gain can cause clipping and distort the ouput. If it's not the gain, then I would check your DMA, DAI and codec configs.

closed 2025-07-25 by github-actions[bot]