thesofproject/sof #5381: [SOF CODEC]how can i use sof/cadence decoder or encoder?

bawanqin1987, 2022-02-17:
i wannna use cadence encoder and decoder,how can i open the cadence codec with my sof?

i check the source support,there are cadence codec supportting,
i now,use the pipe-passthrough-playback.m4 had already test the audio ok
so ,how can i open the cadence codec with my pipeline?

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

lenghonglin, 2022-02-17:
https://github.com/thesofproject/sof/blob/main/zephyr/CMakeLists.txt   it seems zephy sof not support codec_adpater

bawanqin1987, 2022-02-17:
Kernel: {zephyr}
SOF: {1.5.0}
Name of the topology file
Topology: {pipe-passthrough-playback.m4}
Name of the platform(s) on which the bug is observed.
Platform: {CSK6} our new add ,more sames to nxp

dbaluta, 2022-02-17:
@bawanqin1987 Have a look at the pipeline we have for supporting MP3 with i.MX8QXP/i.MX8QM here: 

https://github.com/thesofproject/sof/blob/main/tools/topology/topology1/development/sof-imx8-compr-mp3-wm8960.m4

You will also need tho have the cadence binary libraries: (these are provided by Cadence!!). Once you got them, you need to set some config options:

```
CONFIG_CADENCE_CODEC_MP3_DEC=y
CONFIG_CADENCE_CODEC_MP3_DEC_LIB="path/to/cadence/library/xa_mp3_dec.a"
```

This works and it is tested for NXP platforms. I can assist you with that. For Intel platforms maybe @ranj063 can help you.

lenghonglin, 2022-02-17:
> @bawanqin1987 Have a look at the pipeline we have for supporting MP3 with i.MX8QXP/i.MX8QM here:
> 
> https://github.com/thesofproject/sof/blob/main/tools/topology/topology1/development/sof-imx8-compr-mp3-wm8960.m4
> 
> You will also need to have the cadence binary libraries: (these are provided by Cadence!!). Once you got them, you need to set some config options:
> 
> ```
> CONFIG_CADENCE_CODEC_MP3_DEC=y
> CONFIG_CADENCE_CODEC_MP3_DEC_LIB="path/to/cadence/library/xa_mp3_dec.a"
> ```
> 
> This works and it is tested for NXP platforms. I can assist you with that. For Intel platforms maybe @ranj063 can help you.

`CONFIG_COMP_CODEC_ADAPTER`

dbaluta, 2022-02-17:
> https://github.com/thesofproject/sof/blob/main/zephyr/CMakeLists.txt this file not support `CONFIG_COMP_CODEC_ADAPTER`

We will add this to zephyr in the next few weeks. Anyhow, @bawanqin1987 can try this with proper SOF without Zephyr.

dbaluta, 2022-02-17:
> 

> 1.CONFIG_CADENCE_CODEC_MP3_DEC_LIB="path/to/cadence/library/xa_mp3_dec.a" where i cp this .a ?Where should I maintain the codec library on the file system?
> 2.cadence .a need Do all these. A need to be rebuilt?

1.
You can add Cadence library anywhere on your filestem and provide the correct path to CONFIG_CADENCE_CODEC_MP3_DEC_LIB config option.

Here is on my machine:

```
CONFIG_CADENCE_CODEC_MP3_DEC_LIB="/home/daniel/work/repos/imx-audio-codec/core_lib/pilibxa_mp3_dec/lib/xa_mp3_dec.a"
```

2. I don't think I understand your question. You need to get the library from Cadence (are you a cadence customer?!), I think there are some fees involved. You will need to rebuild just the firmware and link it with Cadence library.

The cadence library it is just a binary. No need to modify it. Use it as it is.

What platforms are you trying to enable?

bawanqin1987, 2022-02-17:
i wanns test with my target board,so i think the .a will be Put on file system,Then the SOF on the AP side delivers the playback service, and calls. A to play after hifi4 starts?

bawanqin1987, 2022-02-17:
then ,i test the sof cadence codec with zpehyr os..and it's does not work? it's not support with zpehyr?

dbaluta, 2022-02-17:
@bawanqin1987 Yes, currently codec_adapter is not supported with zephyr. I have added a first PR to add this support.

Please have a look at:

https://github.com/thesofproject/sof/pull/5384

The only thing that needs to be done is to find a way to link the cadence libraries with zepyr firmware. @lgirdwood any idea? We need a function similar with SOF - > sof_add_static_library

Meanhwile, @bawanqin1987 you can use codec_adapter's passhthrough codec type, to get you started.

lgirdwood, 2022-02-21:
Zephyr builds SOF as a module and imports it's cmake and Kconfig. We should be able to link Zephyr + SOF module to the Cadence archive with a regular Cmake library inclusion.

bawanqin1987, 2022-02-21:
yes we are, my platform is csk6 (arm m0+hifi4 dsp lx7),we had already portting the xos or rtos with processor,but now,we wanna portting zephyr and sof with my processor arch
from now on, our processor with zephyr os ,include sof firmware running ok,sample ap sof alsa topology testbech (such as playback without volume ,adc record and so on ...)running ok
and then we will adaptor the cadence encode/decode on sof with zephyr os.

lenghonglin, 2022-02-28:
I use [sof-imx8mp-compr-mp3-wm8960](https://github.com/thesofproject/sof/blob/main/tools/topology/topology1/development/sof-imx8mp-compr-mp3-wm8960.m4) this tplg file , but it show error 

[image]

dbaluta, 2022-02-28:
@dbaluta this CONFIG option must be set on SOF firmware side:

See: https://github.com/thesofproject/sof/blob/main/src/audio/Kconfig#L441

Besides this depending on what codec you want to enable you will also need to look inside:

https://github.com/thesofproject/sof/blob/main/src/audio/codec_adapter/Kconfig

lenghonglin, 2022-02-28:
> this CONFIG option must be set on SOF firmware side:
>
> See: https://github.com/thesofproject/sof/blob/main/src/audio/Kconfig#L441
>
> Besides this depending on what codec you want to enable you will [...]

Yes, but this is sof side , i wanna know how to let host send CODEC_ADAPTOR  ipc cmd to dsp. i don't find this part code.

dbaluta, 2022-02-28:
The command to create the CODEC_ADAPTER component is already sent, as you see in your screenshot. And it fails.

What I think you want on the kernel side is to have the SND_SOC_SOF_COMPRESS config option enabled so that you can use the compress interface.

This needs that you must have this Compress patch in your tree: https://lkml.org/lkml/2022/1/20/527

One question: what decoder do you want to use: compress one (like MP3) or uncrompressed one like for example (Dolby Audio Processing?).

For DAP there is no need for compress interface as it is PCM based stream.

lenghonglin, 2022-02-28:
> The command to create the CODEC_ADAPTER component is already sent, as you see in your screenshot. And it fails.
> 
> What I think you want on the kernel side is to have the SND_SOC_SOF_COMPRESS config option enabled so that you can use the compress interface.

> This needs that you must have this Compress patch in your tree: https://lkml.org/lkml/2022/1/20/527

Thanks , I try to find SND_SOC_SOF_COMPRESS this part code.
 
> One question: what decoder do you want to use: compress one (like MP3) or uncrompressed one like for example (Dolby Audio Processing?).

Yes, MP3

dbaluta, 2022-02-28:
> Why don't use `SOF_COMP_CODEC_ADAPTOR ` like create [SOF_COMP_DAI](https://github.com/torvalds/linux/blob/871bfa02d08d9c0ed981c50082b7afd367d3700b/sound/soc/sof/topology.c#L1489)

I don't know how to answer this question. Could be that DAI is a special component.

lenghonglin, 2022-02-28:
> > Why don't use `SOF_COMP_CODEC_ADAPTOR ` like create [SOF_COMP_DAI](https://github.com/torvalds/linux/blob/871bfa02d08d9c0ed981c50082b7afd367d3700b/sound/soc/sof/topology.c#L1489)
> 
> I don't know how to answer this question. Could be that DAI is a special component.

Thanks a bundle

lgirdwood, 2022-02-28:
@lenghonglin all the components are being aligned along a single interface being done by @ranj063 and @jxstelter. This is WIP but will converge at some point before SOF 2.4 (Q4 2022)

lenghonglin, 2022-03-17:
@dbaluta @lgirdwood  hello， i can load mp3 decode lib in sof now.  

now i have a question, i see https://github.com/thesofproject/sof/blob/bcccf1253d8b52bfdfc2ba1c4b178d747ee0cf79/src/audio/codec_adapter/codec/cadence.c#L105 this link setup `DEFAULT_CODEC_ID`,   how can i change codec_id in runtime ?

i wanna let codec_adapter support both mp3 decode and aac decodec.

paulstelian97, 2022-03-17:
@lenghonglin That's a good point, with the current state of affairs the codec ID cannot be overwritten. It is a task actively being worked on (since MP3 doesn't work with the current code without changing the default). See #5245 .

lenghonglin, 2022-03-17:
> @lenghonglin That's a good point, with the current state of affairs the codec ID cannot be overwritten. It is a task actively being worked on (since MP3 doesn't work with the current code without changing the default). See #5245 .

So there two way to support both mp3 and aac decode.

1. creat  mp3 decode pipeline and aac decode pipeline.   but this maybe coumse more ram.
2. dynamically setup codec_id.  The codec_id will be sent by the Linux kernel running on Host. like https://github.com/thesofproject/sof/pull/5245 .
Right ?

lenghonglin, 2022-03-17:
> @lenghonglin That's a good point, with the current state of affairs the codec ID cannot be overwritten. It is a task actively being worked on (since MP3 doesn't work with the current code without changing the default). See #5245 .

and It seem this  #5245 is not continuing

dbaluta, 2022-03-17:
> * creat  mp3 decode pipeline and aac decode pipeline.   but this maybe coumse more ram.
>
This is no longer true with latest sources of SOF as we removed the codec_id from topology.
 
> * dynamically setup codec_id.  The codec_id will be sent by the Linux kernel running on Host. like

This is the way we are working now to upstream. Actually, I got back to look at this. Hope to have a solution in 1 week.

For now, just for testing you can replace at compile time DEFAULT_CODEC_ID with MP3 so that you can test it.

lenghonglin, 2022-03-17:
> This is no longer true with latest sources of SOF as we removed the codec_id from topology.
>
>
> This is the way we are working now to upstream. Actually, I got back to look at this. Hope to have a [...]

Thanks for you reply.  `DEFAULT_CODEC_ID with MP3 so that you can test it.` 
 Yes, i am dothing like this , and it work fine now.

` I got back to look at this. Hope to have a solution in 1 week.`  

What can i do for you? I've been needing that lately.

dbaluta, 2022-03-17:
What kernel are you using? What SOF version are you using.  

Not sure if you can help, unless you have some ideas on how can we dinamically set the codec id as discussed in https://github.com/thesofproject/sof/pull/5245

You can help me test this change applied on SOF main, with MP3 and tell me if it still works. Thus we can spare me some cycles.

https://github.com/thesofproject/sof/pull/5559

lenghonglin, 2022-03-17:
> What kernel are you using? What SOF version are you using.

The sof version is v1.5, but i update `codec_apater` part files to latest.   

> Not sure if you can help, unless you have some ideas on how can we dinamically set the codec id as discussed in #5245

OK

> You can help me test this change applied on SOF main, with MP3 and tell me if it still works. Thus we can spare me some cycles.
> 
> #5559

OK, i will test later.