thesofproject/sof #5414: [BUG]xthal_dcache_region_invalidate causes ram data to be modified

lenghonglin, 2022-02-24:
**Describe the bug**

[image]

[image]

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)

**Reproduction Rate**

almost 100%

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

fuyuntsuo, 2022-02-25:
Are these screen captures part of Xtensa Xplorer? Can we run SOF fw on that?

lenghonglin, 2022-02-25:
> Can we run SOF fw on that?

Xtensa Xplorer,  we can run SOF fw

dbaluta, 2022-02-25:
I honestly do not understand the problem, but let me give you some pointers. At a first glance, after you invalidate the cache you re-read the data from RAM (but data from RAM could have been updated by the DMA, so you get different values from what was in the cache).

Next part: data alignment comes from your DMA driver.

You can have a look at `src/audio/host.c`:

host_params:
-> dma_get_attribute(hd->dma, DMA_ATTR_BUFFER_ADDRESS_ALIGNMENT, &addr_align);

Also, how are you able to see the address contens in real time using Xtensa Explorer? Are you using a debugger?

lenghonglin, 2022-02-25:
> I honestly do not understand the problem, but let me give you some pointers. At a first glance, after you invalidate the cache you re-read the data from RAM (but data from RAM could have been updated by the DMA, so you get different values from what was in the cache).
> 
> Next part: data alignment comes from your DMA driver.
> 
> You can have a look at `src/audio/host.c`:
> 
> host_params: -> dma_get_attribute(hd->dma, DMA_ATTR_BUFFER_ADDRESS_ALIGNMENT, &addr_align);
> 
> Also, how are you able to see the address contens in real time using Xtensa Explorer? Are you using a debugger?
> 
> Cc: @paulstelian97

Thanks , i use dummy-dma in host

Yes , i use jlink to debug hifi4.

paulstelian97, 2022-02-25:
This looks like something unrelated is interacting with the sg elem (did some interrupt happen in the meantime?). Indeed this doesn't look like normal behaviour.

However do note that after copying a segment the driver is supposed to update and change to a different sg elem. Remember that elems is a full blown array, not just a simple pointer (you're only showing elems[0] in that debug screen). Also, while in the middle of the copy an elem isn't supposed to change once the copy is done it typically does change.

This does need further analysis.

lgirdwood, 2022-02-25:
Sorry, not following this. @lenghonglin  are you saying that the data pointed to by rptr is being modified or that rptr is being modified ?

I would expect the rptr->data to be modified after invalidate if your local cache copy is not coherent with the data in RAM (e.g. host or DMA has copied data to rptr->data).

closed 2022-02-28 by lenghonglin

fuyuntsuo, 2022-03-01:
Hi @lenghonglin sorry to interrupt. Just wondering how to run SOF fw on Xtensa Xplorer?

Is it a pure emulation? Or kind of debug board overlooking a real system?

lenghonglin, 2022-03-01:
> 

We reserved the JTAG debugging interface of DSP

lenghonglin, 2022-03-01:
> Hi @lenghonglin sorry to interrupt. Just wondering how to run SOF fw on Xtensa Xplorer?
> 
> Is it a pure emulation? Or kind of debug board overlooking a real system?

We reserved the JTAG debugging interface of DSP