thesofproject/sof #5422: [FEATURE] Variable buffer sizes in pipeline

sanzazacuro, 2022-02-24:
**Is your feature request related to a problem? Please describe.**
Is it possible to have different buffer sizes for a pipeline? For instance, a component in a pipeline can have 48 samples input and 500 samples output sent to host?

I tried creating variable widget buffer configs for a pipeline, but the actual frame size in component did not change:
```
pseudo:
NUM_SAMPLES = 500
W_BUFFER(2, NUM_SAMPLES ,PLATFORM_HOST_MEM_CAP)
```
[image]

sanzazacuro, 2022-02-24:
How would ALSA (arecord) react to this?

dbaluta, 2022-02-24:
What worked for me with codec_adapter component (like your custom component is to have some internal deep buffers and thus mitigate having some fixed input / output buffers.

But your question is valid. I think you can setup the buffers sizes from topology files.

lgirdwood, 2022-02-28:
Ack, it is possible to set any sizes for each stage in the pipeline topology. The main thing is to keep the buffer small for latency and large enough for headroom.

closed 2022-03-16 by sanzazacuro