thesofproject/sof #5424: Zephyr LL scheduler support for SOF_SCHEDULE_LL_DMA

dbaluta, 2022-02-24:
For now, Zephyr LL scheduler does not support SOF_SCHEDULE_LL_DMA domain type.

src/schedule/zephyr_ll.c:

```
int zephyr_ll_scheduler_init(struct ll_schedule_domain *domain)
{
        if (domain->type != SOF_SCHEDULE_LL_TIMER) {                                                                                                                                                               
                tr_err(&ll_tr, "zephyr_ll_scheduler_init(): unsupported domain %u",
                      domain->type);
                return -EINVAL;
        }
```

On i.MX all our pipelines use SOF_SCHEDULE_LL_DMA so when switching to zephyr obviously we cannot use Zephyr LL scheduler type.

Is there a plan to support SOF_SCHEDULE_LL_DMA domain type with Zephyr LL scheduler? I've seen that there are topologies with Intel also using this scheduler domain type (e.g Baytrail).

What would you advise us:
- (1) implement LL_DMA with Zephyr LL.
- (2) switch all i.MX topologies to use LL_TIMER?

For now just switching to LL_TIMER on i.MX causes bad sound with an eventual I/O error on ALSA kernel side.

dbaluta, 2022-02-28:
@lgirdwood any idea if the LL Zephyr scheduler will support LL_DMA. What's the direction on this? All our pipelines use LL_DMA domain instead of LL_TIMER and we understand if you switch them to timer or we could implement LL_DMA in LL Zephyr scheduler.

lgirdwood, 2022-02-28:
@dbaluta Fwiw, Zephyr gives some scheduling advantages over xtos mode (preemption based on priority and not IRQ level) that may mean you could simplify here if you wanted ? I would be tempted to try TIMER on NXP at first and then make a decision ?

dbaluta, 2022-02-28:
@lgirdwood we tried the timer stuff but for some reason the sound is distorted and after a while we get the famous I/O error on aplay. 

I will try to make it work with TIMER then.

lgirdwood, 2022-02-28:
> @lgirdwood we tried the timer stuff but for some reason the sound is distorted and after a while we get the famous I/O error on aplay.
> 
> I will try to make it work with TIMER then.

I guess the codec is clock supplier ? One thing to check is that there is overhead in the DAI DMA buffers to cater for drift (Intel uses 3 period size == 3ms) for DAI DMA buffers for drift and to avoid DMA/FW IO collisions on each period.