thesofproject/sof #5429: No audio on GLK Chromebook

J0shuaT, 2022-02-25:
I don't have any audio on my GLK Chromebook. Audio settings shows a dummy output. I saw other forum posts with similar issues, but I couldn't find a definite solution. Any help would be appreciated. Thanks!

**Here are my System specs:**
[image]

**This is the output of `aplay -l`:**

`**** List of PLAYBACK Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 9: HDMI 3 [HDMI 3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 10: HDMI 4 [HDMI 4]
  Subdevices: 1/1
  Subdevice #0: subdevice #0`

**This is the `dmesg` output:** (too long to put here)

https://pastebin.com/LEftsN6R

lgirdwood, 2022-02-25:
It looks like the driver is loading and detecting the DSP but not going any further afaict.
```
[    4.124341] snd_hda_intel 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if info 0x040100
[    4.125309] snd_hda_intel 0000:00:0e.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
```
@plbossart @kv2019i @ujfalusi any ideas ?

kv2019i, 2022-02-25:
This seems like a product with I2S audio codecs not supported in upstream (either machine driver or codec driver missing). Maybe something similar to these: https://github.com/thesofproject/linux/issues/3336

J0shuaT, 2022-02-25:
I tried the solutions described in https://github.com/thesofproject/linux/issues/3336 , but I still don't have audio :(

kv2019i, 2022-02-25:
@J0shuaT Can you share output of "alsa-info.sh" (see https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html#run-alsa-info) and "sudo dmi-decode".  
And third, the NHLT table:
sudo acpidump > dump.bin
gzip dump.bin

These should reveal what kind of audio hardware this system has. If it's a Chromebook, the codec and machines drivers should be upstream. But it's also possible this is hardware not supported by SOF. Anyways, getting hw info will help to tell which case this is.

J0shuaT, 2022-02-25:
**Here is the output of `alsa-info`:** http://alsa-project.org/db/?f=4ffbda9963feb81854cc1af925a91d7e5f46e9f3

**This is the output of `sudo dmi-decode`:** 

`# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.
Table at 0x0FFE5000.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: coreboot
	Version: MrChromebox-4.15
	Release Date: 01/18/2022
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		PC Card (PCMCIA) is supported
		BIOS is upgradeable
		Selectable boot is supported
		ACPI is supported
		Targeted content distribution is supported
	BIOS Revision: 4.15
	Firmware Revision: 0.0

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: HP
	Product Name: Blooglet
	Version: rev4
	Serial Number: Not Specified
	UUID: Not Settable
	Wake-up Type: Reserved
	SKU Number: sku68
	Family: Google_Octopus

Handle 0x0002, DMI type 2, 14 bytes
Base Board Information
	Manufacturer: HP
	Product Name: Blooglet
	Version: rev4
	Serial Number: Not Specified
	Asset Tag: Not Specified
	Features: None
	Location In Chassis: Not Specified
	Chassis Handle: 0x0003
	Type: Motherboard

Handle 0x0003, DMI type 3, 22 bytes
Chassis Information
	Manufacturer: HP
	Type: Laptop
	Lock: Not Present
	Version: Not Specified
	Serial Number: Not Specified
	Asset Tag: Not Specified
	Boot-up State: Safe
	Power Supply State: Safe
	Thermal State: Safe
	Security Status: None
	OEM Information: 0x00000000
	Height: Unspecified
	Number Of Power Cords: 1
	Contained Elements: 0
	SKU Number: Not Specified

Handle 0x0004, DMI type 4, 48 bytes
Processor Information
	Socket Designation: CPU0
	Type: Central Processor
[... 258 lines cut ...]

Handle 0x0012, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

Handle 0x0013, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: Display controller
	Type: Video
	Status: Enabled
	Type Instance: 0
	Bus Address: 0000:00:02.0

Handle 0x0014, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: Multimedia audio controller
	Type: Sound
	Status: Enabled
	Type Instance: 0
	Bus Address: 0000:00:0e.0

Handle 0xFEFF, DMI type 127, 4 bytes
End Of Table`

**This is dump.bin:** [dump.bin.gz](https://github.com/thesofproject/sof/files/8143882/dump.bin.gz)

plbossart, 2022-02-25:
The solutions suggested in #3336 would not work, it's not the same hardware platform. 

You seem to have a bog-standard GLK Chromebook, but you are using a Linux distro with @MrChromebox's firmware.

Please add this file 
[sof-dyndbg.conf.txt](https://github.com/thesofproject/sof/files/8145043/sof-dyndbg.conf.txt)
as /etc/modprobe.d/sof-dyndbg.conf, reboot and attach the dmesg log. Thanks @J0shuaT

MrChromebox, 2022-02-25:
@plbossart there are a few different audio configs for GLK Chromebooks - the ones with rt5682+max98357a work fine, the ones with da7219+max98357a (as above) do not currently

J0shuaT, 2022-02-25:
@plbossart Here is it: https://pastebin.com/Fr2siN4c

plbossart, 2022-03-01:
@MrChromebox I have a "Bobba" GLK device with da7219, works fine outside of the jack detection problem due to UCM.

are you sure it's not the other way around, that the ones with rt5682 are broken?

kv2019i, 2022-03-01:
Kernel driver bug, moving to the appropriate tracker.