thesofproject/sof #5437: [BUG] Sound breaks after a few minutes playing - Dell Chromebook 11 3120 - baytrail max98090 in SOF mode

sourcejedi, 2022-02-26:
**Describe the bug**

Sound playback works initially. After a few minutes playback, sound is replaced with a flat tone.  The driver stops accepting more audio, as can be seen by youtube video stalling.

Sound can always be cleared and restarted simply by switching consoles e.g. ctrl+alt+f1 and back. (This causes pulseaudio to relinquish the device, and then reacquire it). But it will soon break again.

If and only if you pause the video, the tone also goes away in about 5 seconds. Then, you can unpause it and it works again.

If you instead leave it and wait about 3 minutes (?), it seems to recover by itself.

Sound worked on some previous ubuntu kernels, needing only an upgraded alsa-ucm package from debian unstable (which I still have installed).  However, ubuntu was not originally using SOF mode.

**Reproduction Rate**
Reproduces reliably, within five or ten minutes of playback starting.

**Expected behavior**
Able to play an hour-long youtube video and listen to the audio throughout, without interruption.

**Impact**
Effectively the audio is not useful, at least for my purposes.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 49484d818f397764428c665cbf6149be6256ad8c
       i.e. ubuntu mainline kernel 5.16.11-051611-generic.
       (also reproducible with ubuntu non-mainline kernel 5.13.0-30-generic)
    * SOF: v1.8 from sof-bin github
       (Ubuntu's version of SOF did not have the firmware, see #4160).
2) Name of the topology file
    * Topology: sof-byt-max98090.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: "BYT", I guess? Based on the above topology.

See also alsa-info.sh: http://alsa-project.org/db/?f=722800cdf39e22d487a9d0f8a55490d53d1f5f75

**Screenshots or console output**
For ALSA/HDA messages, see alsa-info above.

When this problem happens, it does not cause any error messages in `dmesg` (kernel) or in `journalctl -b` (all, including pulseaudio).

> Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

I can enable debugging as suggested. This is an example where the error happens, then I left it humming for 3 minutes, and it recovered itself.

```
Feb 26 13:13:42 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: pcm: trigger stream 0 dir 0 cmd 0
Feb 26 13:13:42 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc tx: 0x60050000
Feb 26 13:13:42 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc tx succeeded: 0x60050000
Feb 26 13:13:42 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc tx: 0x60030000
Feb 26 13:13:42 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc tx succeeded: 0x60030000
Feb 26 13:13:42 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: pcm: free stream 0 dir 0
Feb 26 13:13:42 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: pcm: close stream 0 dir 0
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: pcm: open stream 0 dir 0
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: period min 192 max 16384 bytes
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: period count 2 max 16
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: buffer max 65536 bytes
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: DAI config 0 matches pcm hw params
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: DAI config 0 matches pcm hw params
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: rate_min: 48000 rate_max: 48000
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: channels_min: 2 channels_max: 2
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: DAI config 0 matches pcm hw params
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: DAI config 0 matches pcm hw params
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: rate_min: 48000 rate_max: 48000
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: channels_min: 2 channels_max: 2
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: pcm: hw params stream 0 dir 0
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: generating page table for 00000000ee045878 size 0x4af4 pages 5
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: stream_tag 0
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc tx: 0x60010000
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc tx succeeded: 0x60010000
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: pcm: stream dir 0, posn mailbox offset is 1329740
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: pcm: trigger stream 0 dir 0 cmd 1
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc tx: 0x60040000
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc tx succeeded: 0x60040000
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc rx: 0x600a0011
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc rx done: 0x600a0011
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc rx: 0x600a0011
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc rx done: 0x600a0011
Feb 26 13:13:58 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc rx: 0x600a0011
```

<snip 5 minutes of continued playback>

```
Feb 26 13:18:43 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc rx: 0x600a0011
Feb 26 13:18:43 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc rx done: 0x600a0011
Feb 26 13:18:43 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc rx: 0x600a0011
Feb 26 13:18:43 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc rx done: 0x600a0011
Feb 26 13:18:43 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc rx: 0x600a0011
Feb 26 13:18:43 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc rx done: 0x600a0011
Feb 26 13:21:08 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc rx: 0x600a0011
Feb 26 13:21:08 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc rx done: 0x600a0011
Feb 26 13:21:08 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc rx: 0x600a0011
Feb 26 13:21:08 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc rx done: 0x600a0011
Feb 26 13:21:08 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc rx: 0x600a0011
Feb 26 13:21:08 candy kernel: sof-audio-acpi-intel-byt 80860F28:00: ipc rx done: 0x600a0011
```

sof-logger isn't telling me anything. perhaps I'm using it wrong.

```
$ sudo sof-logger -l sof-byt.ldc
 TIMESTAMP         (us)              DELTA  C# COMPONENT          LOCATION                      CONTENT	ktime=1100.200s  @  2022-02-26 13:25:44 +0000 GMT
Skipped 880 bytes after the last statement.
```

plbossart, 2022-03-01:
@sourcejedi this seems to be the same DMA issue we're having on BYT.

https://github.com/thesofproject/sof/issues/3868

Please add this to /etc/modprobe.d/alsa-base.conf
options snd_sof sof_debug=1

reboot and retry. Thanks!

sourcejedi, 2022-03-04:
Oh, thank you! Confirmed. Enabling `sof_debug` in `sof_snd` appears to stop the bug.

(And that option doesn't seem to flood the kernel log with debugging messages either).

plbossart, 2022-03-04:
> Oh, thank you! Confirmed. Enabling `sof_debug` in `sof_snd` appears to stop the bug.

Great @sourcejedi, thanks for letting us know. May the force be with you.
 
> (And that option doesn't seem to flood the kernel log with debugging messages either).

this enables the firmware trace which goes in a different ring buffer, so there are no user-visible changes in dmesg.

Danct12, 2022-03-23:
Currently using SOF on the Z8350. This is a different platform (Cherry Trail), but the same problem.

However, enabling `sof_debug` doesn't seem to stop the bug for me, but the bug happens very randomly and I had to restart Pipewire for audio to work again.

mengdonglin, 2022-06-21:
Move BYT platform to maintenance mode and SOF v2.2 will be the last upstream release with support for this platform. There will be no more bug fixing effort on this platform.

closed 2022-06-21 by mengdonglin