thesofproject/sof #5438: [BUG] No audio output on topology on HP Laptop 17-cn0xxx

ZacharyTheGamer200, 2022-02-26:
**Describe the bug**
On my HP Laptop 17-cn0xxx the audio does not work on Linux using the latest Linux Firmware release. The audio worked in a prior release in August and as recently as a few months ago but that driver may've been highly unstable because back then the system would completely freeze to the extent that no logs were recorded I don't know what the cause was. The current Linux-firmware release does not cause this freezing issue but no audio output is produced as no soundcards are detected.
What have you tried to diagnose or workaround this issue?
I've run speaker tests at the hardware level and determined that the issue was that the soundcard wasn't being detected.

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)
Try to play any audio on an HP Laptop 17-cn0xxx laptop.
**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
Does the reproduction rate vary with any other configuration or user action, if so please describe and show the new reproduction rate.
All the time.
**Expected behavior**
A clear and concise description of what you expected to happen.
I expected to hear audio from the browser or, at least, the speaker tests and none happened.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper) showstopper. I have to use my Windows partition until this is fixed.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
I don't know how to find a commit hash, so I'm putting everything I know here.
    * Kernel: {Mainline}
    * SOF: {Early February release, linux-firmware 20220209.6342082-1}
2) Name of the topology file
I don't know what a topology file is.
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {Arch Linux}

If anyone can explain what the stuff I don't know is or how I can provide any further information please tell me.

lgirdwood, 2022-02-28:
@ZacharyTheGamer200 can you provide your dmesg output at boot and then any dmesg output after playback with aplay.

ZacharyTheGamer200, 2022-02-28:
Here's the output immediately after boot after typing sudo dmesg: https://termbin.com/au1m
For aplay output I downloaded some random dummy wav file from here https://file-examples.com/index.php/sample-audio-files/sample-wav-download/. After playing that the dmesg output is as follows: https://termbin.com/r5v8

lgirdwood, 2022-02-28:
@ZacharyTheGamer200 it looks like we are missing a component driver (amey the codec) see.
```
[    1.277117] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    1.277138] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    1.319345] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    1.319363] sof-audio-pci-intel-tgl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    1.319377] sof-audio-pci-intel-tgl 0000:00:1f.3: enabling device (0000 -> 0002)
[    1.319544] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    1.319594] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    1.324307] Userspace governor deprecated: use thermal netlink notification instead
[    1.325888] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[    1.336112] sof-audio-pci-intel-tgl 0000:00:1f.3: codec #0 probe error, ret: -2
[    1.337824] sof-audio-pci-intel-tgl 0000:00:1f.3: codec #2 probe error, ret: -2
[    1.338018] sof-audio-pci-intel-tgl 0000:00:1f.3: no hda codecs found!
[    1.338019] sof-audio-pci-intel-tgl 0000:00:1f.3: warning: No matching ASoC machine driver found
[    1.338019] sof-audio-pci-intel-tgl 0000:00:1f.3: error: no matching ASoC machine driver found - aborting probe
[    1.338020] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to get machine info -19
[    1.338135] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sof_probe_work failed err: -19
```
The HDA codec is not probing above. @plbossart @kv2019i @ujfalusi fyi.

ZacharyTheGamer200, 2022-02-28:
I have no experience in firmware programming so I'm curious what would a fix for this issue look like?

ZacharyTheGamer200, 2022-02-28:
Are there any other commands I could execute or tests I could run to diagnose the exact cause of the HDA codec not being probed? (namely, whether the issue is the drivers or the integration.)

ujfalusi, 2022-02-28:
@ZacharyTheGamer200, can you enable the dynamic debugging [1] and share the dmesg?

[1] https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html?highlight=dyndbg#enable-dynamic-debug

ujfalusi, 2022-02-28:
@ZacharyTheGamer200, you mentioned that audio worked before on the laptop. Can you roll back `sof-firmware` or the `linux` package to see which might have caused the regression?

There were a brief regression with 5.16, but it got resolved with 5.16.3.

ZacharyTheGamer200, 2022-02-28:
To ensure I have all the correct options enabled here's a printout of my /etc/modprobe.d/sof-dyndbg.conf: https://termbin.com/c3qp
Here's the dmesg before attempting to play any audio:https://termbin.com/2quq
After attempting to play some audio from youtube: https://termbin.com/89mh
I'm play around with downgrading sof-firmware and will report back.

ZacharyTheGamer200, 2022-02-28:
I tried downgrading to sof-firmware 1.9.1 which I think I used on the prior install but that didn't fix it. I'm going to try downgrading to Linux 5.16.3 now.

ujfalusi, 2022-02-28:
@ZacharyTheGamer200, do you sound card at all? what does `cat /proc/asound/cards` show?
Can you share the output of `alsa-info.sh` as well? Also the output of `lspci`.

The audio probe aborts because probing of the two HDA codec fails and because of that we can not find the correct machine driver. Interestingly the debugprints from `snd-sof-intel-hda-common` is still missing.

Is this laptop have NVIDIA gpu?

plbossart, 2022-03-01:
@ZacharyTheGamer200 if you have HDAudio issues, you first need to try the HDaudio legacy driver. see https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html

please add this to /etc/modprobe.d/alsa-base.conf

options snd-intel-dspcfg dsp_driver=1

reboot and report the log of 'alsa-info'

Thanks!

ujfalusi, 2022-03-01:
@ZacharyTheGamer200, for the alsa-info.sh error: `sudo pacman -S tree`

ZacharyTheGamer200, 2022-03-01:
So you can verify I edited the alsa-base.conf file correctly: https://termbin.com/6t4hm
Output of alsa-info after doing this edit and rebooting: https://termbin.com/mm2l 
The output of the file alsa-info produced: https://termbin.com/lm80
Thanks @ujfalusi for helping me resolve the error.

ujfalusi, 2022-03-01:
> So you can verify I edited the alsa-base.conf file correctly: https://termbin.com/6t4hm Output of alsa-info after doing this edit and rebooting: https://termbin.com/mm2l The output of the file alsa-info produced: https://termbin.com/lm80 Thanks @ujfalusi for helping me resolve the error.

@ZacharyTheGamer200, can you also share the content of the alsa-info report?
"Your ALSA information is in `/tmp/alsa-info.txt.some_random_characters`"
`cat /tmp/alsa-info.txt.some_random_characters`
It was `/tmp/alsa-info.txt.0KpwuC9uPU` in the latest log you have shared.

Thanks

ujfalusi, 2022-03-01:
> @ZacharyTheGamer200, can you also share the content of the alsa-info report? "Your ALSA information is in `/tmp/alsa-info.txt.some_random_characters`" `cat /tmp/alsa-info.txt.some_random_characters` [...]

Nevermind, the third text is the content.

ujfalusi, 2022-03-01:
The two HDA code is 10ec0236 (Realtek ALC3204) and 80862812 (HDMI)
In your logs:
```
[    1.534367] sof-audio-pci-intel-tgl 0000:00:1f.3: HDA codec #0 probed OK: response: 10ec0236
[    1.535186] hdaudio ehdaudio0D0: loading codec module: hdaudio:v10EC0236r00100002a01

[    1.535869] hdaudio ehdaudio0D0: loading codec module: snd-hda-codec-generic
[    1.536513] sof-audio-pci-intel-tgl 0000:00:1f.3: codec #0 probe error, ret: -2
[    1.536619] sof-audio-pci-intel-tgl 0000:00:1f.3: HDA codec #2 probed OK: response: 80862812
[    1.537317] hdaudio ehdaudio0D2: loading codec module: hdaudio:v80862812r00100000a01

[    1.538095] hdaudio ehdaudio0D2: loading codec module: snd-hda-codec-generic
[    1.538624] sof-audio-pci-intel-tgl 0000:00:1f.3: codec #2 probe error, ret: -2
[    1.538758] sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state change: 0 -> 1
[    1.538892] sof-audio-pci-intel-tgl 0000:00:1f.3: no hda codecs found!
```
It is still probing the SOF driver instead of the legacy HDA.

I do have a laptop with the same codec (it is a Dell laptop) it probes just fine (but I'm forcing the SOF driver):
```
[  181.193077] sof-audio-pci-intel-tgl 0000:00:1f.3: HDA codec #0 probed OK: response: 10ec0236
[  181.194144] hdaudio ehdaudio0D0: loading codec module: hdaudio:v10EC0236r00100002a01

[  181.202237] sof-audio-pci-intel-tgl 0000:00:1f.3: HDA codec #2 probed OK: response: 80862812
[  181.203017] hdaudio ehdaudio0D2: loading codec module: hdaudio:v80862812r00100000a01
```

I think the reason is that intel-dspcfg is in both alsa-base.conf and in sof-dyndbg.conf. Sorry about the hassle, but can you:
`rm /etc/modprobe.d/alsa-base.conf`
and edit the `/etc/modprobe.d/sof-dyndbg.conf` file, add the driver selection to the existing line, so it will look like this:
`options snd_intel_dspcfg dyndbg=+p dsp_driver=1`

In addition, can you enabled the dynamic debugging fort he HDA drivers to get some hints from there:
```
options snd-hda-codec-realtek dyndbg=+p
options snd-hda-codec-generic dyndbg=+p
options snd-hda-codec-hdmi dyndbg=+p
options snd-hda-intel dyndbg=+p
options snd-hda-codec dyndbg=+p
```
And reboot.

ujfalusi, 2022-03-01:
@ZacharyTheGamer200, fwiw the distro kernel I'm using is `5.16.10-artix1-1` (Artix == Arch w/o systemd)

kv2019i, 2022-03-01:
HDA codecs not probing is a driver issue, moving.