thesofproject/sof #5439: Hp Chromebook elite c1030 no audio rt5682 SOF

psammarco, 2022-02-26:
**Describe the issue**
I have got a HP chromebook elite c1030 which features a 10th gent Intel Cometlake i7-10610U chip and a rt5682 soundcard. I have been hammering my head for over a week now trying to get audio to work, but so far I have had no luck whatsoever.
I enabled all available SND and SOF related kernel modules selected in the default ChromeOS kernel config, but it made little to no difference. The soundcard is not recognized and I have exhausted all ideas. I am not sure whether I am missing any step or if I am doing it wrong.

**How the soundcard is seen in ChromeOS**
```
// lspci
00:1f.3 Multimedia audio controller: Intel Corporation Device 02c8
	DeviceName: Multimedia audio controller
	Subsystem: Intel Corporation Device 7270
	Flags: bus master, fast devsel, latency 64, IRQ 150
	Kernel driver in use: sof-audio-pci
	Kernel modules: snd_hda_intel, snd_sof_pci

// dmesg
[    4.613556] sof_rt5682 sof_rt5682: info: override FE DAI link SSP0-Codec
[    4.613560] sof_rt5682 sof_rt5682: info: override FE DAI link dmic01
[    4.613562] sof_rt5682 sof_rt5682: info: override FE DAI link dmic16k
[    4.613563] sof_rt5682 sof_rt5682: info: override FE DAI link iDisp1
[    4.613564] sof_rt5682 sof_rt5682: info: override FE DAI link iDisp2
[    4.613566] sof_rt5682 sof_rt5682: info: override FE DAI link iDisp3
[    4.613567] sof_rt5682 sof_rt5682: info: override FE DAI link SSP1-Codec
[    4.614155] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:12:0 Kernel ABI 3:11:0
[    4.614159] sof-audio-pci 0000:00:1f.3: warn: topology ABI is more recent than kernel
[    4.628675] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name DETECT SINK 9 not handled
[    4.637724] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[    4.649642] sof_rt5682 sof_rt5682: Linked as a consumer to ehdaudio0D2
[    4.653193] sof_rt5682 sof_rt5682: rt5682-aif1 <-> SSP0 Pin mapping ok
[    4.653214] sof_rt5682 sof_rt5682: dmic-hifi <-> DMIC01 Pin mapping ok
[    4.653221] sof_rt5682 sof_rt5682: dmic-hifi <-> DMIC16k Pin mapping ok
[    4.653231] sof_rt5682 sof_rt5682: intel-hdmi-hifi1 <-> iDisp1 Pin mapping ok
[    4.653238] sof_rt5682 sof_rt5682: intel-hdmi-hifi2 <-> iDisp2 Pin mapping ok
[    4.653245] sof_rt5682 sof_rt5682: intel-hdmi-hifi3 <-> iDisp3 Pin mapping ok
[    4.653255] sof_rt5682 sof_rt5682: HiFi <-> SSP1 Pin mapping ok
[    4.653284] sof_rt5682 sof_rt5682: snd-soc-dummy-dai <-> DMIC16k 8 mapping ok
[    4.653304] sof_rt5682 sof_rt5682: snd-soc-dummy-dai <-> Port1 0 mapping ok
[    4.653319] sof_rt5682 sof_rt5682: snd-soc-dummy-dai <-> DMIC01 1 mapping ok
[    4.653334] sof_rt5682 sof_rt5682: snd-soc-dummy-dai <-> HDMI1 2 mapping ok
[    4.653348] sof_rt5682 sof_rt5682: snd-soc-dummy-dai <-> HDMI2 3 mapping ok
[    4.653366] sof_rt5682 sof_rt5682: snd-soc-dummy-dai <-> HDMI3 4 mapping ok
[    4.653381] sof_rt5682 sof_rt5682: snd-soc-dummy-dai <-> Speakers 5 mapping ok
[    4.654070] input: sof-sof_rt5682 Headset Jack as /devices/pci0000:00/0000:00:1f.3/sof_rt5682/sound/card0/input8
[    4.654134] input: sof-sof_rt5682 HDMI/DP, pcm=2 Jack as /devices/pci0000:00/0000:00:1f.3/sof_rt5682/sound/card0/input9
[    4.654170] input: sof-sof_rt5682 HDMI/DP, pcm=3 Jack as /devices/pci0000:00/0000:00:1f.3/sof_rt5682/sound/card0/input10
[    4.654209] input: sof-sof_rt5682 HDMI/DP, pcm=4 Jack as /devices/pci0000:00/0000:00:1f.3/sof_rt5682/sound/card0/input11
[ 1311.411478] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 2285.178766] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 3136.276805] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 3177.712465] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 3205.181909] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 3358.586906] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 3446.723222] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 3468.073716] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 3496.838068] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 3537.407335] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 3603.766275] sof_rt5682 sof_rt5682: calling platform_pm_suspend+0x0/0x49 @ 17040, parent: 0000:00:1f.3
[ 3603.766483] sof_rt5682 sof_rt5682: platform_pm_suspend+0x0/0x49 returned 0 after 197 usecs
[ 3603.784955] sof-audio-pci 0000:00:1f.3: calling pci_pm_suspend+0x0/0x1c3 @ 17040, parent: pci0000:00
[ 3603.915822] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 3603.959618] sof-audio-pci 0000:00:1f.3: pci_pm_suspend+0x0/0x1c3 returned 0 after 170559 usecs
[ 3604.101237] sof-audio-pci 0000:00:1f.3: calling pci_pm_suspend_late+0x0/0x36 @ 17040, parent: pci0000:00
[ 3604.101240] sof-audio-pci 0000:00:1f.3: pci_pm_suspend_late+0x0/0x36 returned 0 after 0 usecs
[ 3604.116367] sof-audio-pci 0000:00:1f.3: calling pci_pm_suspend_noirq+0x0/0x2a3 @ 17040, parent: pci0000:00
[ 3604.128268] sof-audio-pci 0000:00:1f.3: pci_pm_suspend_noirq+0x0/0x2a3 returned 0 after 11606 usecs
[ 3621.594794] sof-audio-pci 0000:00:1f.3: calling pci_pm_resume_noirq+0x0/0xe2 @ 17040, parent: pci0000:00
[ 3621.609605] sof-audio-pci 0000:00:1f.3: pci_pm_resume_noirq+0x0/0xe2 returned 0 after 14456 usecs
[ 3621.759904] sof-audio-pci 0000:00:1f.3: calling pci_pm_resume+0x0/0xc3 @ 17040, parent: pci0000:00
[ 3621.877266] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 3621.912489] sof-audio-pci 0000:00:1f.3: pci_pm_resume+0x0/0xc3 returned 0 after 148999 usecs
[ 3621.982764] sof_rt5682 sof_rt5682: calling platform_pm_resume+0x0/0x40 @ 17040, parent: 0000:00:1f.3
[ 3621.982775] sof_rt5682 sof_rt5682: platform_pm_resume+0x0/0x40 returned 0 after 6 usecs
[ 3955.378059] sof_rt5682 sof_rt5682: calling platform_pm_suspend+0x0/0x49 @ 18704, parent: 0000:00:1f.3
[ 3955.378260] sof_rt5682 sof_rt5682: platform_pm_suspend+0x0/0x49 returned 0 after 190 usecs
[ 3955.395952] sof-audio-pci 0000:00:1f.3: calling pci_pm_suspend+0x0/0x1c3 @ 18704, parent: pci0000:00
[ 3955.524189] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 3955.570542] sof-audio-pci 0000:00:1f.3: pci_pm_suspend+0x0/0x1c3 returned 0 after 170488 usecs
[ 3955.712076] sof-audio-pci 0000:00:1f.3: calling pci_pm_suspend_late+0x0/0x36 @ 18704, parent: pci0000:00
[ 3955.712078] sof-audio-pci 0000:00:1f.3: pci_pm_suspend_late+0x0/0x36 returned 0 after 0 usecs
[ 3955.726328] sof-audio-pci 0000:00:1f.3: calling pci_pm_suspend_noirq+0x0/0x2a3 @ 18704, parent: pci0000:00
[ 3955.738106] sof-audio-pci 0000:00:1f.3: pci_pm_suspend_noirq+0x0/0x2a3 returned 0 after 11492 usecs
[71709.657811] sof-audio-pci 0000:00:1f.3: calling pci_pm_resume_noirq+0x0/0xe2 @ 18704, parent: pci0000:00
[71709.672601] sof-audio-pci 0000:00:1f.3: pci_pm_resume_noirq+0x0/0xe2 returned 0 after 14435 usecs
[71709.823697] sof-audio-pci 0000:00:1f.3: calling pci_pm_resume+0x0/0xc3 @ 18704, parent: pci0000:00
[71709.933826] sof-audio-pci 0000:00:1f.3: firmware boot complete
[71709.972543] sof-audio-pci 0000:00:1f.3: pci_pm_resume+0x0/0xc3 returned 0 after 145346 usecs
[71710.042857] sof_rt5682 sof_rt5682: calling platform_pm_resume+0x0/0x40 @ 18704, parent: 0000:00:1f.3
[71710.042867] sof_rt5682 sof_rt5682: platform_pm_resume+0x0/0x40 returned 0 after 5 usecs
[71730.968157] sof-audio-pci 0000:00:1f.3: firmware boot complete
```

**Environment**
kernel: 5.16.11-mainline
Fedora 35
alsa-sof-firmware  1.9.3

_**ChromeOS logs**_
[chromeos-aplay.txt](https://github.com/thesofproject/sof/files/8147502/chromeos-aplay.txt)
[chromeos-dmesg.txt](https://github.com/thesofproject/sof/files/8147503/chromeos-dmesg.txt)
[chromeos-kern-config.txt](https://github.com/thesofproject/sof/files/8147504/chromeos-kern-config.txt)
[chromeos-lsmod.txt](https://github.com/thesofproject/sof/files/8147505/chromeos-lsmod.txt)
[chromeos-lspci.txt](https://github.com/thesofproject/sof/files/8147506/chromeos-lspci.txt)
_**Fedora logs**_
[aplay.txt](https://github.com/thesofproject/sof/files/8147516/aplay.txt)
[dmesg.txt](https://github.com/thesofproject/sof/files/8147517/dmesg.txt)
[kern_config.txt](https://github.com/thesofproject/sof/files/8147518/kern_config.txt)
[lsmod.txt](https://github.com/thesofproject/sof/files/8147519/lsmod.txt)
[lspci.txt](https://github.com/thesofproject/sof/files/8147520/lspci.txt)
http://alsa-project.org/db/?f=df081a6b15554f21a4040a05944dd4c43eae0c08

Any help will be much appreciated.
Thanks

lgirdwood, 2022-02-28:
Looks like a duplicate of https://github.com/thesofproject/sof/issues/5429

psammarco, 2022-02-28:
Hello @lgirdwood, as far as my understanding goes this is a rt5682+max98357a, however the CPU is not a GLK but a CML rather.

On ChromeOS it is recognized as such:
```
[ 3955.395820] max98357a MX98357A:00: acpi_subsys_suspend+0x0/0x55 returned 0 after 0 usecs
[ 3955.705832] max98357a MX98357A:00: calling acpi_subsys_suspend_late+0x0/0x4e @ 18704, parent: 0000:00:1f.3

```
however, I can't get the MX98357A interface recognized under Fedora. There is simply no mention of it in dmesg, and I don't think there is a direct rt5682+max98357a CML kernel module as there is for GLK, CONFIG_SND_SOC_INTEL_GLK_RT5682_MAX98357A_MACH.

if we look at lsmod from fedora we can see that snd_soc_max98357a is not even being used when it should according to chromeos lsmod.
```
[pietro@172-12-0-191 ~]$ lsmod |grep -i snd
snd_hda_codec_hdmi     65536  1
snd_sof_pci_intel_cnl    16384  0
snd_sof_intel_hda_common    77824  1 snd_sof_pci_intel_cnl
snd_sof_pci            16384  2 snd_sof_intel_hda_common,snd_sof_pci_intel_cnl
soundwire_intel        36864  1 snd_sof_intel_hda_common
snd_sof_intel_hda      16384  1 snd_sof_intel_hda_common
snd_sof_xtensa_dsp     16384  1 snd_sof_intel_hda_common
snd_sof               110592  2 snd_sof_pci,snd_sof_intel_hda_common
snd_soc_skl           135168  0
snd_soc_hdac_hda       20480  2 snd_sof_intel_hda_common,snd_soc_skl
snd_soc_sst_ipc        16384  1 snd_soc_skl
snd_soc_sst_dsp        28672  1 snd_soc_skl
snd_soc_acpi_intel_match    49152  3 snd_sof_intel_hda_common,snd_soc_skl,snd_sof_pci_intel_cnl
snd_soc_acpi           16384  3 snd_soc_acpi_intel_match,snd_sof_intel_hda_common,snd_soc_skl
snd_hda_ext_core       24576  4 snd_sof_intel_hda_common,snd_soc_hdac_hda,snd_soc_skl,snd_sof_intel_hda
snd_hda_intel          45056  0
snd_intel_dspcfg       20480  3 snd_hda_intel,snd_sof_intel_hda_common,snd_soc_skl
snd_intel_sdw_acpi     16384  2 snd_sof_intel_hda_common,snd_intel_dspcfg
snd_hda_codec         135168  3 snd_hda_codec_hdmi,snd_hda_intel,snd_soc_hdac_hda
snd_hwdep              16384  1 snd_hda_codec
snd_hda_core           86016  8 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_ext_core,snd_hda_codec,snd_sof_intel_hda_common,snd_soc_hdac_hda,snd_soc_skl,snd_sof_intel_hda
snd_soc_rt5682_i2c     16384  0
snd_soc_rt5682         77824  1 snd_soc_rt5682_i2c
snd_soc_rl6231         16384  1 snd_soc_rt5682
snd_soc_max98357a      16384  0
[pietro@172-12-0-191 ~]$ 
```
ChromeOS lsmod
```
[pietro@172-12-0-191 ~]$ grep -i snd Documents/chromebook-c1030-chromeos-logs/lsmod.txt 
snd_seq_dummy          16384  0
snd_seq                57344  5 snd_seq_dummy
snd_seq_device         16384  1 snd_seq
snd_soc_sof_rt5682     20480  3
snd_soc_hdac_hdmi      32768  1 snd_soc_sof_rt5682
snd_soc_dmic           16384  1
snd_sof_pci            16384  0
snd_sof_xtensa_dsp     16384  1 snd_sof_pci
snd_sof_intel_hda_common    53248  1 snd_sof_pci
snd_soc_hdac_hda       20480  1 snd_sof_intel_hda_common
snd_sof_intel_hda      16384  1 snd_sof_intel_hda_common
snd_sof                73728  2 snd_sof_pci,snd_sof_intel_hda_common
snd_soc_acpi_intel_match    28672  2 snd_sof_pci,snd_sof_intel_hda_common
snd_soc_acpi           16384  2 snd_sof_pci,snd_soc_acpi_intel_match
snd_hda_ext_core       16384  4 snd_sof_intel_hda_common,snd_soc_hdac_hdmi,snd_soc_hdac_hda,snd_sof_intel_hda
snd_intel_nhlt         16384  1 snd_sof_intel_hda_common
snd_hda_codec         110592  1 snd_soc_hdac_hda
snd_hwdep              16384  1 snd_hda_codec
snd_hda_core           61440  6 snd_hda_ext_core,snd_hda_codec,snd_sof_intel_hda_common,snd_soc_hdac_hdmi,snd_soc_hdac_hda,snd_sof_intel_hda
snd_soc_rt5682         65536  2 snd_soc_sof_rt5682
snd_soc_rl6231         16384  1 snd_soc_rt5682
snd_soc_max98357a      16384  1
[pietro@172-12-0-191 ~]$ 
```
sound card related kernel config
```
[pietro@172-12-0-191 ~]$ grep -iE 'rt5682|max98357a' /usr/src/linux-5.16.11/.config |grep -v '#'
CONFIG_SND_SOC_INTEL_SKL_NAU88L25_MAX98357A_MACH=m
CONFIG_SND_SOC_INTEL_DA7219_MAX98357A_GENERIC=m
CONFIG_SND_SOC_INTEL_BXT_DA7219_MAX98357A_COMMON=m
CONFIG_SND_SOC_INTEL_BXT_DA7219_MAX98357A_MACH=m
CONFIG_SND_SOC_INTEL_KBL_DA7219_MAX98357A_MACH=m
CONFIG_SND_SOC_INTEL_GLK_DA7219_MAX98357A_MACH=m
CONFIG_SND_SOC_INTEL_GLK_RT5682_MAX98357A_MACH=m
CONFIG_SND_SOC_INTEL_SOF_RT5682_MACH=m
CONFIG_SND_SOC_INTEL_CML_LP_DA7219_MAX98357A_MACH=m
CONFIG_SND_SOC_INTEL_SOF_CML_RT1011_RT5682_MACH=m
CONFIG_SND_SOC_MAX98357A=m
CONFIG_SND_SOC_RT5682=m
CONFIG_SND_SOC_RT5682_I2C=m
CONFIG_SND_SOC_RT5682_SDW=m
CONFIG_SND_SOC_RT5682S=m
[pietro@172-12-0-191 ~]$ 
```
ChromeOS sound card related kernel config
```
[pietro@172-12-0-191 ~]$ grep -iE 'rt5682|max98357a' Documents/chromebook-c1030-chromeos-logs/hp-chromebook-c1030-chromeos-kern-config.txt |grep -v '#'
CONFIG_SND_SOC_INTEL_SKL_NAU88L25_MAX98357A_MACH=m
CONFIG_SND_SOC_INTEL_DA7219_MAX98357A_GENERIC=m
CONFIG_SND_SOC_INTEL_BXT_DA7219_MAX98357A_MACH=m
CONFIG_SND_SOC_INTEL_KBL_DA7219_MAX98357A_MACH=m
CONFIG_SND_SOC_INTEL_SOF_CML_RT1011_RT5682_MACH=m
CONFIG_SND_SOC_INTEL_SOF_RT5682_MACH=m
CONFIG_SND_SOC_INTEL_CML_LP_DA7219_MAX98357A_MACH=m
CONFIG_SND_SOC_MAX98357A=m
CONFIG_SND_SOC_RT5682=m
[pietro@172-12-0-191 ~]$ 
```

Could someone confirm whether I am doing something wrong or if this soundcard on Cometlake is still not working?

Thanks

lgirdwood, 2022-02-28:
> however, I can't get the MX98357A interface recognized under Fedora. There is simply no mention of it in dmesg, and I don't think there is a direct rt5682+max98357a CML kernel module as there is for GLK, CONFIG_SND_SOC_INTEL_GLK_RT5682_MAX98357A_MACH.

The codec may have a different I2C address than expected by the driver. Can you scan the I2C bus (using some userspace tools) and see what's found ?

psammarco, 2022-02-28:
> The codec may have a different I2C address than expected by the driver. Can you scan the I2C bus (using some userspace tools) and see what's found ?

Some output here. Let me know if you need something more specific.
```
[pietro@172-12-0-191 ~]$ sudo i2cdetect -l
i2c-0	smbus     	SMBus I801 adapter at efa0      	SMBus adapter
i2c-1	i2c       	i915 gmbus dpb                  	I2C adapter
i2c-2	i2c       	i915 gmbus dpc                  	I2C adapter
i2c-3	i2c       	i915 gmbus misc                 	I2C adapter
i2c-4	i2c       	i915 gmbus dpd                  	I2C adapter
i2c-5	i2c       	AUX A/DDI A/PHY A               	I2C adapter
i2c-6	i2c       	AUX B/DDI B/PHY B               	I2C adapter
i2c-7	i2c       	AUX C/DDI C/PHY C               	I2C adapter
i2c-8	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-9	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-10	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
[pietro@172-12-0-191 ~]$ 
```
Also, is this issue somewhat related to this https://www.spinics.net/lists/alsa-devel/msg101261.html ?
Thanks

lgirdwood, 2022-02-28:
> Also, is this issue somewhat related to this https://www.spinics.net/lists/alsa-devel/msg101261.html ?
> Thanks

Yes, it could be, can you try the patch ? (or confirm you have it in your kernel) 

I cant see the codec in the list above, I would probably not expect it to report the codec name (since it requires a register read to get the codec ID), but I would have at least expected to see a device being reported on the bus.

plbossart, 2022-03-01:
I may be wrong but some Maxim codecs don't even have an I2C address. You won't find it in that case...

please add this file 
[sof-dyndbg.conf.txt](https://github.com/thesofproject/sof/files/8158375/sof-dyndbg.conf.txt) as /etc/modprobe.d/sof-dyndbg.conf

reboot 

and attach the results of 'alsa-info'

plbossart, 2022-03-01:
@psammarco are you using @MrChromebox's firmware here?

psammarco, 2022-03-01:
> > Also, is this issue somewhat related to this https://www.spinics.net/lists/alsa-devel/msg101261.html ?
> > Thanks
> 
> Yes, it could be, can you try the patch ? (or confirm you have it in your kernel)
> 
I have not tried the patch yet and I am not sure if it is already present in the mainline code.

psammarco, 2022-03-01:
> I may be wrong but some Maxim codecs don't even have an I2C address. You won't find it in that case...
>
> please add this file [...]

Here it is http://alsa-project.org/db/?f=96db906f0bff329dff50916857ecd06196ec44fd

> @psammarco are you using @MrChromebox's firmware here?
And yes
```
pietro@172-12-0-191 ~]$ dmesg |grep -i BIOS |tail -2
[    0.000000] SMBIOS 3.3.0 present.
[    0.000000] DMI: HP Jinlon/Jinlon, BIOS MrChromebox-4.14 07/25/2021
[pietro@172-12-0-191 ~]$ 
```
Thanks for the support. Much appreciated!

plbossart, 2022-03-01:
@MrChromebox yes you need an ACPI _HID to probe a platform driver, but there's no I2C support for this device. the speaker output is enabled with the 'sdmode' - just a GPIO that toggles when you want the amplifier to generate an output.

max98357a->sdmode = devm_gpiod_get_optional(&pdev->dev,
				"sdmode", GPIOD_OUT_LOW);

plbossart, 2022-03-01:
````
config SND_SOC_MAX98357A
	tristate "Maxim MAX98357A CODEC"
	depends on GPIOLIB
````
-> no dependency on I2C

MrChromebox, 2022-03-01:
@plbossart you're right, I must have been thinking of other Maxim parts that are both codec and amp (eg, 98373).

plbossart, 2022-03-01:
@psammarco there is no I2C interface on this amplifier, so nothing to add/invent.

psammarco, 2022-03-01:
Oh I see. So what is the issue then? Apology if the question may sound silly.

plbossart, 2022-03-01:
I think I know what's going on:

````
[    7.992598] snd_hda_intel 0000:00:1f.3: NHLT table not found
````

that means we can't detect whether there are dmics or not on this device.

And most likely the matching in sound/hda/intel-dspcfg.c fails because it's not a Chromebook.

plbossart, 2022-03-01:
````
!!DMI Information
!!---------------

Manufacturer:      HP
Product Name:      Jinlon
Product Version:   rev5
Firmware Version:  MrChromebox-4.14
System SKU:        sku1
Board Vendor:      HP
Board Name:        Jinlon
````

What we use is 		

````c
.dmi_table = (const struct dmi_system_id []) {
			{
				.ident = "Google Chromebooks",
				.matches = {
					DMI_MATCH(DMI_SYS_VENDOR, "Google"),
				}
			},
			{}
		}
````

so it's not really a big surprise that the SOF driver is not selected and the firmware is not found in the right place.

MrChromebox, 2022-03-01:
@plbossart google (coreboot) lets OEMs override the SMBIOS data, which is why HP is listed as manufacturer vs Google. I'll disable that for my next release.

@psammarco shoot me an email for a test build with this corrected and we'll see that fixes things

psammarco, 2022-03-02:
> ```
> !!DMI Information
> !!---------------
>
> Manufacturer:      HP
> Product Name:      Jinlon
> Product Version:   rev5
> Firmware Version:  MrChromebox-4.14
> System SKU:        sku1
> Board Vendor:      HP [...]

Awesome finding!!!

> @plbossart google (coreboot) lets OEMs override the SMBIOS data, which is why HP is listed as manufacturer vs Google. I'll disable that for my next release.
> 
> @psammarco shoot me an email for a test build with this corrected and we'll see that fixes things

Done. Thanks mate :)

psammarco, 2022-03-07:
The change implemented by @MrChromebox which changed the OEM from HP to Google did the trick. I have now got audio, however only out of the jack. 

I wonder if this is a driver SOF issue or something else related to coreboot. 

Thanks

MrChromebox, 2022-03-07:
> I wonder if this is a driver SOF issue or something else related to coreboot.

it's not on the coreboot side, that's not within the scope of what system firmware does. The DMI system vendor change was to work around a larger problem with Linux drivers using DMI to handle different hardware classes/cases, because it was easier to do so there.

psammarco, 2022-03-07:
Oh okay @MrChromebox, thanks for the clarification and the help. Should this issue/bug be closed then?

MrChromebox, 2022-03-07:
in my testing, only the headphone device is detected/working on Cometlake Chromebooks, using Manjaro 21.x w/proprietary drivers as my test bed. Internal speakers are not available/selectable. Can you replicate that?

psammarco, 2022-03-07:
> in my testing, only the headphone device is detected/working on Cometlake Chromebooks, using Manjaro 21.x w/proprietary drivers as my test bed. Internal speakers are not available/selectable. Can you replicate that?

Yes. It is exactly the same behaviour you have described.

Denisuu, 2022-03-08:
> in my testing, only the headphone device is detected/working on Cometlake Chromebooks, using Manjaro 21.x w/proprietary drivers as my test bed. Internal speakers are not available/selectable. Can you replicate that?

Also the exact same behaviour on my Gemini Lake Chromebook on both your FW and Breath.

MrChromebox, 2022-03-08:
> Also the exact same behaviour on my Gemini Lake Chromebook on both your FW and Breath.

don't you mean the opposite? on GLK the speakers work but not headphone output

Denisuu, 2022-03-08:
[image]

Internal speakers are not available/selectable only headphones is selectable but sound comes out of the speakers.

MrChromebox, 2022-03-08:
> Internal speakers are not available/selectable only headphones is selectable but sound comes out of the speakers.

right, and on CML it only comes out of the headphones. So similar but not identical issues

plbossart, 2022-03-22:
@psammarco speaker issues are in the realm of machine driver/gpio issues, it's not strictly related to SOF proper. I don't think anyone in our team has this hardware variant so it'll be hard to suggest any solution.

psammarco, 2022-03-23:
> @psammarco speaker issues are in the realm of machine driver/gpio issues, it's not strictly related to SOF proper. I don't think anyone in our team has this hardware variant so it'll be hard to suggest any solution.

I understand. So could it be that I may be missing a GPIO driver required by the speaker?

plbossart, 2022-03-24:
@psammarco yes a GPIO sounds like a likely suspect, it's what is used to enable/disable playback on the Maxim amplifier.

psammarco, 2022-03-24:
> @psammarco yes a GPIO sounds like a likely suspect, it's what is used to enable/disable playback on the Maxim amplifier.

Alright then. Back to the kernel configuration. Hopefully a driver missing ends up being the issue.

paulmenzel, 2022-04-05:
Could be related to coreboot using probe functionality only supported by the Chromium OS Linux kernel and not yet upstreamed to the Linux kernel, cf. [[PATCH] CHROMIUM: i2c: Add device property for probing](https://lore.kernel.org/all/20211220210643.47842-1-pmenzel@molgen.mpg.de/#r).

lgirdwood, 2022-04-05:
> Could be related to coreboot using probe functionality only supported by the Chromium OS Linux kernel and not yet upstreamed to the Linux kernel, cf. [[PATCH] CHROMIUM: i2c: Add device property for probing](https://lore.kernel.org/all/20211220210643.47842-1-pmenzel@molgen.mpg.de/#r).

@paulmenzel will you be upstreaming this ?

MrChromebox, 2022-04-05:
@lgirdwood the "probed" patch is a dirty hack due to coreboot generating invalid ACPI statuses for some i2c devices, setting everything to "present/working" and forcing the OS driver to determine if that's correct. It will not, and should not, be upstreamed

plbossart, 2022-04-05:
The gpio support and I2C probe are orthogonal topics, aren't they? The maxim amp used in these devices don't use I2C so whatever happens for I2C is likely irrelevant.

psammarco, 2022-04-05:
> Could be related to coreboot using probe functionality only supported by the Chromium OS Linux kernel and not yet upstreamed to the Linux kernel, cf. [[PATCH] CHROMIUM: i2c: Add device property for probing](https://lore.kernel.org/all/20211220210643.47842-1-pmenzel@molgen.mpg.de/#r).

I already use this patch which got the touchscreen to work. Had no effect on the speaker unfortunately. 

@plbossart  could the topology which handles pin routing and speaker detection be missing?

plbossart, 2022-04-05:
The topology is DSP-centric, there's no pin routing or detection at the hardware level.

psammarco, 2022-04-07:
> The topology is DSP-centric, there's no pin routing or detection at the hardware level.

I am out of ideas at this point. Thanks for the help though.

plbossart, 2023-01-05:
no update in 9+ months, closing. If there's a need please open a new issue based on more recent kernel+firmware. thanks!

closed 2023-01-05 by plbossart