thesofproject/sof #5440: No output on glkda7219 + glkrt5682 (OCTOPUS) Chromebook devices

Denisuu, 2022-02-27:
**Describe the bug**
Gemini lake processor Chromebooks that have both `sof-glkrt5682max`and `sof-glkda7219max` don't have working audio. 
(Only dummy output).

**Reproduction Rate**
Sound doesn't work OOB on devices listed below. I've tried multiple different `firmware-sof-signed` packages ranging from 1.6.1 to 1.9.1 none seem to work.

**Impact**
Devices that are confirmed to have this issue are: 
`BLOOG`
`BLOOGLET`
`SPARKY360`

[Source](https://github.com/cb-linux/breath/issues/29)

The overlay tree of my device HP 14a BLOOGLET:
```
├── chromeos-base
│   ├── chromeos-bsp-octopus
│   │   ├── chromeos-bsp-octopus-0.0.2-r48.ebuild
│   │   ├── chromeos-bsp-octopus-9999.ebuild
│   │   ├── files
│   │   │   ├── 93-powerd-overrides.rules
│   │   │   ├── 99-chromeos-octopus-usb-charge-mode.rules
│   │   │   ├── 99-chromeos-octopus-usb-runtime-suspend.rules
│   │   │   ├── blooglet
│   │   │   │   ├── audio
│   │   │   │   │   ├── cras-config
│   │   │   │   │   │   ├── board.ini
│   │   │   │   │   │   ├── card2_settings
│   │   │   │   │   │   ├── card_settings
│   │   │   │   │   │   └── dsp.ini
│   │   │   │   │   └── ucm-config
│   │   │   │   │       ├── sof-glkda7219max.1loc
│   │   │   │   │       │   ├── HiFi.conf
│   │   │   │   │       │   └── sof-glkda7219max.conf
│   │   │   │   │       └── sof-glkrt5682max.1loc
│   │   │   │   │           ├── HiFi.conf
│   │   │   │   │           └── sof-glkrt5682max.conf
│   │   │   │   └── thermal
│   │   │   │       └── dptf.dv
```

Unconfirmed devices that also have `sof-glkrt5682max` and `sof-glkda7219max` and are likely to be affected:
`BLOOGUARD` 
`BLORB`
`BOBBA`
`BOBBA360`
`DORP`
`DROID`
`PHASER`
`PHASER360`
`SPARKY`
`VORTICON`
`VORTININJA`

[source](https://chromeos.guide/osdev/overlays/overlay-octopus)

**Environment**

- Sound doesn´t work in Crouton (ChromeOS Chroot)
- Ubuntu based distro´s through stock kernel (using [Breath](https://github.com/cb-linux/breath))
- Manjaro or Ubuntu based distro's using [Mr.Chromebox](https://mrchromebox.tech/) UEFI firmware.

Sound only works in ChromeOS. 

**Log files**
All logs are from Xubuntu using Breath on my own device BLOOGLET)

Output from ChromeOS (Crosh) shell:
```
chronos@localhost / $ cat /proc/asound/cards
 0 [sofglkda7219max]: sof-glkda7219ma - sof-glkda7219max
                      HP-Bloog-rev4-Bloog
```
`dmesg` snippit:
```
[   10.718788] sof-audio-pci-intel-apl 0000:00:0e.0: error: cl_copy_fw: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[   10.719336] sof-audio-pci-intel-apl 0000:00:0e.0: error: extended rom status:  0x81000012 0x2c 0x0 0x0 0x0 0x0 0x1551100 0x0
[   10.719338] sof-audio-pci-intel-apl 0000:00:0e.0: error: load fw failed ret: -110
[   10.719364] sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to reset DSP
[   10.719367] sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to boot DSP firmware -110
[   10.769790] sof-audio-pci-intel-apl 0000:00:0e.0: error: hda_dsp_core_reset_enter: timeout on HDA_DSP_REG_ADSPCS read
[   10.769795] sof-audio-pci-intel-apl 0000:00:0e.0: error: dsp core reset failed: core_mask 1
[   10.770940] sof-audio-pci-intel-apl 0000:00:0e.0: error: sof_probe_work failed err: -110
```

Denisuu, 2022-02-27:
**Full logs**

<details>
  <summary>dmesg</summary>

```
[    0.000000] Linux version 5.10.88 (milkydeveloper@GF63) (gcc (Ubuntu 11.2.0-13ubuntu1) 11.2.0, GNU ld (GNU Binutils for Ubuntu) 2.37) #1 SMP PREEMPT Tue Dec 28 15:13:11 CST 2021
[    0.000000] Command line: cros_secure console=tty1 root=/dev/sda2 i915.modeset=1 rootwait rw fbcon=logo-pos:center,logo-count:1 loglevel=0 splash
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[3]:  576, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  640, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1b, context size is 704 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000000fff] type 16
[    0.000000] BIOS-e820: [mem 0x0000000000001000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000010000000-0x0000000012150fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000012151000-0x00000000799f4fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000799f5000-0x0000000079ffffff] type 16
[    0.000000] BIOS-e820: [mem 0x000000007a000000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d0000000-0x00000000d0ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe042000-0x00000000fe043fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed17fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000017fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: HP Bloog/Bloog, BIOS Google_Bloog.11297.275.0 05/16/2021
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Initial usec timer 3946499
[    0.000000] tsc: Detected 1094.400 MHz processor
[    0.000014] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000017] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000023] last_pfn = 0x180000 max_arch_pfn = 0x400000000
[    0.000026] MTRR default type: uncachable
[    0.000027] MTRR fixed ranges enabled:
[    0.000029]   00000-9FFFF write-back
[    0.000030]   A0000-BFFFF uncachable
[    0.000031]   C0000-FFFFF write-back
[    0.000032] MTRR variable ranges enabled:
[    0.000033]   0 base 0000000000 mask 7F80000000 write-back
[    0.000035]   1 base 007B000000 mask 7FFF000000 uncachable
[    0.000036]   2 base 007C000000 mask 7FFC000000 uncachable
[    0.000037]   3 base 00B0000000 mask 7FF0000000 write-combining
[    0.000038]   4 base 0100000000 mask 7F80000000 write-back
[    0.000039]   5 disabled
[    0.000039]   6 disabled
[    0.000040]   7 disabled
[    0.000041]   8 disabled
[    0.000041]   9 disabled
[    0.000140] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000245] last_pfn = 0x799f5 max_arch_pfn = 0x400000000
[    0.003912] Using GB pages for direct mapping
[    0.004317] ACPI: Early table checksum verification disabled
[    0.004322] ACPI: RSDP 0x00000000000F0000 000024 (v02 COREv4)
[    0.004326] ACPI: XSDT 0x0000000079B060E0 00005C (v01 COREv4 COREBOOT 00000000 CORE 00000000)
[    0.004333] ACPI: FACP 0x0000000079B09340 0000F4 (v04 COREv4 COREBOOT 00000000 CORE 00000000)
[    0.004339] ACPI: DSDT 0x0000000079B06280 0030B2 (v02 COREv4 COREBOOT 20110725 INTL 20180531)
[    0.004343] ACPI: FACS 0x0000000079B06240 000040
[    0.004347] ACPI: FACS 0x0000000079B06240 000040
[    0.004350] ACPI: SSDT 0x0000000079B09440 001794 (v02 COREv4 COREBOOT 0000002A CORE 0000002A)
[    0.004354] ACPI: MCFG 0x0000000079B0ABE0 00003C (v01 COREv4 COREBOOT 00000000 CORE 00000000)
[    0.004357] ACPI: TPM2 0x0000000079B0AC20 00004C (v04 COREv4 COREBOOT 00000000 CORE 00000000)
[    0.004360] ACPI: APIC 0x0000000079B0AC70 00006C (v02 COREv4 COREBOOT 00000000 CORE 00000000)
[    0.004364] ACPI: NHLT 0x0000000079B0ACE0 001A8F (v05 COREv4 COREBOOT 00000000 CORE 00000000)
[    0.004368] ACPI: HPET 0x0000000079B0E770 000038 (v01 COREv4 COREBOOT 00000000 CORE 00000000)
[... 1104 lines cut ...]
snd_soc_sst_ipc
snd_sof
snd_sof_intel_hda
snd_sof_intel_hda_common
snd_sof_pci
snd_sof_pci_intel_apl
snd_sof_xtensa_dsp
uvcvideo
videobuf2_vmalloc

!!ALSA/HDA dmesg
!!--------------

!!Packages installed
!!--------------------

ii  alsa-oss                              1.1.8-1                                                              amd64        ALSA wrapper for OSS applications
ii  alsa-tools                            1.2.2-1                                                              amd64        Console based ALSA utilities for specific hardware
ii  alsa-tools-gui                        1.2.2-1                                                              amd64        GUI based ALSA utilities for specific hardware
ii  alsa-topology-conf                    1.2.4-1                                                              all          ALSA topology configuration files
ii  alsa-ucm-conf                         1.2.4-2ubuntu1.4                                                     all          ALSA Use Case Manager configuration files
ii  alsa-utils                            1.2.4-1ubuntu3                                                       amd64        Utilities for configuring and using ALSA
```

</details>

J0shuaT, 2022-02-27:
I also submitted an issue for this #5429 . I also have the same type of device as you, HP Chromebook 14a (BLOOGLET)

lgirdwood, 2022-02-28:
So I think #5429 is different, as it looks like we are missing an ingredient (suspect codec??) driver/config upstream (and hence the SOF driver is not fully being bound). This issue is different as driver binding takes place but the FW fails to load.

@Denisuu I suspect this is the FW failing to be verified by the signing mechanism. i.e. the key used to validate is different to the key used to sign the FW you have tried. Each SOF release is signed with two keys (production and community) giving us two binaries for each target (same version and functionality,but signed with different keys) can you try both FWs to rule out the wrong key. You may need to manually copy the FW files here to test this.

Denisuu, 2022-02-28:
> So I think #5429 is different, as it looks like we are missing an ingredient (suspect codec??) driver/config upstream (and hence the SOF driver is not fully being bound). This issue is different as driver binding takes place but the FW fails to load.

I think it's different because he ran this script [sof-audio-setup](https://github.com/cb-linux/breath/blob/main/bin/sof-setup-audio) as instructed in the audio section of the [Breath docs.](https://cb-linux.github.io/breath/docs.html#/?id=audio) Can you confirm this? @J0shuaT 

> I suspect this is the FW failing to be verified by the signing mechanism. i.e. the key used to validate is different to the key used to sign the FW you have tried. Each SOF release is signed with [...]

Previously I tried the packages below, then the ones for Ubuntu Jammy through the same site because they go up to 1.9.1
[Binary package “firmware-sof-signed”](https://launchpad.net/ubuntu/impish/+package/firmware-sof-signed)

I don't quite understand what I need to do now? Do I get the latest 2.0 package from here: [Thesofproject releases](https://github.com/thesofproject/sof/releases) then follow the documentation here? [thesofproject.github.io](https://thesofproject.github.io/latest/index.html) Where do I need to manually copy FW files?

lgirdwood, 2022-02-28:
@Denisuu yes, please follow the docs. Also please take a look here
https://github.com/thesofproject/sof-bin/tree/main/v2.0.x/sof-v2.0

The sof-bin repo has scripts to update and install releases. There are Intel signed and community signed FW binaries. 

IIUC the SOF kernel will detect a chromebook and try the community signed FW rather than trying the link in /lib/firmware/intel/sof/sof-glk.ri. 

You may need to manually overwrite the community sof-glk.ri (under the community directory) and the soft link sof-glk.ri with the version from sof-bin you want to force using a different signed binary.

Denisuu, 2022-02-28:
@lgirdwood Ok, I read the documentation and entered the commands below. Audio still doesn't work, did I miss something or did someting wrong?

Commands I entered:
```
git clone https://github.com/thesofproject/sof-bin.git
cd sof-bin/
mv ~/sof-bin/v2.0.x/sof-v2.0/community/* ~/sof-bin/v2.0.x/sof-v2.0/
rm -r  ~/sof-bin/v2.0.x/sof-v2.0/community/
rm -r  ~/sof-bin/v2.0.x/sof-v2.0/intel-signed/
cd v2.0.x/
sudo rsync -a sof-v2.0   /lib/firmware/intel/
sudo ln -s sof-v2.0      /lib/firmware/intel/sof
sudo ln -s sof-tplg-v2.0 /lib/firmware/intel/sof-tplg
sudo rsync tools-v2.0/*  /usr/local/bin
sudo init 6
sudo dmesg
```

dmesg snippit:
```
[    7.481664] snd_hda_intel 0000:00:0e.0: attach allowed to drvr sof-audio-pci-intel-apl [internal device]
[    8.941696] pci 0000:00:0e.0: attach allowed to drvr sof-audio-pci-intel-apl [internal device]
[    8.941728] sof-audio-pci-intel-apl 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if info 0x040100
[    8.942055] sof-audio-pci-intel-apl 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if 0x040100
[    8.942183] sof-audio-pci-intel-apl 0000:00:0e.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops)
[    8.961116] sof-audio-pci-intel-apl 0000:00:0e.0: attach allowed to drvr snd_soc_skl [internal device]
[    8.968125] sof-audio-pci-intel-apl 0000:00:0e.0: use msi interrupt mode
[    8.976912] sof-audio-pci-intel-apl 0000:00:0e.0: attach allowed to drvr snd_hda_intel [internal device]
[    8.979009] sof-audio-pci-intel-apl 0000:00:0e.0: hda codecs found, mask 4
[    8.981088] sof-audio-pci-intel-apl 0000:00:0e.0: Direct firmware load for intel/sof/sof-glk.ri failed with error -2
[    8.981094] sof-audio-pci-intel-apl 0000:00:0e.0: error: request firmware intel/sof/sof-glk.ri failed err: -2
[    8.981099] sof-audio-pci-intel-apl 0000:00:0e.0: you may need to download the firmware from https://github.com/thesofproject/sof-bin/
[    8.981101] sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to load DSP firmware -2
[    8.989379] max98357a MX98357A:00: GPIO lookup for consumer sdmode
[    8.989382] max98357a MX98357A:00: using ACPI for GPIO lookup
[    8.989385] acpi MX98357A:00: GPIO: looking up sdmode-gpios
[    8.989387] acpi MX98357A:00: GPIO: looking up sdmode-gpio
[    8.989390] acpi MX98357A:00: GPIO: _DSD returned MX98357A:00 0 0 0
[    8.989466] gpio gpiochip1: Persistence not supported for GPIO 15
[    8.995230] sof-audio-pci-intel-apl 0000:00:0e.0: error: sof_probe_work failed err: -2
```

<details>
  <summary>full dmseg</summary>

```
[    0.000000] Linux version 5.10.88 (milkydeveloper@GF63) (gcc (Ubuntu 11.2.0-13ubuntu1) 11.2.0, GNU ld (GNU Binutils for Ubuntu) 2.37) #1 SMP PREEMPT Tue Dec 28 15:13:11 CST 2021
[    0.000000] Command line: cros_secure console=tty1 root=/dev/sda2 i915.modeset=1 rootwait rw fbcon=logo-pos:center,logo-count:1 loglevel=0 splash
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[3]:  576, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  640, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1b, context size is 704 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000000fff] type 16
[    0.000000] BIOS-e820: [mem 0x0000000000001000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000010000000-0x0000000012150fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000012151000-0x00000000799f4fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000799f5000-0x0000000079ffffff] type 16
[    0.000000] BIOS-e820: [mem 0x000000007a000000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d0000000-0x00000000d0ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe042000-0x00000000fe043fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed17fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000017fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[... 790 lines cut ...]
[   10.978711] rtw_8822ce 0000:01:00.0: start vif ec:2e:98:8a:37:1b on port 0
[   20.384560] input: MX Master 3 Keyboard as /devices/virtual/misc/uhid/0005:046D:B023.0001/input/input6
[   20.384689] input: MX Master 3 Mouse as /devices/virtual/misc/uhid/0005:046D:B023.0001/input/input7
[   20.384768] hid-generic 0005:046D:B023.0001: input,hidraw0: BLUETOOTH HID v0.15 Keyboard [MX Master 3] on ec:2e:98:8a:37:1a
[   20.451197] logitech-hidpp-device 0005:046D:B023.0001: HID++ 4.5 device connected.
[   20.781411] input: Logitech Wireless Mouse MX Master 3 as /devices/virtual/misc/uhid/0005:046D:B023.0001/input/input9
[   20.781667] logitech-hidpp-device 0005:046D:B023.0001: input,hidraw0: BLUETOOTH HID v0.15 Keyboard [Logitech Wireless Mouse MX Master 3] on ec:2e:98:8a:37:1a
[   23.956807] wlp1s0: authenticate with dc:39:6f:43:9c:5f
[   24.355149] wlp1s0: send auth to dc:39:6f:43:9c:5f (try 1/3)
[... cut ...]
[   25.257959] wlp1s0: send auth to dc:39:6f:43:9c:60 (try 1/3)
[   25.264590] wlp1s0: authenticated
[   25.264960] wlp1s0: associate with dc:39:6f:43:9c:60 (try 1/3)
[   25.271108] wlp1s0: RX AssocResp from dc:39:6f:43:9c:60 (capab=0x1431 status=0 aid=5)
[   25.271139] rtw_8822ce 0000:01:00.0: sta dc:39:6f:43:9c:60 joined with macid 0
[   25.271627] wlp1s0: associated
[   25.296090] IPv6: ADDRCONF(NETDEV_CHANGE): wlp1s0: link becomes ready
[   25.362433] wlp1s0: Limiting TX power to 20 (20 - 0) dBm as advertised by dc:39:6f:43:9c:60
```
</details>

lgirdwood, 2022-02-28:
> [    8.981088] sof-audio-pci-intel-apl 0000:00:0e.0: Direct firmware load for intel/sof/sof-glk.ri failed with error -2
> [    8.981094] sof-audio-pci-intel-apl 0000:00:0e.0: error: request firmware [...]

Can you try and copy and test (one by one) the sof-glk.ri files (from the sof-bin intel-signed and community-signed directories repo) to ```/lib/firmware/intel/sof/```

This will allow the SOF driver to try both the intel and community signed GLK FW binaries.

MrChromebox, 2022-02-28:
There's no issue loading the community firmware on GLK Chromebooks, it works fine here on mine. The issue appears to be isolated to models using the the da7219+max98357a combo; ones using rt5682+max98357a are working (at least for main speaker output).

Denisuu, 2022-03-01:
> Can you try and copy and test (one by one) the sof-glk.ri files (from the sof-bin rep, intel-signed and community-signed directories) to `/lib/firmware/intel/sof/`
> 
> This will allow the SOF driver to try both the intel and community signed GLK FW binaries.

@lgirdwood  I just tried the intel-signed files, same thing. What do you mean one by one? You mean mixing intel-signed and community signed. That's a lot of possible combinations!
 
> There's no issue loading the community firmware on GLK Chromebooks, it works fine here on mine. The issue appears to be isolated to models using the the da7219+max98357a combo; ones using rt5682+max98357a are working (at least for main speaker output).

@MrChromebox I think it might be all devices that have `da7219`.

CASTA doesn't have working audio either. It gives exactly the same output as BLOOGLET when showing the contents of `/proc/asound/cards`

<details>
  <summary>BLOOGLET</summary>

```
 0 [sofglkda7219max]: sof-glkda7219ma - sof-glkda7219max
                      HP-Bloog-rev4-Bloog
```
</details>
<details>
  <summary>CASTA</summary>

```
0 [sofglkda7219max]: sof-glkda7219ma - sof-glkda7219max
                     Google-Casta-rev6
```
</details>

I'm going to switch to Manjaro on your custom firmware again for testing because I'm not sure if I should run [Milkydeveloper's script](https://github.com/cb-linux/breath/blob/main/bin/sof-setup-audio) each time I make a change. I uncommented line 20, tried with and without the script each time.

I don't even get to see the dummy output on Breath I always get: `Establishing connection to PulseAudio. Please wait...`

aiChaoSONG, 2022-03-01:
The kernel still request a firmware under /lib/firmware/intel/sof/sof-glk.ri, so I suspect that you are using an old kernel. New kernel will request /lib/firmware/intel/sof/community/sof-glk.ri
```
Direct firmware load for intel/sof/sof-glk.ri failed with error -2
request firmware intel/sof/sof-glk.ri failed err: -2
```

Denisuu, 2022-03-01:
I just flashed mr.Chromebox UEFI firmware and installed Manjaro and updated to the latest kernel. Tried installing `sof-bin` community as mentioned above. Then tried installing the latest `sof-firmware` from AUR. 

Still no output device! :(

**sof-firmware and kernel version:**
```
[denisuu@BLOOGLET ~]$ pacman -Qi sof-firmware | grep Version
Version         : 2.0-1

[denisuu@BLOOGLET ~]$ uname -r
5.16.2-2-rt19-MANJARO

[denisuu@BLOOGLET ~]$ lsb_release -r
Release:	21.2.4
```
**cards:** 
`/proc/asound/cards` doesn't show: `sof-glkda7219ma - sof-glkda7219max` like it does on stock ChromeOS kernel.

```
[denisuu@BLOOGLET ~]$ cat /proc/asound/cards
 0 [PCH            ]: HDA-Intel - HDA Intel PCH
                      HDA Intel PCH at 0x91418000 irq 122
```

**Snippit of dmesg:**
```
[denisuu@BLOOGLET ~]$ sudo dmesg | grep -e sof- -e glk -e snd -e hda
[    2.752550] Modules linked in: snd_compress syscopyarea ac97_bus cros_ec_typec(+) sysfillrect processor_thermal_device_pci_legacy(+) sysimgblt processor_thermal_device snd_pcm_dmaengine snd_hwdep i2c_i801(+) fb_sys_fops fjes(-) intel_lpss_pci(+) processor_thermal_rfim typec snd_pcm cec i2c_smbus processor_thermal_mbox lpc_ich tpm_tis_spi intel_lpss processor_thermal_rapl roles cros_usbpd_notify rfkill tpm_tis_core intel_gtt intel_rapl_common idma64 dwc3_pci snd_timer agpgart intel_soc_dts_iosf tpm i2c_hid_acpi cros_ec_lpcs snd intel_vbtn int3403_thermal cros_ec chromeos_tbmc soc_button_arr[denisuu@BLOOGLET ~]$ lsb_release -r
Release:	21.2.4
ay sparse_keymap i2c_hid elants_i2c rng_core soundcore elan_i2c int340x_thermal_zone video int3400_thermal acpi_thermal_rel cros_kbd_led_backlight intel_pmc_bxt mac_hid coreboot_table drm uinput crypto_user fuse bpf_preload ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 mmc_block serio_raw atkbd libps2 sdhci_pci crc32c_intel cqhci i8042 sdhci xhci_pci mmc_core xhci_pci_renesas serio
[    2.921116] snd_hda_intel 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if info 0x040100
[    3.224460] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/glk_dmc_ver1_04.bin (v1.4)
[    3.281825] snd_hda_intel 0000:00:0e.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
```
**Full dmesg:**

<details>
  <summary>Manjaro dmesg</summary>

```
[    0.000000] microcode: microcode updated early to revision 0x1c, date = 2021-05-10
[    0.000000] Linux version 5.16.2-2-rt19-MANJARO (bernhard@mk5) (gcc (GCC) 11.2.0, GNU ld (GNU Binutils) 2.38) #1 SMP PREEMPT_RT Tue Feb 15 09:37:27 CET 2022
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.16-rt-x86_64 root=UUID=9611c2fc-b133-4a7d-9323-6415934aa544 rw quiet apparmor=1 security=apparmor udev.log_priority=3
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[3]:  576, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  640, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1b, context size is 704 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 2032
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000000fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000001000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000fa68fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000fa69000-0x000000000fa69fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000000fa6a000-0x000000000fc7afff] usable
[    0.000000] BIOS-e820: [mem 0x000000000fc7b000-0x000000000fc80fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000000fc81000-0x000000000fc84fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000fc85000-0x000000000fc86fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000000fc87000-0x000000000feaefff] usable
[    0.000000] BIOS-e820: [mem 0x000000000feaf000-0x000000000ff2efff] type 20
[    0.000000] BIOS-e820: [mem 0x000000000ff2f000-0x000000000fff2fff] reserved
[... 895 lines cut ...]
[ 1412.999243] audit: type=1105 audit(1646135544.077:165): pid=2151 uid=1000 auid=1000 ses=1 subj==unconfined msg='op=PAM:session_open grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[ 1413.013242] audit: type=1106 audit(1646135544.091:166): pid=2151 uid=1000 auid=1000 ses=1 subj==unconfined msg='op=PAM:session_close grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[ 1413.013299] audit: type=1104 audit(1646135544.091:167): pid=2151 uid=1000 auid=1000 ses=1 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[... cut ...]
[ 2091.354806] audit: type=1110 audit(1646136222.433:185): pid=2341 uid=1000 auid=1000 ses=1 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[ 2091.358208] audit: type=1105 audit(1646136222.436:186): pid=2341 uid=1000 auid=1000 ses=1 subj==unconfined msg='op=PAM:session_open grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
```

</details>

lgirdwood, 2022-03-01:
> @lgirdwood I just tried the intel-signed files, same thing. What do you mean one by one? You mean mixing intel-signed and community signed. That's a lot of possible combinations!

Sorry, I mean trying both sof-glk.ri files for the SOF version you want. One is signed by the Intel key and the other by the community key. This would help and rule out signing issues.

plbossart, 2022-03-01:
@Denisuu your last dmesg contains nothing related to SOF, possibly due to another bug
````
 RIP: 0010:cros_ec_check_features+0xa/0xa0
````

You may want to do a blacklist snd-sof-pci-intel-apl, reboot, dmesg -C, modprobe snd-sof-pci-intel-apl, that way we can see what's really going on.

Also add this file 
[sof-dyndbg.conf.txt](https://github.com/thesofproject/sof/files/8162478/sof-dyndbg.conf.txt)
as /etc/modprobe.d/sof-dyndbg.conf prior to the reboot, it'll show a more verbose dmesg log.

J0shuaT, 2022-03-01:
@Denisuu I have the same problem. I am running Ubuntu with Mr. Chromeboxes UEFI test firmware, and `/proc/asound/cards`, shows this:

` 0 [PCH            ]: HDA-Intel - HDA Intel PCH
                      HDA Intel PCH at 0x91418000 irq 124`

J0shuaT, 2022-03-01:
@plbossart I tried blacklisting snd-sof-pci-intel-apl, since I have the same issue as @Denisuu . Here is my `dmesg` output:

`[    0.000000] microcode: microcode updated early to revision 0x1a, date = 2020-10-23
[    0.000000] Linux version 5.16.10-051610-generic (kernel@kathleen) (gcc (Ubuntu 11.2.0-16ubuntu1) 11.2.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #202202161442 SMP PREEMPT Wed Feb 16 14:49:20 UTC 2022
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.16.10-051610-generic root=UUID=e3019cda-b08a-449c-affe-9e7210ccbf9c ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[3]:  576, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  640, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1b, context size is 704 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 2032
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000000fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000001000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000fa4cfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000fa4d000-0x000000000fa4dfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000000fa4e000-0x000000000fc7afff] usable
[    0.000000] BIOS-e820: [mem 0x000000000fc7b000-0x000000000fc80fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000000fc81000-0x000000000fc84fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000fc85000-0x000000000fc86fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000000fc87000-0x000000000feaefff] usable
[    0.000000] BIOS-e820: [mem 0x000000000feaf000-0x000000000fff2fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000fff3000-0x000000000fff6fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000fff7000-0x000000000fffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000000ffff000-0x000000000fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000010000000-0x0000000012150fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000012151000-0x0000000079aedfff] usable
[    0.000000] BIOS-e820: [mem 0x0000000079aee000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffae0000-0x00000000ffb1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000017fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by EDK II
[    0.000000] efi: SMBIOS=0xffe8000 SMBIOS 3.0=0xffe6000 ACPI=0xfffe000 ACPI 2.0=0xfffe014 MEMATTR=0xfa3b418 MOKvar=0xfa37000 RNG=0xffa4a98 
[    0.000000] efi: seeding entropy pool
[    0.000000] random: fast init done
[    0.000000] SMBIOS 3.3.0 present.
[    0.000000] DMI: HP Blooglet/Blooglet, BIOS MrChromebox-4.15 01/18/2022
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 1094.400 MHz processor
[    0.000024] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000029] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000038] last_pfn = 0x180000 max_arch_pfn = 0x400000000
[    0.000082] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000293] last_pfn = 0x79aee max_arch_pfn = 0x400000000
[    0.004725] e820: update [mem 0x0fa37000-0x0fa37fff] usable ==> reserved
[    0.004760] Using GB pages for direct mapping
[    0.005927] Secure boot disabled
[    0.005928] RAMDISK: [mem 0x08a8d000-0x0b620fff]
[    0.005936] ACPI: Early table checksum verification disabled
[    0.005942] ACPI: RSDP 0x000000000FFFE014 000024 (v02 COREv4)
[    0.005948] ACPI: XSDT 0x000000000FA4DF18 00006C (v01 COREv4 COREBOOT 00000000      01000013)
[    0.005957] ACPI: FACP 0x000000000FFFC000 000114 (v06 COREv4 COREBOOT 00000000 CORE 20210331)
[    0.005966] ACPI: DSDT 0x000000000FFF8000 0032D8 (v02 COREv4 COREBOOT 20110725 INTL 20210331)
[    0.005972] ACPI: FACS 0x0000000079AFF240 000040
[    0.005976] ACPI: SSDT 0x000000000FC85000 001803 (v02 COREv4 COREBOOT 0000002A CORE 20210331)
[    0.005981] ACPI: MCFG 0x000000000FFF7000 00003C (v01 COREv4 COREBOOT 00000000 CORE 20210331)
[    0.005986] ACPI: TPM2 0x000000000FC80000 00004C (v04 COREv4 COREBOOT 00000000 CORE 20210331)
[    0.005991] ACPI: APIC 0x000000000FC7F000 000062 (v03 COREv4 COREBOOT 00000000 CORE 20210331)
[    0.005995] ACPI: NHLT 0x000000000FC7D000 001A93 (v05 COREv4 COREBOOT 00000000 CORE 00000000)
[    0.006000] ACPI: DBG2 0x000000000FC7C000 000061 (v00 COREv4 COREBOOT 00000000 CORE 20210331)
[    0.006005] ACPI: HPET 0x000000000FC7B000 000038 (v01 COREv4 COREBOOT 00000000 CORE 20210331)
[... 761 lines cut ...]
[  423.172057] audit: type=1400 audit(1646152483.323:54): apparmor="DENIED" operation="open" profile="snap.snap-store.ubuntu-software" name="/etc/PackageKit/Vendor.conf" pid=1835 comm="snap-store" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[... cut ...]
[  836.505087] systemd[1]: systemd 248.3-1ubuntu8.2 running in system mode. (+PAM +AUDIT +SELINUX +APPARMOR +IMA +SMACK +SECCOMP +GCRYPT +GNUTLS -OPENSSL +ACL +BLKID +CURL +ELFUTILS -FIDO2 +IDN2 -IDN +IPTC +KMOD +LIBCRYPTSETUP -LIBFDISK +PCRE2 -PWQUALITY -P11KIT -QRENCODE +BZIP2 +LZ4 +XZ +ZLIB +ZSTD -XKBCOMMON +UTMP +SYSVINIT default-hierarchy=unified)
[  836.522256] systemd[1]: Detected architecture x86-64.
[  854.828556] audit: type=1326 audit(1646152914.921:71): auid=1000 uid=1000 gid=1000 ses=4 subj==snap.firefox.firefox (enforce) pid=6527 comm="MainThread" exe="/snap/firefox/1025/usr/lib/firefox/firefox" sig=0 arch=c000003e syscall=314 compat=0 ip=0x7fc487ecc89d code=0x50000`

Denisuu, 2022-03-01:
@lgirdwood I tried both, so I don't think it's a signing issue.
@J0shuaT Thanks for checking! Does sof get loaded at all on your machine running Ubuntu?

@plbossart I did all the steps as described. I also blacklisted `kvm` and `kvm_intel`  because the machine won't fully shut-down, it hangs on 'kvm: exiting hardware virtualization'

Here' s the output of dmesg: 

<details>
<summary> modprobe snd-sof-pci-intel-apl</summary>

```
[denisuu@BLOOGLET ~]$ sudo dmesg -C
[denisuu@BLOOGLET ~]$ modprobe snd-sof-pci-intel-apl
[denisuu@BLOOGLET ~]$ sudo dmesg
[   43.284104] audit: type=1106 audit(1646152810.361:91): pid=1148 uid=1000 auid=1000 ses=1 subj==unconfined msg='op=PAM:session_close grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[   43.284214] audit: type=1104 audit(1646152810.361:92): pid=1148 uid=1000 auid=1000 ses=1 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[   46.433488] audit: type=1101 audit(1646152813.510:93): pid=1150 uid=1000 auid=1000 ses=1 subj==unconfined msg='op=PAM:accounting grantors=pam_unix,pam_permit,pam_time acct="denisuu" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[   46.434604] audit: type=1110 audit(1646152813.511:94): pid=1150 uid=1000 auid=1000 ses=1 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[   46.437975] audit: type=1105 audit(1646152813.515:95): pid=1150 uid=1000 auid=1000 ses=1 subj==unconfined msg='op=PAM:session_open grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[   46.517556] audit: type=1106 audit(1646152813.594:96): pid=1150 uid=1000 auid=1000 ses=1 subj==unconfined msg='op=PAM:session_close grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[   46.517659] audit: type=1104 audit(1646152813.594:97): pid=1150 uid=1000 auid=1000 ses=1 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[   64.546438] audit: type=1101 audit(1646152831.623:98): pid=1162 uid=1000 auid=1000 ses=1 subj==unconfined msg='op=PAM:accounting grantors=pam_unix,pam_permit,pam_time acct="denisuu" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[   64.547505] audit: type=1110 audit(1646152831.624:99): pid=1162 uid=1000 auid=1000 ses=1 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[   64.550940] audit: type=1105 audit(1646152831.628:100): pid=1162 uid=1000 auid=1000 ses=1 subj==unconfined msg='op=PAM:session_open grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
``` 

</details>

plbossart, 2022-03-01:
@Denisuu @J0shuaT please attach your dmesg logs, it's horrible if you copy/paste in GitHub

J0shuaT, 2022-03-01:
@plbossart Here: https://pastebin.com/G5rDaEBn

plbossart, 2022-03-01:
> @plbossart Here: https://pastebin.com/G5rDaEBn

I meant use the GitHub "attach files' below the comment box. and sorry but no there is still nothing related to SOF in there.

J0shuaT, 2022-03-01:
> > @plbossart Here: https://pastebin.com/G5rDaEBn
> 
> I meant use the GitHub "attach files' below the comment box. and sorry but no there is still nothing related to SOF in there.

I am supposed to attach the dmesg as .txt file?

Denisuu, 2022-03-01:
The same here, still nothing sof related in dmesg. 

Could it be because it detects the wrong cards?

Manjaro & Ubuntu (Mr.Chromebox UEFI):
```
[denisuu@BLOOGLET ~]$ cat /proc/asound/cards
 0 [PCH            ]: HDA-Intel - HDA Intel PCH
                      HDA Intel PCH at 0x91418000 irq 122
```
Stock ChromeOS & Ubuntu (MilkyDeveloper's Breath):
```
chronos@localhost / $ cat /proc/asound/cards
 0 [sofglkda7219max]: sof-glkda7219ma - sof-glkda7219max
                      HP-Bloog-rev4-Bloog
```

Would be easiest that I revert to stock ChromeOS kernel and use Breath to boot Linux for further testing? At least we get this error.

```
Direct firmware load for intel/sof/sof-glk.ri failed with error -2
request firmware intel/sof/sof-glk.ri failed err: -2
```

I removed the 2nd comment with all the expandable logs and attached them as txt files to the 1st post. I did the same for all other posts.

plbossart, 2022-03-01:
> > > @plbossart Here: https://pastebin.com/G5rDaEBn
> > 
> > 
> > I meant use the GitHub "attach files' below the comment box. and sorry but no there is still nothing related to SOF in there.
> 
> I am supposed to attach the dmesg as .txt file?

yes. .txt or .log extensions work fine.

plbossart, 2022-03-01:
@Denisuu please attach the full dmesg log and alsa-info.
I have no idea what 'Breath' is, nor why you don't have the firmware in /lib/firmware/intel/sof/community/sof-glk.ri
To repeat: if the firmware files are not found there is nothing the SOF driver can do...

J0shuaT, 2022-03-01:
@plbossart this is Breath: https://cb-linux.github.io/breath/

J0shuaT, 2022-03-01:
@plbossart i'll just add my `alsa-info` here too: http://alsa-project.org/db/?f=05ce4765b25d7d68e3c1b41bd3cda9ed271f81c8

plbossart, 2022-03-01:
Same reply as in https://github.com/thesofproject/sof/issues/5439#issuecomment-1055847673, the DMI info does not match what we expect in a Chromebook so NOTHING will work.

@MrChromebox this is a disconnect here:

````
Manufacturer:      HP
Product Name:      Blooglet
Product Version:   rev4
Firmware Version:  MrChromebox-4.15
System SKU:        sku68
Board Vendor:      HP
Board Name:        Blooglet
````

v. what we use

````
.dmi_table = (const struct dmi_system_id []) {
			{
				.ident = "Google Chromebooks",
				.matches = {
					DMI_MATCH(DMI_SYS_VENDOR, "Google"),
				}
			},
			{}
		}
````

MrChromebox, 2022-03-01:
@plbossart same issue then, ugh. Will prevent the SMBIOS overrride.

Denisuu, 2022-03-01:
@plbossart 

So on ChromeOS you have 4 ways or running Linux:

- Crostini: Embedded Linux which runs in a VM on top of ChromeOS (Working audio)
- Crouton: Chroot (No working audio)
- Breath: Use the stock ChromeOS kernel to boot Linux from USB (or replace ChromeOS)
- FW change: This turns your laptop into a regular laptop (in theory not on GLK Chromebooks though)

The fist post was on Breath, I switched to FW change because @aiChaoSONG said it might be due to an old kernel. Which might be true since ChromeOS still uses  `4.14.256-18461-g6eaaee8fb397` 

[alsa-info.txt](https://github.com/thesofproject/sof/files/8164680/alsa-info.txt)
[dmesg.txt](https://github.com/thesofproject/sof/files/8164682/dmesg.txt)

plbossart, 2022-03-01:
@Denisuu The kernel version is meaningless for ChromeOS. It's only a base version, but the Chrome team will backport hundreds of patches, so you cannot know just with a version number what the kernel supports.

What we recommend is 5.10+ as a minimum for stability. Clearly a 4.14 base won't cut it or will only work with an ancient firmware.

MrChromebox, 2022-03-01:
@Denisuu @J0shuaT you'll need updated UEFI firmware from me to proceed further in troubleshooting (assuming the SMBIOS change doesn't resolve the issue completely). Not sure how to handle this for Breath though.

Denisuu, 2022-03-01:
@MrChromebox Ok. I read the 'Hp Chromebook elite c1030 no audio rt5682 SOF' issue to understand what's happening. Let us know when it's ready, no rush for me.

I'm hoping once we get it working on of the 2 it will be easier to get it working on the other.

MrChromebox, 2022-03-01:
@Denisuu shoot me an email with the board name and I'll build an update for you to try

J0shuaT, 2022-03-01:
@MrChromebox could I also try the updated firmware?

MrChromebox, 2022-03-01:
@J0shuaT email me with your board name

J0shuaT, 2022-03-01:
@MrChromebox alright, thanks.

J0shuaT, 2022-03-01:
@plbossart @MrChromebox I just flashed the firmware with the fix:
[image]

I still have the dummy audio output, but here is the new `alsa-info` output: http://alsa-project.org/db/?f=6a0ac2d7553fcf8746db94cc1c4351e0a486a99e

plbossart, 2022-03-01:
I think the DMI matching is still in the weeds.

here's what you have:
Manufacturer:      Google
Product Name:      Blooglet
Product Version:   rev4
Firmware Version:  MrChromebox-4.15
System SKU:        sku68
Board Vendor:      HP
Board Name:        Blooglet

here's what I have.

Manufacturer:      Google
Product Name:      Bobba
Product Version:   rev2
Firmware Version:  Google_Bobba.11825.0.2019_03_06_2015
System SKU:        sku2
Board Vendor:      Google
Board Name:        Bobba

I think DMI_SYS_VENDOR is the Board vendor, no?

MrChromebox, 2022-03-02:
> I think DMI_SYS_VENDOR is the Board vendor, no?

that would be DMI_BOARD_VENDOR, or at least I'd hope. I purposefully avoided changing the baseboard vendor to make this patch upstreamable

plbossart, 2022-03-02:
@MrChromebox that would be the rational explanation but from the results above I think there's some sort of inversion.

Denisuu, 2022-03-02:
I think the DMI information should be like this:
```
!!DMI Information
!!---------------

Manufacturer:      HP
Product Name:      Bloog
Product Version:   rev4
Firmware Version:  Google_Bloog.11297.275.0
System SKU:        sku68
Board Vendor:      HP
Board Name:        Bloog
```
`Blooglet` is listed as `Bloog` in the recovery file.

I went back to Breath to test for MilkyDeveloper: [Breath alsa-info](http://alsa-project.org/db/?f=33d2c0d009e8f0a9112fe94c48f49579b6cfed6d)
If I need to check something SMBIOS related in ChromeOS let me know.

iamnina, 2022-03-02:
> There's no issue loading the community firmware on GLK Chromebooks, it works fine here on mine. The issue appears to be isolated to models using the the da7219+max98357a combo; ones using rt5682+max98357a are working (at least for main speaker output).

Hey @MrChromebox , you stated rt5682+max98357a are working, does this apply for CometLake or just gemini? I have exactly that configuration. 
Is the fix applied in your test uefi rom? I would love to test, I have DRAGONAIR . ( ps. its nina, from reddit ) 
Cheers!

MrChromebox, 2022-03-02:
> Hey @MrChromebox , you stated rt5682+max98357a are working, does this apply for CometLake or just gemini? I have exactly that configuration.

just GLK. Cometlake seems to have a completely different issue.

J0shuaT, 2022-03-04:
@Denisuu I am on Manjaro with Mr. Chromeboxes firmware, and I can also confirm that the audio works! However, there is an audio scaling bug explained here: https://github.com/thesofproject/sof/issues/5361
[image]

lgirdwood, 2022-03-04:
@J0shuaT this could be the codec driver kcontrol has the wrong shift or mask values for this volume.

Denisuu, 2022-03-04:
> @Denisuu I am on Manjaro with Mr. Chromeboxes firmware, and I can also confirm that the audio works! However, there is an audio scaling bug explained here: #5361

Yea I have the same issue. When lowering sound from 100% to 6% volume stays the same, when going below 4% only the right speaker turns off. Looks like the same issue as you posted.

I've been using Manjaro eversince MrChromebox recommeded it, I absolutely love it! Especially the AUR and the ability to install AUR packages through GUI so I'm staying on Manjaro. But, I wanted to test Ubuntu again out of curiosity though. When you install [firmware-sof-signed_1.9-1ubuntu1_all.deb](https://launchpad.net/ubuntu/jammy/+package/firmware-sof-signed) audio works too. Still it takes around 12min to boot and trackpad is furstrating as hell. 

I don't have much time to troubleshoot until next weekend as I have an exam coming up. I'll probably test a bit more on Breath too and try to get Manjaro working on it (or Arch with GUI).

J0shuaT, 2022-03-04:
@lgirdwood Is there anything I can do to fix it? Maybe an application that limits the volume?

J0shuaT, 2022-03-07:
@Denisuu could you explain how you got working audio on Ubuntu with Mr. Chromeboxes firmware?

J0shuaT, 2022-03-08:
Nvm, it works OOB on Ubuntu 22.04!

lgirdwood, 2022-03-08:
> @lgirdwood Is there anything I can do to fix it? Maybe an application that limits the volume?

I would ask the Maxim folks to check the codec driver kcontrol for this volume is aligned to the codec registers (the shift or mask bits may be wrong). @ryans-lee is there someone who can help check the max98357a codec driver ?

J0shuaT, 2022-03-10:
> I would ask the Maxim folks to check the codec driver kcontrol for this volume is aligned to the codec registers (the shift or mask bits may be wrong). @ryans-lee is there someone who can help check [...]

Where should I go to ask?

lgirdwood, 2022-03-21:
> Where should I go to ask?

Maybe the fasted solution is to check the ASoC Maxim codec drivers authors (with maxim email addresses) and email them as well as CCing the ALSA devel mailing list.

Denisuu, 2022-05-07:
Reddit u/nsx2brz posted a workaround to get headphone-jack working on GLK devices. 
I can confirm it works on my HP 14a BLOOGLET on Manjaro kernel 5.17.1-3

Original post: [Workaround for GLK audio (sof-glkda7219max)](https://www.reddit.com/r/chrultrabook/comments/ui6ghn/workaround_for_glk_audio_sofglkda7219max/)

Run the following script as root: 
<details>
  <summary>Audio-jack fix</summary>

```
#!/bin/bash

echo Adding microphone to Pulseaudio

grep -qxF 'load-module module-alsa-source device=hw:0,99' /etc/pulse/default.pa || echo 'load-module module-alsa-source device=hw:0,99' >> /etc/pulse/default.pa

echo Adding headphone to Pulseaudio

grep -qxF 'load-module module-alsa-sink device=hw:0,1' /etc/pulse/default.pa || echo 'load-module module-alsa-sink device=hw:0,1' >> /etc/pulse/default.pa

echo Adding headset microphone to Pulseaudio

grep -qxF 'load-module module-alsa-source device=hw:0,1' /etc/pulse/default.pa || echo 'load-module module-alsa-source device=hw:0,1' >> /etc/pulse/default.pa

echo Restarting Pulseaudio

killall pulseaudio

echo Configuring headphone/headset mixer

amixer cset name='Playback Digital Volume' 111
amixer cset name='Playback Digital Switch' 1
amixer cset name='Out DACL Mux' DAIL
amixer cset name='Out DACR Mux' DAIR
amixer cset name='Mixer Out FilterL DACL Switch' 1
amixer cset name='Mixer Out FilterR DACR Switch' 1
amixer cset name='ST Mixer Out FilterL Out FilterL Switch' 1
amixer cset name='ST Mixer Out FilterR Out FilterR Switch' 1
amixer cset name='Headphone Jack Switch' on
amixer cset name='Headset Mic Switch' on
amixer cset name='Mic Volume' 5
amixer cset name='Capture Digital Volume' 111
amixer cset name='Capture Digital Switch' 1
amixer cset name='Mixin Volume' 10
amixer cset name='Mixin Switch' 1
amixer cset name='Mixer In Mic Switch' 1
amixer cset name='Out DAIL Mux' ADC
amixer cset name='Out DAIR Mux' ADC
amixer cset name='Pin5-Port0 Mux' 1
amixer cset name='Pin5-Port1 Mux' 1
amixer cset name='Pin5-Port2 Mux' 1
amixer cset name='Pin6-Port0 Mux' 2
amixer cset name='Pin6-Port1 Mux' 2
amixer cset name='Pin6-Port2 Mux' 2
amixer cset name='Gain Ramp Rate' 1
```
</details>

closed 2022-05-07 by Denisuu

ferrellsl, 2022-06-20:
Thanks Denisuu.  I confirm that the script above enables the audio jack and the mic on an Acer Chromebook CB311-9H and an HP Chromebook 14G6-4020-8 both running Ubuntu 22 and flashed with MrChromebox's UEFI firmware.  Thank you so much!

abhi-deshpande, 2024-02-29:
Hi Guys..! I know this is a old discussion, but I thought it would be useful to share the I trick found.
I have a Lenovo IdeaPad 3 Chromebook (Octopus Lick). Yesterday, I converted it to a Linux box by installing custom firmware by MrChromebox (Version MrChromebox-4.22.2-5-g822782a1001) and installed Linux Mint 21.3 Virginia.

Everything was working, except audio issues. I researched whole day to find a possible solution. Running the WeirdTreeThing audio script did not do anything. Changing some configs here and there did not help either. Finally I found an old gist [here](https://gist.github.com/unixabg/4ca86477b6ad0504d8b9486f1a160057) by unixabg. Although it was for Samsung Chromebook 4, it configures the same glkda7219max card, so I ran it and found that all my audio hardware is perfectly working. This script is similar to the above GLK Workaround, but it adds some ucm conf files.

Just the headset mic did not worked. But I found that the "Noisy Mic" that the script mentions is the headset mic. So I went to alsamixer and increased PGA3.0 3 Master to about 54%. Et Voila..! It started working. To prevent noise, we can mute it when not in use via pavucontrol.

I now have a 100% working audio setup. I need to switch devices manually, and sometimes microphones need some delay to start capture. But still it is worth..!

tarniye, 2024-03-12:
Sound work on linuxmint mate
[image]

tarniye, 2024-03-12:
[image]