thesofproject/sof #5442: [BUG] FW status on IPC timeout after 5.0 seconds in test 17_09_TestMulticoreKdWithResetCore1PlaybackD0ix16000Hz24b32b

kkarask, 2022-02-28:
**Describe the bug**
IPC timeout was found in test 17_09_TestMulticoreKdWithResetCore1PlaybackD0ix16000Hz24b32b

FW status on IPC timeout after 5.0 seconds: 0x00000005 0x00000000 0x00004000

In this test multi core with keyword detection feature with D0i3 support is checked.
Playback and capture on SSP interface on internal loopback. Playback pipeline on Core 1
and capture pipeline on Core 0

FW was build with chrome config.

**Topology**
```
Topology:

              pipe_plb1 core 1)
                     +-------------------------+
    +-----------+    | +------+   +----------+ |
    | Host1 plb |----->|Buffer|-->| Dai      |-----------------------------+
    +-----------+    | +------+   +----------+ |                           |
                     +-------------------------+                           |
                                                                           |
              pipe_cap core 0                                              v
    +--------------------------------------+                           +-------+
    | +----------+   +------+   +-----+    |                           |       |
    | | KD dummy |<--|Buffer|<--| Dai  |<----------------------------- |  SSP0 |
    | +----------+   +------+   +-----+    |                           |       |
    +--------------------------------------+                           +-------+
```

**To Reproduce**
Run test 17_09_TestMulticoreKdWithResetCore1PlaybackD0ix16000Hz24b32b with Diagnostic Driver

**Reproduction Rate**
~20%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF: Branch: main,  Hash: 58ce6e68608fdbec49a9835dfb8b76e0fa848187
2) Name of the platform(s) on which the bug is observed.
    * Platform: TGL

**Screenshots or console output**
[TGL_17_09_TestMulticoreKdWithResetCore1PlaybackD0ix16000Hz24b32b.zip](https://github.com/thesofproject/sof/files/8152266/TGL_17_09_TestMulticoreKdWithResetCore1PlaybackD0ix16000Hz24b32b.zip)

dkrzysiak, 2022-07-11:
New errors appeared, 'Nonetype' and 'DSP Panic'
[2022_07_08_14_07_52-731cceb8-1251-48dc-ae83-887c5a332d60.zip](https://github.com/thesofproject/sof/files/9085117/2022_07_08_14_07_52-731cceb8-1251-48dc-ae83-887c5a332d60.zip)
.

antonirx, 2023-01-24:
Not reproduced on TGL: 8bfcf59980b7302bc3621dc3b889ac279ad81ae4

closed 2023-01-24 by wszypelt