thesofproject/sof #5446: [BUG] [ADL][ZEPHYR] Failed to setup widget PIPELINE.5.SSP2.OUT

kv2019i, 2022-02-28:
**Describe the bug**
With fix to https://github.com/thesofproject/sof/issues/5018 , a new issue is uncovered with latest main of SOF+Zephyr.
sof-test multiple-pipeline-playback.sh fails with nocodec topology.

**To Reproduce**
export TPLG=/lib/firmware/intel/sof-tplg/sof-tgl-nocodec.tplg
cd sof-test/test-case
./multiple-pipeline-playback.sh -l 1

**Reproduction Rate**
100%

**Expected behavior**
sof-test case should pass

**Impact**
critical, audio only recovers via DSP power/D3 cycle

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 03fe3da8ee26f74ece29ed9d064806a56710c1bc (sof/topic/sof-dev) ASoC: SOF: sof-client support: sync up with upstream copyright date
    * SOF: main + https://github.com/thesofproject/sof/pull/5443 (main 6f71808e3ee74a8782f1a65cf7154b7f04df6ea2 (origin/main) xtensa-build-zephyr.py: add ipc4 build support for tgl )
    * Zephyr: main + https://github.com/zephyrproject-rtos/zephyr/pull/43267 (main 04ea1ee1afabb620d419a99c399d39d4ad721e9c (origin/main) boards: arm: sam: Enable hwinfo on supported boards )
2) Name of the topology file
    * Topology: sof-tgl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL-LP RVP

**Screenshots or console output**
Kernel:
```
[  400.548186] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x40070000: GLB_PM_MSG: CORE_ENABLE
[  400.548285] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x40070000: GLB_PM_MSG: CORE_ENABLE
[  400.548288] sof-audio-pci-intel-tgl 0000:00:1f.3: Core 1 powered up
[  400.548293] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[  400.548450] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x30100000 (msg/reply size: 48/20): -13
[  400.548456] sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to setup widget PIPELINE.5.SSP2.OUT
[  400.548461] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x40070000: GLB_PM_MSG: CORE_ENABLE
[  400.548559] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x40070000: GLB_PM_MSG: CORE_ENABLE
[  400.548563] sof-audio-pci-intel-tgl 0000:00:1f.3: Core 1 powered down
```

FW:
```
      268890.822649] (          19.010416) c2 pipe         2.11  ......./pipeline-graph.c:195  INFO pipeline_free()
[      269188.895553] (         298.072906) c0 ipc                  src/ipc/ipc3/handler.c:1579 INFO ipc: new cmd 0x40070000
[      269210.666386] (          21.770832) c0 ipc                  src/ipc/ipc3/handler.c:690  INFO ipc: pm core mask 0x1 -> enable
[      271823.270449] (        2612.604004) c0 ipc                  src/ipc/ipc3/handler.c:1579 INFO ipc: new cmd 0x40070000
[      271844.572531] (          21.302082) c0 ipc                  src/ipc/ipc3/handler.c:690  INFO ipc: pm core mask 0x5 -> enable
[      272065.301689] (         220.729156) c0 ipc                  src/ipc/ipc3/handler.c:1579 INFO ipc: new cmd 0x30100000
[      272151.135019] (          85.833328) c2 pipe               ......./pipeline-graph.c:114  INFO pipeline new pipe_id 1 priority 0
[      272529.416254] (         378.281250) c0 ipc                  src/ipc/ipc3/handler.c:1579 INFO ipc: new cmd 0x30010000
[      272630.562083] (         101.145828) c2 component             src/ipc/ipc3/helper.c:298  INFO comp new dai <c2b00d27-ffbc-4150-a51a-245c79c5e54b> type 2 id 1.2
[      272694.364164] (          63.802082) c2 power              ../cavs/lib/pm_runtime.c:184  INFO en_ssp_power index 0
[      272714.207913] (          19.843750) c2 power              ../cavs/lib/pm_runtime.c:193  INFO en_ssp_power I2SLCTL 00000101
[      272734.676662] (          20.468750) c2 dai                           src/lib/dai.c:164  INFO dai_get type 1 index 0 new sref 1
```

FW etrace
```
569447: SHM: FW ABI 0x3014001 DBG ABI 0x5003000 tags SOF:v2.0-rc1-556-g229646b3108f-dirty zephyr:zephyr-v3.0.0-501-gb433813f5fb5 src hash 0x229646b3 (ldc hash 0x229646b3)
633193 zephyr_ll_scheduler_init(): unsupported domain 2
1095872 ipc_process_on_core(): core #2 is disabled
1097208 zephyr_ll_scheduler_init(): unsupported domain 2
```
```
```

kv2019i, 2022-02-28:
This seems to be very timing sensitive:

# minimal failing case:
```
aplay   -D hw:0,2 -d 13 -c 2 -r 48000 -f S16_LE /dev/zero -q &
aplay   -D hw:0,3 -d 10 -c 2 -r 48000 -f S16_LE /dev/zero -q
```

# working case (just add delay)
```
aplay   -D hw:0,2 -d 13 -c 2 -r 48000 -f S16_LE /dev/zero -q &
sleep 1
aplay   -D hw:0,3 -d 10 -c 2 -r 48000 -f S16_LE /dev/zero -q
```

lgirdwood, 2022-03-04:
@kv2019i zephyr part is merged.

kv2019i, 2022-03-07:
This is now fixed in latest main (SOF + Zephyr), so closing. There is a remaining issue tracked in https://github.com/thesofproject/sof/issues/5477

closed 2022-03-07 by kv2019i

keqiaozhang, 2022-03-08:
Confirmed that this issue is fixed.