thesofproject/sof #5448: Apollolake  DMIC16k

EMLommers, 2022-02-28:
Why is dmic16k not supported?

If I include DMIC16k I get errors like:
soc_pcm_hw_params failed (-22)
soc_dpcm_fe_dai_hw_params failed (-22)

Is this a sof-ap.ri issue or feature not implemented?

plbossart, 2022-03-01:
@EMLommers can you attach the log of 'alsa-info' and provide the exact commands that you used.

DMIC16k means it's 16kHz.

EMLommers, 2022-03-01:
[    0.148205] ACPI: Added _OSI(Linux-Dell-Video)
[    0.148206] ACPI: Added _OSI(Linux-Lenovo-NV-HDMI-Audio)
[    0.148208] ACPI: Added _OSI(Linux-HPI-Hybrid-Graphics)
--
[    8.286340] intel_rapl_common: Found RAPL domain dram
[    8.397944] sof-audio-pci-intel-apl 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if 0x040100
[    8.747714] sof-audio-pci-intel-apl 0000:00:0e.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    8.753319] sof-audio-pci-intel-apl 0000:00:0e.0: use msi interrupt mode
[    8.793168] sof-audio-pci-intel-apl 0000:00:0e.0: hda codecs found, mask 4
[    8.795713] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware info: version 2:0:0-b678a
[    8.795722] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware: ABI 3:20:0 Kernel ABI 3:18:0
[    8.795726] sof-audio-pci-intel-apl 0000:00:0e.0: warn: FW ABI is more recent than kernel
[    8.795738] sof-audio-pci-intel-apl 0000:00:0e.0: unknown sof_ext_man header type 3 size 0x30
[    8.844647] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware info: version 2:0:0-b678a
[    8.844660] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware: ABI 3:20:0 Kernel ABI 3:18:0
[    8.844664] sof-audio-pci-intel-apl 0000:00:0e.0: warn: FW ABI is more recent than kernel
[    8.920790] iwlwifi 0000:01:00.0 wlp1s0: renamed from wlan0
--
[    9.282736] NET: Registered PF_ALG protocol family
[    9.465539] snd_soc_intel_hda_dsp_common: loading out-of-tree module taints kernel.
[    9.465636] snd_soc_intel_hda_dsp_common: module verification failed: signature and/or required key missing - tainting kernel
[    9.476948] sof-audio-pci-intel-apl 0000:00:0e.0: Topology: ABI 3:20:0 Kernel ABI 3:18:0
[    9.476961] sof-audio-pci-intel-apl 0000:00:0e.0: warn: topology ABI is more recent than kernel
[    9.488514] sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: Parent card not yet available, widget card binding deferred
[    9.490237] da7219 i2c-DLGS7219:00: supply VDD not found, using dummy regulator
--
[    9.518841] bxt_da7219_max98357a bxt_da7219_mx98357a: ASoC: source widget Capture overwritten
[    9.536885] input: sof-bxtda7219max Headset Jack as /devices/pci0000:00/0000:00:0e.0/bxt_da7219_mx98357a/sound/card0/input8
[    9.537252] input: sof-bxtda7219max HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:0e.0/bxt_da7219_mx98357a/sound/card0/input9
[    9.537705] input: sof-bxtda7219max HDMI/DP,pcm=6 as /devices/pci0000:00/0000:00:0e.0/bxt_da7219_mx98357a/sound/card0/input10
[    9.538050] input: sof-bxtda7219max HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:0e.0/bxt_da7219_mx98357a/sound/card0/input11
[   12.216144] da7219 i2c-DLGS7219:00: SRM failed to lock
--
[   12.721169] wlp1s0: associated
[   12.762915] sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc error for 0x60010000 size 20
[   12.762932] sof-audio-pci-intel-apl 0000:00:0e.0: error: hw params ipc failed for stream 1
[   12.762936] sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:0e.0: -22
[   12.763193]  DMIC16k: ASoC: soc_pcm_hw_params() failed (-22)
[   12.763408]  DMIC16k: ASoC: dpcm_fe_dai_hw_params failed (-22)
[   12.765311] sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc error for 0x60010000 size 20
[   12.765322] sof-audio-pci-intel-apl 0000:00:0e.0: error: hw params ipc failed for stream 1
[   12.765325] sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:0e.0: -22
[   12.765593]  DMIC16k: ASoC: soc_pcm_hw_params() failed (-22)
[   12.765810]  DMIC16k: ASoC: dpcm_fe_dai_hw_params failed (-22)
[   12.766884] sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc error for 0x60010000 size 20
[   12.766890] sof-audio-pci-intel-apl 0000:00:0e.0: error: hw params ipc failed for stream 1
[   12.766893] sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:0e.0: -22
[   12.767109]  DMIC16k: ASoC: soc_pcm_hw_params() failed (-22)
[   12.767319]  DMIC16k: ASoC: dpcm_fe_dai_hw_params failed (-22)
[   12.770119] sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc error for 0x60010000 size 20
[   12.770131] sof-audio-pci-intel-apl 0000:00:0e.0: error: hw params ipc failed for stream 1
[   12.770135] sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:0e.0: -22
[   12.770411]  DMIC16k: ASoC: soc_pcm_hw_params() failed (-22)
--
[   12.804570] wlp1s0: Limiting TX power to 20 (20 - 0) dBm as advertised by 92:5c:14:2a:0b:d1
[   12.809627] sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc error for 0x60010000 size 20
[   12.809646] sof-audio-pci-intel-apl 0000:00:0e.0: error: hw params ipc failed for stream 1
[   12.809651] sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:0e.0: -22
[   12.810243]  DMIC16k: ASoC: soc_pcm_hw_params() failed (-22)
[   12.810830]  DMIC16k: ASoC: dpcm_fe_dai_hw_params failed (-22)
[   12.815334] sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc error for 0x60010000 size 20
[   12.815352] sof-audio-pci-intel-apl 0000:00:0e.0: error: hw params ipc failed for stream 1
[   12.815356] sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:0e.0: -22
[   12.815734]  DMIC16k: ASoC: soc_pcm_hw_params() failed (-22)
[   12.815964]  DMIC16k: ASoC: dpcm_fe_dai_hw_params failed (-22)
[   12.818282] sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc error for 0x60010000 size 20
[... cut ...]
[   33.186881]  DMIC16k: ASoC: soc_pcm_hw_params() failed (-22)
[   33.187111]  DMIC16k: ASoC: dpcm_fe_dai_hw_params failed (-22)
[   33.226095] sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc error for 0x60010000 size 20
[   33.226113] sof-audio-pci-intel-apl 0000:00:0e.0: error: hw params ipc failed for stream 1
[   33.226117] sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:0e.0: -22
[   33.226409]  DMIC16k: ASoC: soc_pcm_hw_params() failed (-22)

lgirdwood, 2022-03-01:
@singalsu any comments ?

EMLommers, 2022-03-01:
http://alsa-project.org/db/?f=4a89153604b5867a98920fa34f5ba8d3f1569d2e

EMLommers, 2022-03-01:
Do you need my .tplg file?

lgirdwood, 2022-03-01:
> Do you need my .tplg file?

Yes, the text (M4) version is better.

plbossart, 2022-03-01:
@EMLommers what exact command did you use? DMIC16k means you have to use a 16khz sample rate for arecord.

plbossart, 2022-03-01:
> I noticed in the driver (bxt_da7219_max98537a.c) it should be possible and wanted to test.
> 
> Attached files. and how I created them [sof-tplg-and-sof-tplg-platform-convert.zip](https://github.com/thesofproject/sof/files/8162498/sof-tplg-and-sof-tplg-platform-convert.zip)

that cannot work, it's not possible to just do string substitutions in binary topology files.

In addition, we already released a topology in sof-bin, see https://github.com/thesofproject/sof-bin/blob/main/v2.0.x/sof-tplg-v2.0/sof-apl-da7219.tplg

EMLommers, 2022-03-01:
That cannot work? Test it and I show you it is possible!
The tplg for apl I did as well. it works like a charm!

plbossart, 2022-03-01:
@EMLommers You can choose to ignore our technical advice but I can also choose to stop providing answers. Over and out.

EMLommers, 2022-03-01:
Why are you linking to a tplg file, which is not working!?  The routes of iDisp3 Tx , iDisp2 Tx, IDisp1 Tx are not set.
Use my script and the routes are working. It is not technical advice. If there is an official sof topology available I will use it.

plbossart, 2022-03-01:
This was fixed already in https://github.com/thesofproject/sof/pull/5401.
Binary attached:

[sof-apl-da7219.tplg.gz](https://github.com/thesofproject/sof/files/8163674/sof-apl-da7219.tplg.gz)

EMLommers, 2022-03-01:
I know thank you very much! But this does not include the dmic16k yet.

plbossart, 2022-03-01:
why do you need it? it's not very useful in Linux.

EMLommers, 2022-03-01:
I do not need it, just liked to test possibilities of the driver, noticed it was available in the CML (7219+98357a) and GLK toplogy, but not in the APL.

EMLommers, 2022-03-01:
I am running a Asus C223NA Chromebook, and was working on on the ucm, to get audio fixed. In the original google ucm are the hotword and 1mic and 2mic ucms, depending on the model type. Like section below was not working.

SectionModifier."Hotword Model en_us".0 {
	EnableSequence [
		cdev "hw:sof-bxtda7219max"
		cset-tlv "name='hwd_in hwd 0 mdl params' /opt/google/bxt-hotword-support/en_us.hwd-blob"
	]

	DisableSequence [
	]
}

plbossart, 2022-03-01:
@EMLommers you can remove all this modifier, this is not supported with SOF public releases. this hotwording thing is only part of ChromeOS and released by Google, not by this team.

EMLommers, 2022-10-02:
I understand... 
ChromeOS is also moving to SOF and using sspXc_be and fe .. 
LG fully understand the issue and how to resolve the sound issues, it's politics... ACPI

closed 2022-10-02 by EMLommers