thesofproject/sof #5457: [FEATURE] Branch protection rules for `mt8195/*`

afq984, 2022-03-02:
**Is your feature request related to a problem? Please describe.**
Prevent accidentally overwriting code in `mt8195/*` branches which are used to build firmware releases.
Currently the only branch that matches this pattern is `mt8195/v0.4`.

**Describe the solution you'd like**
Set branch protection rules on `mt8195/*`.

**Describe alternatives you've considered**
Be careful to not use `git push --force ...`

lgirdwood, 2022-03-02:
@afq984 who should have Write access ?

dbaluta, 2022-03-02:
@lgirdwood since we are here could also give write access for imx-stable-v2.0 only to @dbaluta @paulstelian97 and @TangleZ

lgirdwood, 2022-03-02:
@dbaluta Sure. 
@balakishorepati @bhiregou any similar request ?

lgirdwood, 2022-03-02:
@dbaluta @afq984 done - please try.

cujomalainey, 2022-03-02:
@afq984 are we good to close this?

afq984, 2022-03-03:
I'll let @dbaluta close this

afq984, 2022-03-03:
@lgirdwood ah, can the protection rules be lifted to just "protection" only? i.e. drop all extra protection such as signed commits, PR, status checks, just prevent force push & deletion

afq984, 2022-03-03:
Works for me, thanks again!

bhiregoudar, 2022-03-03:
> @dbaluta Sure. @balakishorepati @bhiregou any similar request ?

@lgirdwood : We come back tomorrow on this.

bhiregoudar, 2022-03-08:
> > @dbaluta Sure. @balakishorepati @bhiregou any similar request ?
> 
> @lgirdwood : We come back tomorrow on this.

Could you help add @bhiregoudar @balakishorepati @sunilkumardommati @maruthi-machani for amd-rn-stable branch

lgirdwood, 2022-03-09:
> > > @dbaluta Sure. @balakishorepati @bhiregou any similar request ?
> > 
> > 
> > @lgirdwood : We come back tomorrow on this.
> 
> Could you help add @bhiregoudar @balakishorepati @sunilkumardommati @maruthi-machani for amd-rn-stable branch

Done, but I can't add @maruthi-machani yet until GH invite is accepted. @maruthi-machani please let me know when you accept the invite.

dbaluta, 2022-03-09:
All good for imx-stable-v2.0 branch. @lgirdwood can this rules be inherited for future imx-stable-* branches as @cujomalainey suggested?

lgirdwood, 2022-03-10:
> All good for imx-stable-v2.0 branch. @lgirdwood can this rules be inherited for future imx-stable-* branches as @cujomalainey suggested?

Yes, I've made the wildcard "imx-*" so rules should apply to all IMX branches.

lgirdwood, 2022-03-10:
Good to close ?

maruthi-machani, 2022-03-14:
> Done, but I can't add @maruthi-machani yet until GH invite is accepted. @maruthi-machani please let me know when you accept the invite.

Accepted the requested.