thesofproject/sof #5470: [BUG][IPC4] SSP playback ipc timed out with SOF ipc4

XiaoyunWu6666, 2022-03-04:
**Describe the bug**
ipc timed out when playback with SSP 

**To Reproduce**
 aplay   -Dhw:0,0 -r 48000 -c 2 -f S16_LE -d 3 /dev/zero -v -q
**Reproduction Rate**
100%
**Environment**
Kernel Branch: integration/ipc4-wip
Kernel Commit: [a80183ea](https://github.com/thesofproject/linux/commit/a80183ea)
SOF Branch: main
SOF Commit: [9709ef81dcb8](https://github.com/thesofproject/sof/commit/9709ef81dcb8)
Zephyr Commit: [7e9e99379679](https://github.com/zephyrproject-rtos/zephyr/commit/7e9e99379679)
 Topology:cavs-tgl-nocodec.tplg
 Platform: ADLP_RVP_NOCODEC_IPC4
[image]

**Screenshots or console output**
IO error when playing 
[dmesg]
```
[   58.916988] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget pipeline.1 setup complete
[   58.916992] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Create widget copier.host.1.1 instance 0 - pipe 1 - core 0
[   58.917004] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x40000004|0x10015: MOD_INIT_INSTANCE [data size: 84]
[   59.420879] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x40000004|0x10015
[   59.420891] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to create module copier.host.1.1
[   59.420898] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x12010000|0x0: GLB_DELETE_PIPELINE
[   59.924914] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x12010000|0x0
[   59.924926] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to free pipeline widget pipeline.1
[   59.924931] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to set up connected widgets
[   59.925305] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[   59.925908] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x14001e successful
[   59.925918] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 1 state 3
[   59.925924] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13010003|0x0: GLB_SET_PIPELINE_STATE
[   60.428906] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13010003|0x0
[   60.428918] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to pause pipeline 1
[   60.428923] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[   60.428929] kernel:  Port0: ASoC: trigger FE cmd: 1 failed: -110
[   60.429003] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[   60.429008] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 1 state 3
[   60.429012] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13010003|0x0: GLB_SET_PIPELINE_STATE
[   60.932899] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13010003|0x0
[   60.932911] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to pause pipeline 1
[   60.932923] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x140000 successful
[   60.932930] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[   60.932936] kernel:  Port0: ASoC: trigger FE cmd: 0 failed: -110
[   61.437375] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: free stream 0 dir 0
[   61.437402] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 1 state 3
[   61.437416] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13010003|0x0: GLB_SET_PIPELINE_STATE
[   61.941046] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13010003|0x0
[   61.941082] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to pause pipeline 1
[   61.941136] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_free on 0000:00:1f.3: -110
[   61.941375] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: close stream 0 dir 0

```

lgirdwood, 2022-03-04:
@mwasko @marcinszkudlinski  fyi - @RanderWang should be able to look at this on Monday unless someone can solve it today.

RanderWang, 2022-03-04:
@lgirdwood there is a bug in Linux driver, so we can't fix it in FW only

XiaoyunWu6666, 2022-03-15:
This cannot be observed in IPC4 test results now because of another blocker issue : DSP firmware cannot boot successfully with sof + IPC4 because of module UUID out of range.
Potential fix PR on zephyr side via @aiChaoSONG and @RanderWang  https://github.com/zephyrproject-rtos/zephyr/pull/43571/files

closed 2022-03-18 by lgirdwood