thesofproject/sof #5473: [BUG]

bruceyurepo, 2022-03-04:
**Describe the bug**
When I use sof-fuzzer to load tplg to qemu dsp, it will always crash .

**To Reproduce**
1. use the latest sof source code (9709ef81dcb8e9630c55) and build sof-byt.ri
2. start up the the qemu dsp by using byt with kernel sof-byt.ri
3. try to use sof-fuzzer to load sof-byt-nocodec.tplg
4. sof-fuzzer will crash as sof-fuzzer 

**Reproduction Rate**
100%

**Expected behavior**
should load tplg success

**Impact**
None

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Qemu DSP with BYT

**Screenshots or console output**

#./sof-fuzzer -t sof-byt-nocodec.tplg -p byt 

initialising platform byt
registering iram-mem
registering dram-mem
registering shim-io
registering mbox-io
msg: id 0 msg 64 size 16 type 2
irq: masking 0 mask 0 active 0
ipc: host box 0x400 size 0x400
ipc: dsp box 0x0 size 0x400
ipc: FW version major: 2 minor: 0 tag: 9709e
irq: send done interrupt 0x4000000070028800
irq: masking 3 mask 0 active 2
FW boot complete
debug: topology parsing start
debug type: 8, size: 0x73 count: 1 index: 0
debug type: 5, size: 0x780 count: 7 index: 1
debug number of DAPM widgets 7

Segmentation fault

___________________________________________________
the output from DSP QEMU:

build/xtensa-softmmu/qemu-system-xtensa -cpu baytrail -M adsp_byt -nographic -kernel /home/e0004400/sof/sof/build_byt_gcc/sof-byt.ri -semihosting
bridge-io: qemu-bridge-iram-mem fd 9 region 1 at 0x7f5f4adf1000 allocated 81920 bytes
bridge-io: qemu-bridge-dram-mem fd 10 region 2 at 0x7f5f4ad6d000 allocated 163840 bytes
bridge-io: qemu-bridge-dmac0-io fd 11 region 3 at 0x7f5f4adf0000 allocated 4096 bytes
bridge-io: qemu-bridge-dmac1-io fd 12 region 4 at 0x7f5f4adef000 allocated 4096 bytes
bridge-io: qemu-bridge-ssp0-io fd 13 region 5 at 0x7f5f4adee000 allocated 4096 bytes
bridge-io: qemu-bridge-ssp1-io fd 14 region 6 at 0x7f5f4aded000 allocated 4096 bytes
bridge-io: qemu-bridge-ssp2-io fd 15 region 7 at 0x7f5f4adec000 allocated 4096 bytes
bridge-io: qemu-bridge-shim-io fd 16 region 8 at 0x7f5f4adeb000 allocated 4096 bytes
bridge-io: qemu-bridge-mbox-io fd 17 region 9 at 0x7f5f4adea000 allocated 4096 bytes
bridge-io-mq: added /qemu-io-parent-byt
bridge-io-mq: added /qemu-io-child-byt
bridge-io: 0 messages are currently on child queue.
header size=0x12a70 modules=0x1 abi=0x1 size=16
new module size 0x12a64 blocks 0xf type 0x0
data: 0xff2c0000 size 0x16c
block 0 type 0x1 size 0x16c ==>  offset 0xc0000
data: 0xff2c0400 size 0x16c
block 1 type 0x1 size 0x16c ==>  offset 0xc0400
data: 0xff2c05bc size 0x8
block 2 type 0x1 size 0x8 ==>  offset 0xc05bc
data: 0xff2c05dc size 0x8
block 3 type 0x1 size 0x8 ==>  offset 0xc05dc
data: 0xff2c05fc size 0x4
block 4 type 0x1 size 0x4 ==>  offset 0xc05fc
data: 0xff2c061c size 0x4
block 5 type 0x1 size 0x4 ==>  offset 0xc061c
data: 0xff2c063c size 0x8
block 6 type 0x1 size 0x8 ==>  offset 0xc063c
data: 0xff2c0658 size 0x4
block 7 type 0x1 size 0x4 ==>  offset 0xc0658
data: 0xff2c065c size 0x18
block 8 type 0x1 size 0x18 ==>  offset 0xc065c
data: 0xff2c067c size 0x8
block 9 type 0x1 size 0x8 ==>  offset 0xc067c
data: 0xff2c06c0 size 0x113f8
block 10 type 0x1 size 0x113f8 ==>  offset 0xc06c0
data: 0xff300008 size 0xa70
block 11 type 0x2 size 0xa70 ==>  offset 0x100008
data: 0xff300a78 size 0x10
block 12 type 0x2 size 0x10 ==>  offset 0x100a78
data: 0xff300a88 size 0x7b0
block 13 type 0x2 size 0x7b0 ==>  offset 0x100a88
data: 0xff3050b8 size 0x6c
block 14 type 0x2 size 0x6c ==>  offset 0x1050b8
QEMU 4.2.0 monitor - type 'help' for more information
(qemu) irq: SHIM_IMRLPESC masking 0 mask 0 active 0
irq: SC send busy interrupt 0xc2
irq: SHIM_IMRLPESC masking 1 mask 1 active 0
irq: SHIM_IMRLPESC masking 0 mask 0 active 0
irq: SC send busy interrupt 0xc5
irq: SHIM_IMRLPESC masking 1 mask 1 active 0
irq: SHIM_IMRLPESC masking 0 mask 0 active 0
IRQ: from DMAC 1 status 0 block 0 tfr 0
IRQ: from DMAC 1 status 0 block 0 tfr 0
irq: IMRD masking 0 mask 0 active 0
irq: PIMR masking 0 mask 38 active 0
irq: send busy interrupt 0x80000000
bridge-io: msg send: 0 type 2 msg 64 size 16 ret 0
bridge-io: msg recv 0 type 2 size 16 msg 64
msg: id 0 msg 64 size 16 type 2
IRQ: from HOST status 3 mask 0 active 3 cmd 0
irq: IMRD masking 3 mask 1 active 2
irq: IMRD masking 3 mask 0 active 3
irq: IMRD masking 3 mask 2 active 1
irq: IMRD masking 3 mask 3 active 0
irq: IMRD masking 3 mask 2 active 1
irq: send done interrupt 0x40000000
bridge-io: msg send: 1 type 2 msg 64 size 16 ret 0
irq: IMRD masking 3 mask 0 active 3

lgirdwood, 2022-03-04:
@bruceyurepo are you able to run under valgrind and produce a error dump. e.g.
```valgrind -tool=memcheck ./sof-fuzzer -t sof-byt-nocodec.tplg -p byt```

bruceyurepo, 2022-03-04:
Hi @lgirdwood 

When I use valgrind to test, seems no **Segmentation fault** but a **IPC timeout**

please see below dump log:

_test@ubuntu:~/sof/sof/tools/build_tools/fuzzer$ sudo LD_LIBRARY_PATH=$PATH:`pwd`/sof_parser/build/ valgrind --tool=memcheck ./sof-fuzzer -t ../topology/sof-byt-nocodec.tplg -p byt
==11525== Memcheck, a memory error detector
==11525== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==11525== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==11525== Command: ./sof-fuzzer -t ../topology/sof-byt-nocodec.tplg -p byt
==11525==
initialising platform byt
registering iram-mem
registering dram-mem
registering shim-io
registering mbox-io
=====msg: id 0 msg 64 size 16 type 2
=====irq: masking 0 mask 0 active 0
=====ipc: host box 0x400 size 0x400
=====ipc: dsp box 0x0 size 0x400
=====ipc: FW version major: 2 minor: 0 tag: 9709e
=====irq: send done interrupt 0x4000000070028800
=====irq: masking 3 mask 0 active 1
FW boot complete
=====msg: id 1 msg 64 size 16 type 2
=====irq: masking 3 mask 0 active 0
=====ipc: reply msg from DSP
error: incorrect DSP reply
==11525== Thread 2 io-bridge-byt:
==11525== Use of uninitialised value of size 8
==11525==    at 0x53B57E1: _itoa_word (_itoa.c:180)
==11525==    by 0x53B8EDD: vfprintf (vfprintf.c:1642)
==11525==    by 0x53C1E13: fprintf (fprintf.c:32)
==11525==    by 0x10B92E: ??? (in /home/test/sof/sof/tools/build_tools/fuzzer/sof-fuzzer)
==11525==    by 0x10BDD2: ??? (in /home/test/sof/sof/tools/build_tools/fuzzer/sof-fuzzer)
==11525==    by 0x10ED34: ??? (in /home/test/sof/sof/tools/build_tools/fuzzer/sof-fuzzer)
==11525==    by 0x10F096: ??? (in /home/test/sof/sof/tools/build_tools/fuzzer/sof-fuzzer)
==11525==    by 0x10C453: ??? (in /home/test/sof/sof/tools/build_tools/fuzzer/sof-fuzzer)
==11525==    by 0x4EB22A4: ??? (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.5600.4)
==11525==    by 0x59C76DA: start_thread (pthread_create.c:463)
==11525==    by 0x547E61E: clone (clone.S:95)
==11525==
==11525== Conditional jump or move depends on uninitialised value(s)
==11525==    at 0x53B57E8: _itoa_word (_itoa.c:180)
==11525==    by 0x53B8EDD: vfprintf (vfprintf.c:1642)
==11525==    by 0x53C1E13: fprintf (fprintf.c:32)
==11525==    by 0x10B92E: ??? (in /home/test/sof/sof/tools/build_tools/fuzzer/sof-fuzzer)
==11525==    by 0x10BDD2: ??? (in /home/test/sof/sof/tools/build_tools/fuzzer/sof-fuzzer)
==11525==    by 0x10ED34: ??? (in /home/test/sof/sof/tools/build_tools/fuzzer/sof-fuzzer)
==11525==    by 0x10F096: ??? (in /home/test/sof/sof/tools/build_tools/fuzzer/sof-fuzzer)
==11525==    by 0x10C453: ??? (in /home/test/sof/sof/tools/build_tools/fuzzer/sof-fuzzer)
==11525==    by 0x4EB22A4: ??? (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.5600.4)
==11525==    by 0x59C76DA: start_thread (pthread_create.c:463)
==11525==    by 0x547E61E: clone (clone.S:95)
==11525==
==11525== Conditional jump or move depends on uninitialised value(s)
==11525==    at 0x53B8FE4: vfprintf (vfprintf.c:1642)
==11525==    by 0x53C1E13: fprintf (fprintf.c:32)
==11525==    by 0x10B92E: ??? (in /home/test/sof/sof/tools/build_tools/fuzzer/sof-fuzzer)
==11525==    by 0x10BDD2: ??? (in /home/test/sof/sof/tools/build_tools/fuzzer/sof-fuzzer)
==11525==    by 0x10ED34: ??? (in /home/test/sof/sof/tools/build_tools/fuzzer/sof-fuzzer)
==11525==    by 0x10F096: ??? (in /home/test/sof/sof/tools/build_tools/fuzzer/sof-fuzzer)
==11525==    by 0x10C453: ??? (in /home/test/sof/sof/tools/build_tools/fuzzer/sof-fuzzer)
==11525==    by 0x4EB22A4: ??? (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.5600.4)
==11525==    by 0x59C76DA: start_thread (pthread_create.c:463)
==11525==    by 0x547E61E: clone (clone.S:95)
==11525==
==11525== Conditional jump or move depends on uninitialised value(s)
==11525==    at 0x53B9B1C: vfprintf (vfprintf.c:1642)
[... 132 lines cut ...]
==11525==    by 0x10C453: ??? (in /home/test/sof/sof/tools/build_tools/fuzzer/sof-fuzzer)
==11525==    by 0x4EB22A4: ??? (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.5600.4)
==11525==    by 0x59C76DA: start_thread (pthread_create.c:463)
==11525==    by 0x547E61E: clone (clone.S:95)
==11525==
ipc: header 0x30010000 size 64 reply 20
=====irq: masking 0 mask 2 active 1
error: IPC timeout
ipc: header 0x30010000 size 64 reply 20
==11525==
==11525== HEAP SUMMARY:
==11525==     in use at exit: 2,120,613,716 bytes in 18 blocks
==11525==   total heap usage: 27 allocs, 9 frees, 2,120,621,678 bytes allocated
==11525==
==11525== LEAK SUMMARY:
==11525==    definitely lost: 0 bytes in 0 blocks
==11525==    indirectly lost: 0 bytes in 0 blocks
==11525==      possibly lost: 272 bytes in 1 blocks
==11525==    still reachable: 2,120,613,444 bytes in 17 blocks
==11525==         suppressed: 0 bytes in 0 blocks
==11525== Rerun with --leak-check=full to see details of leaked memory
==11525==
==11525== For counts of detected and suppressed errors, rerun with: -v
==11525== Use --track-origins=yes to see where uninitialised values come from
==11525== ERROR SUMMARY: 97 errors from 20 contexts (suppressed: 0 from 0)_

bruceyurepo, 2022-03-07:
Any update here ?

closed 2022-06-08 by bruceyurepo