thesofproject/sof #5474: [FEATURE] Easier/scalable way to resize memory maps

cujomalainey, 2022-03-04:
**Is your feature request related to a problem? Please describe.**
Resizing memory regions to fit various use cases is not trivial in the current implementation nor is it "configurable" in the sense that you can multiple layouts and easily pick from them

**Describe the solution you'd like**
We need to first split the device memory layout from the region layout, then provide a configurable way to size the regions

**Describe alternatives you've considered**
N/A, current method is a pain and hard to understand

**Additional context**
No idea what/how zephyr is going to affect this problem, might already be solved by the system

cujomalainey, 2022-03-04:
@lkoenig FYI

lgirdwood, 2022-03-04:
Zephyr allocator should fix this.