thesofproject/sof #5480: [BUG][Zephyr] DMA_LL scheduler init fails on secondary cores

kv2019i, 2022-03-07:
**Describe the bug**
DMA_LL scheduler is not supported in Zephyr builds. 

Additionally following error is printed for each secondary core boot:
[       25515.832319] (         403.124969) c1 zll-schedule       src/schedule/zephyr_ll.c:527  ERROR zephyr_ll_scheduler_init(): unsupported domain 2

Related to https://github.com/thesofproject/sof/issues/5477

lgirdwood, 2022-03-07:
@kv2019i this is intentional - DMA driven work is asynchronous and better served under Zephyr as DMA triggered IRQs and preemptable worker threads (with varying thread priority)

kv2019i, 2022-03-07:
@lgirdwood We should probably get rid of the error print and leave this out from the build.

lgirdwood, 2022-03-08:
> @lgirdwood We should probably get rid of the error print and leave this out from the build.

I would leave it but make the message more meaningful. This make it obvious for anyone transferring a topology.