thesofproject/sof #5493: [BUG][TGL-H] firmware failed to boot on TGL-H laptop

keqiaozhang, 2022-03-09:
**Describe the bug**
Firmware failed to boot on TGL-H laptop. Confirmed that this is a firmware regression. It should be caused by https://github.com/thesofproject/sof/commit/95d887251ee40397300c570568d079ac9f9c7db4
error message:
```
[  317.017988] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware download successful, booting...
[  319.031586] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  319.031597] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware boot failure due to timeout
[  319.031601] sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_IN_PROGRESS (2)
[  319.031639] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown ROM status value 80000005
[  319.031727] sof-audio-pci-intel-tgl 0000:00:1f.3: invalid header size 0x3010202. FW oops is bogus
[  319.031731] sof-audio-pci-intel-tgl 0000:00:1f.3: unexpected fault 0xdead000f trace 0x00000222
[  319.031736] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  319.031739] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to boot DSP firmware -5
[  319.031742] sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state change: 2 -> 3
[  319.034273] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0x3010303 successful
[  319.034861] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0x303 successful
[  319.034867] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[  319.036050] sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state change: 3 -> 0
[  319.036120] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sof_probe_work failed err: -5
```

**To Reproduce**
check the dmesg after booting 

**Reproduction Rate**
100%

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
*kernel: sof-dev/8a2fe3d05
  *SOF: tip main 
 * Platform: TGLH_SKU0A70_HDA

**dmesg**

[dmesg.txt](https://github.com/thesofproject/sof/files/8211430/dmesg.txt)

marc-hb, 2022-03-09:
@keqiaozhang thanks for bisecting this, very useful. Could also list which TGL models are _not_ affected?

marc-hb, 2022-03-09:
I think I have a good guess. i think it's not  95d887251ee40 but it's the next commit 0affed20e9d9:

```shell
case $platform in ...
         ....
		tgl|tgl-h)
			PLATFORM="tgplp"
        ...
```

```

marc-hb, 2022-03-09:
@keqiaozhang do you have a device where you can test revert #5495?

keqiaozhang, 2022-03-09:
> @keqiaozhang thanks for bisecting this, very useful. Could also list which TGL models are _not_ affected?

This issue affects only one platform: TGLH_SKU0A70_HDA.

PR #5501 (2022-03-09): xtensa-build-all.sh: Use appropriate defconfig for tgl-h platform
Recent combination of tgl and tgl-h building resulted in use of an invalid defconfig file for tgl-h platform. This PR reverts that change.

Fixes #5493
closed 2022-03-09 by ranj063